project-chip/connectedhomeip #32489: [Workflow] [ESP32] Chef - ESP32 CI Examples failed

lboue, 2024-03-07:
### Workflow issues

## Chef - ESP32 CI Examples failed
Chef - ESP32 CI Examples failed. See my job [22399109167](https://github.com/lboue/connectedhomeip/actions/runs/8190963525/job/22399109167#logs)

```
Cmd: ./chef.py -cbr -d rootnode_dimmablelight_bCwGYSDpoe -t esp32
Error: Process completed with exit code 1.
```

From logfile
```
-- Building ESP-IDF components for target esp32
Solving dependencies requirements
..................................Updating lock file at /__w/connectedhomeip/connectedhomeip/examples/chef/esp32/dependencies.lock
Processing 10 dependencies:
[1/10] espressif/cbor (0.6.0~1)
[2/10] espressif/esp_diag_data_store (1.0.1)
[3/10] espressif/esp_diagnostics (1.0.2)
[4/10] espressif/esp_encrypted_img (2.1.0)
[5/10] espressif/esp_insights (1.0.1)
[6/10] espressif/esp_secure_cert_mgr (2.4.1)
[7/10] espressif/led_strip (1.1.0-alpha)
[8/10] espressif/mdns (1.2.5)
[9/10] espressif/rmaker_common (1.4.5)
[10/10] idf (5.1.2)
-- ** Start of Cluster List **
-- ** End of Cluster List **
-- Found Python3: /__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/bin/python3.9 (found version "3.9.5") found components: Interpreter 
CMake Error at /opt/espressif/esp-idf/tools/cmake/component.cmake:250 (message):
  ERROR: 2 problems were found in the manifest file
  /__w/connectedhomeip/connectedhomeip/examples/chef/esp32/managed_components/espressif__mdns/idf_component.yml:

  Invalid manifest format

  Unknown keys: repository_info

  

  SUGGESTION: This component may be using a newer version of the component
  manager.

  You can try to update the component manager by running:

      /__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/bin/python -m pip install --upgrade idf-component-manager

  

Call Stack (most recent call first):
  /opt/espressif/esp-idf/tools/cmake/build.cmake:574 (__component_get_requirements)
  /opt/espressif/esp-idf/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:56 (project)

-- Configuring incomplete, errors occurred!
See also "/__w/connectedhomeip/connectedhomeip/examples/chef/esp32/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /__w/connectedhomeip/connectedhomeip/examples/chef/esp32/build/log/idf_py_stderr_output_3883 and /__w/connectedhomeip/connectedhomeip/examples/chef/esp32/build/log/idf_py_stdout_output_3883
Traceback (most recent call last):
  File "/__w/connectedhomeip/connectedhomeip/examples/chef/./chef.py", line 1011, in <module>
    sys.exit(main())
  File "/__w/connectedhomeip/connectedhomeip/examples/chef/./chef.py", line 704, in main
    shell.run_cmd("idf.py build")
  File "/__w/connectedhomeip/connectedhomeip/examples/chef/stateful_shell.py", line 139, in run_cmd
    raise RuntimeError(
RuntimeError: Error. Nonzero return code.
Returncode: 2
Cmd: idf.py build
Traceback (most recent call last):
  File "/__w/connectedhomeip/connectedhomeip/./examples/chef/chef.py", line 1011, in <module>
    sys.exit(main())
  File "/__w/connectedhomeip/connectedhomeip/./examples/chef/chef.py", line 407, in main
    shell.run_cmd(command)
  File "/__w/connectedhomeip/connectedhomeip/examples/chef/stateful_shell.py", line 139, in run_cmd
    raise RuntimeError(
RuntimeError: Error. Nonzero return code.
Returncode: 1
Cmd: ./chef.py -cbr -d rootnode_dimmablelight_bCwGYSDpoe -t esp32
```
Can you fix the ESP32 CI?

### Platform

esp32

bzbarsky-apple, 2024-03-07:
@lboue Wasn't this fixed in #32474?

lboue, 2024-03-08:
> @lboue Wasn't this fixed in #32474?

No. I tried by hand (commit id b278978aa3fd92e2f6aec39b7f007fdf79ad8a5c) and it still fails.

lboue, 2024-03-08:
I restarted 'Chef - ESP32 CI Examples'  job and it's OK now.
Thanks.

closed 2024-03-08 by lboue