project-chip/connectedhomeip #32515: [1.3 SVE ] test case TC-OPSTATE-2.5 test failed

haijun1656, 2024-03-09:
### Reproduction steps

running according with  TC-OPSTATE-2.5， step-14（operationalstate start 1 1）cannot running success fisrt time after the step-13（Restart DUT and repeat step 5）is done； but if we Restart DUT， we runnning command twice（operationalstate start 1 1）, the first command will be timeout, and the sencond will success;
and we do some other test to verify the case:
1. ./chip-tool interactive start
2. basicinformation read cluster-revision 1 0 --success
3.restart DUT
4. basicinformation read cluster-revision 1 0 ----timeout
5. basicinformation read cluster-revision 1 0--- success

so, at chip interactive  mode, the first command will be timeout,  and the sencond command will be success, after DUT reboot

[TC-OPSTATE-2-5(2).txt](https://github.com/project-chip/connectedhomeip/files/14544704/TC-OPSTATE-2-5.2.txt)
[Basicinformation-OPSTATE-2-5-test.txt](https://github.com/project-chip/connectedhomeip/files/14544708/Basicinformation-OPSTATE-2-5-test.txt)

### Bug prevalence

Problems must arise

### GitHub hash of the SDK that was being used

9f6d627e0262e1d023986291948bb4e845be803e

### Platform

other

### Type

Platform Issue

haijun1656, 2024-03-09:
[OPSTATE-2.5-DUT-log-模组log.txt](https://github.com/project-chip/connectedhomeip/files/14545348/OPSTATE-2.5-DUT-log-.log.txt)
attach a DUT log

haijun1656, 2024-03-09:
[TC-OPSTATE-2.5-CHIP.txt](https://github.com/project-chip/connectedhomeip/files/14545859/TC-OPSTATE-2.5-CHIP.txt)
[TC-OPSTATE-2.5-KVS.txt](https://github.com/project-chip/connectedhomeip/files/14545860/TC-OPSTATE-2.5-KVS.txt)

i use chip-all-clusters-app on raspberrypi to simulate it, and the problem still occurs;
to start  chip-all-clusters-app with comand: ./chip-all-clusters-app --KVS /home/kvs

bzbarsky-apple, 2024-03-11:
This is expected behavior if you are using chip-tool in interactive mode without re-establishing CASE sessions explicictly: when you restart the DUT, the next attempt to use the session will fail (because the DUT no longer has the session), that will make the session as defunct, and the next attempt to send a command will use a new session.

closed 2024-03-11 by bzbarsky-apple