project-chip/connectedhomeip #32533: [Build] Bootstraph.sh fails with pip protobuf error

samryanuofl, 2024-03-11:
### Build issue(s)

1. Trying to build v1.2.0.0 tag (52e4a643)
2. Clone, git submodules and run `source scripts/bootstrap.sh`
3. Get the error: 
```
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/linux:linux_x64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --capture-output --module piptools --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._compile_requirements_metadata_path_list.txt -- compile --resolver=backtracking --allow-unsafe --output-file python/gen/matter_build_venv/compiled_requirements.txt python/gen/matter_build_venv/generated_requirements.txt ../../scripts/setup/requirements.build.txt
  ERROR: Cannot install -r python/gen/matter_build_venv/generated_requirements.txt (line 86), protobuf and protobuf~=3.20.1 because these package versions have conflicting dependencies.
```
4. See attached log for full error
5. Running Ubuntu 22.04.1 LTS
6. Previously I was able to run this build on the same VM without issues
...
[build.log](https://github.com/project-chip/connectedhomeip/files/14560219/build.log)

bzbarsky-apple, 2024-03-11:
@andy31415

sanlingye, 2024-03-12:
have u solved this question?
i meet it too

samryanuofl, 2024-03-12:
Git bisect tells me it was fixed by this commit: https://github.com/project-chip/connectedhomeip/commit/19d03a0c07f645ca906c7677a07b22dd49e9b9fa

So it should be fixed by updating to v1.2.0.1.

I don't understand the bootstrap script well enough to know why the 1.2.0.0 build started failing, I guess one of the online dependencies changed?

@sanlingye

tehampson, 2024-03-12:
Hmm... Interesting that you found 19d03a0 to be the fix.

I am just wondering in your build environment when you hit the issue on v1.2.0.0 tag (https://github.com/project-chip/connectedhomeip/commit/52e4a6435f74b20fb1bf355cc67587c3a3853488), was that a clean build? As in did you remove  `.environment/` before running bootstrap?

tehampson, 2024-03-12:
Also after updating the git submodule, what is the commit id `third_party/pigweed/`

tehampson, 2024-03-13:
@samryanuofl do you mind answer the questions above when you are able to reproduce the issue?

bzbarsky-apple, 2024-04-11:
We could consider moving the tags to a different revision, but that does require creating a new changeset branching off the current tag, right?

andy31415, 2024-04-15:
I can create a branch out of the latest tag if we need some way to keep updating. I am unsure how long we are maintaining old versions - for my own needs, I generally maintain matter and potentially 1 version behind,

andy31415, 2024-04-15:
I created https://github.com/project-chip/connectedhomeip/tree/v1.2-branch from v1.2.0.1

andy31415, 2024-10-21:
Assuming closed for now: we have a fix for these bits checked in various branches.

closed 2024-10-21 by andy31415