project-chip/connectedhomeip #32587: [Workflow]  the QEMU tests are still failing in CI

yufengwangca, 2024-03-15:
### Workflow issues

1.
2.
3.
...
https://github.com/project-chip/connectedhomeip/actions/workflows/qemu.yaml?query=branch%3Amaster

### Platform

esp32

andy31415, 2024-03-20:
Do we have the logs of what was happening here? the link seems to show all actions (which now are green) and I cannot find an example failure with logs.

andy31415, 2024-03-20:
I see some failures with `The hosted runner: GitHub Actions ... lost communication with the server. Anything in your workflow that terminates the runner process, starves it for CPU/Memory, or blocks its network access can cause this error.`

was this the error? Something else?

closed 2024-04-19 by mergify[bot]