project-chip/connectedhomeip #32589: [Build] Bootstraph.sh fails with pip building PyYAML

mringwal, 2024-03-15:
### Build issue(s)

Trying to build master (15866ed085e07e03b0e564cf41f4932b3f71b5b5)

Clone, git submodules and run source scripts/bootstrap.sh

Get the error:

[511/1099] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --capture-output --module piptools --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._compile_requirements_metadata_path_list.txt -- compile --resolver=backtracking --allow-unsafe --output-file python/gen/matter_build_venv/compiled_requirements.txt python/gen/matter_build_venv/generated_requirements.txt ../../scripts/setup/requirements.build.txt
FAILED: python/gen/matter_build_venv/compiled_requirements.txt

Full log: [buildlog-mac.txt](https://github.com/project-chip/connectedhomeip/files/14621196/buildlog-mac.txt)

### Platform

darwin

### Anything else?

Running macOS 14.4 with current Xcode
This is the first time I try to setup matter for macOS, but used esp-matter before.
I've deleted ./environment before

mringwal, 2024-03-16:
Trying on other systems gave some insight - on a similar Mac but without Python 3.12 from Homebrew, bootstrap succeeded. 

Going back to my system, I was able to complete the bootstrap by temporarily disabling Homebrew (skipping `opt/homebrew/bin/brew shellenv`). After bootstrap succeeded, compiling the all-clusters example worked eve with Homebrew enabled again.

Please test if you can bootstrap with Python3 from up-to-date Homebrew. I'm happy to dig out more details about my system if this doesn't happen on other systems.

musachy, 2024-03-26:
Same error here but I cannot get it to work even with Mac's pip/python. Replicated in M2.

noahp, 2024-04-30:
This patch does it, I'll open a PR:

```diff
diff --git a/scripts/setup/constraints.txt b/scripts/setup/constraints.txt
index b0e9aed4ae..2f91fa14e2 100644
--- a/scripts/setup/constraints.txt
+++ b/scripts/setup/constraints.txt
@@ -49,7 +49,7 @@ colorama==0.4.6
     #   west
 coloredlogs==15.0.1
     # via -r requirements.all.txt
-construct==2.10.54
+construct==2.10.70
     # via
     #   -r requirements.esp32.txt
     #   esp-coredump
@@ -163,7 +163,7 @@ portpicker==1.5.2
     # via
     #   -r requirements.all.txt
     #   mobly
-prompt-toolkit==3.0.38
+prompt-toolkit==3.0.43
     # via ipython
 protobuf==4.24.4
     # via
@@ -208,7 +208,7 @@ python-socketio==4.6.1
     # via -r requirements.esp32.txt
 pytz==2022.7.1
     # via pandas
-pyyaml==6.0
+pyyaml==6.0.1
     # via
     #   esptool
     #   idf-component-manager
```

closed 2024-05-02 by mergify[bot]

mringwal, 2024-05-04:
@noahp Thanks!

PR #33242 (2024-05-02): Support building chip-tool with python3.12
This updates 2 python packages to versions compatible with python3.12, fixing the following errors when running
`./scripts/examples/gn_build_example.sh examples/chip-tool BUILDFOLDER`:

1. `pyyaml` 6.0 build error: `AttributeError: cython_sources`, addressed [here](https://github.com/yaml/pyyaml/pull/702)
2. `construct` import error:

   ``` File "/tmp/pip-install-ifgk4tul/construct_6d60304f85e249759f6c184ea89f1317/construct/core.py", line 3, in <module> import struct, io, binascii, itertools, collections, pickle, sys, os, tempfile, hashlib, importlib [... cut]