project-chip/connectedhomeip #32616: [chip-tool] Is there a way to look up the node information (eg, the list of all node-ids) for all paired devices?

whqxbs, 2024-03-18:
### Test issue(s)

In the process of using and developing the Matter protocol, there are many scenarios where you need to query all the devices paired under a particular Controller.

For example, when I use chip-tool, I may have paired a lot of devices, but then for various reasons, for example, I forgot the node-id of a certain device, and I can't control it through the chip-tool command.

Is there an API or a command that allows me to list the node-id of all the devices that have been paired to the chip-tool I'm using?

Looking forward to your reply sincerely! : )

bzbarsky-apple, 2024-03-19:
There is not right now.

bzbarsky-apple, 2024-03-19:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/31504

bzbarsky-apple, 2024-03-19:
That said, if the devices are on the network, you could do an operational browse (using avahi or dns-sd or whatever DNS-SD inspection tool you want) and look for the instance names with the right compressed fabric ID...

closed 2024-03-19 by bzbarsky-apple

whqxbs, 2024-03-19:
Does this mean that if a device that was once paired if now not on the network and I forget the nodeid, I can no longer query it via chiptool and have to resort to some other method of recording the nodeid myself?
And then I have to pair it again, right？

bzbarsky-apple, 2024-03-20:
>  I can no longer query it via chiptool and have to resort to some other method of recording the nodeid myself?

Right now, correct.

>  And then I have to pair it again, right？

Well, if it's (1) still paired but (2) no longer on the network, you can't really pair it again without resetting it or something...  So I am not sure what you are asking.