project-chip/connectedhomeip #32629: Python tests has to be updated by adding the PICS conditions as per the test plan

KishokG, 2024-03-19:
### Feature Area

Other

### Test Case

TC-MESS-1.1 & 2.1

### Reproduction steps

d

### Bug prevalence

Whenever I do this 

### GitHub hash of the SDK that was being used

eb2f8f6ca95c181577415bc4b3e7cc240d5b2edd

### Platform

raspi

KishokG, 2024-03-19:
@jamesharrow  In this PR: https://github.com/project-chip/connectedhomeip/pull/32464 the TC-EEM-2.1 and TC-EPm-2.1 tests only resolved. Other than that, all the scripts which is mentioned above has to be updated with the PICS guard.

- TC_EPM_2_1 already has PICS guard checks. But it seems, the script doesen't checks the PICS. Becasue while we run the script without adding the PICS files all the steps gets PASSING without skipping any steps.

jamesharrow, 2024-03-19:
That’s expected when you use the example energy management app since it supports all of the required features for these tests.

the tests would likely fail if you used a different DUT which didn’t.

jamesharrow, 2024-03-19:
Note the PR explains the rationale for not using PICS everywhere - when discussed with other CSA members the approach of moving away from PICS to asking the device what attributes it supports is felt to be better in the longer term.

So EVSE test cases don’t rely on PICS but check the supported attributes attribute. Ideally we should change all of the test cases to use that approach instead of PICS, but given the urgency of the request I used PICS / pics_guard to check the PICS for resolving 1.3 SVE issues on EPM/EEM.

But I think this needs looking at in a future matter release across all test cases to try to move away from PICS.