project-chip/connectedhomeip #32642: [CERT-TEST-FAILURE] - TC-S-2.2 - Test failed

Ashwinigrl, 2024-03-20:
### Feature Area

Area:Scenes

### Test Case

TC-S-2.2

### Reproduction steps

1. Execute the below mentioned command to put DUT into a commissionable state : .`/chip-all-clusters-app`
2. Once DUT reach the commission able state sent below mentioned command on TH. : `./chip-tool pairing onnetwork 1 20202021`
3. Once after successful pairing provide the below commands:

**TH sends a AddScene command to DUT with SceneID field set to 255(Value grater than Max value mentioned in the constraint)**

`./chip-tool scenesmanagement add-scene 0x0001 255 6000000 "" '[{"clusterID": "0x0300", "attributeValueList":[{"attributeID": "0x4001", "attributeValue": "0x01"}]}]' 1 1`

**Actual Outcome:**

> [1710928429.344989][15787:15789] CHIP:DMG: },
> [1710928429.345007][15787:15789] CHIP:DMG: Received Command Response Data, Endpoint=1 Cluster=0x0000_0062 Command=0x0000_0000
> [1710928429.345017][15787:15789] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0062 Command 0x0000_0000
> [1710928429.345036][15787:15789] CHIP:TOO:   AddSceneResponse: {
> [1710928429.345044][15787:15789] CHIP:TOO:     status: 0
> [1710928429.345047][15787:15789] CHIP:TOO:     groupID: 1
> [1710928429.345049][15787:15789] CHIP:TOO:     sceneID: 255
> [1710928429.345052][15787:15789] CHIP:TOO:    }

**Expected Outcome:**

> [1710928429.344989][15787:15789] CHIP:DMG: },
> [1710928429.345007][15787:15789] CHIP:DMG: Received Command Response Data, Endpoint=1 Cluster=0x0000_0062 Command=0x0000_0000
> [1710928429.345017][15787:15789] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0062 Command 0x0000_0000
> [1710928429.345036][15787:15789] CHIP:TOO:   AddSceneResponse: {
> [1710928429.345044][15787:15789] CHIP:TOO:     status: 133
> [1710928429.345047][15787:15789] CHIP:TOO:     groupID: 1
> [1710928429.345049][15787:15789] CHIP:TOO:     sceneID: 255
> [1710928429.345052][15787:15789] CHIP:TOO:    }

- Same for  ViewScene Command, RemoveScene Command, StoreScene Command, RecallScene Command. CopyScene Command.

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

d2eab7f07ae9289fe14aa7e13dba7e64904ef80e

### Platform

raspi

### Anything else?

**Summary:**

According to the specifications, The SceneID field in the AddScene Command has a constraint of a maximum value 254. However, when attempting to send the AddScene command with a SceneID field value of 255, the response value should be "Invalid Command" instead of "SUCCESS".

**Specification References:**

- https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/Scenes.adoc#92-addscene-command.

**References Log:**
[Scenes.txt](https://github.com/project-chip/connectedhomeip/files/14664708/Scenes.txt)

jinbiaochenstayhungry, 2024-03-26:
hello, 
where did u get the scenesmanagement command work，i had check the code and specification，but still cant find chip-tool that supports this command，which really support is scenes， but still， i found scenesmanagement in some yaml file ，which makes   me confuse.

Survensa, 2024-03-27:
@jinbiaochenstayhungry, could you please verify the commit or branch you are currently in? We utilized https://github.com/project-chip/connectedhomeip/commit/d2eab7f07ae9289fe14aa7e13dba7e64904ef80e to validate this test case

lpbeliveau-silabs, 2024-04-03:
@Ashwinigrl From the Matter Application Clusters section 1.4.7.3.2:

> For non-nullable attributes, any value that is not a valid numeric value for the attribute’s type after accounting for constraints SHALL be considered to have the maximum legal value in the attribute’s constrained range.

So the expected behavior here is success.

We currently apply the truncation when doing a recall scene, so a way to test that this is applied would be to call recall scene and to read the attribute from the cluster and confirm its values was affected.

Ashwinigrl, 2024-05-23:
As per the above comment, closing this Issue.

closed 2024-05-23 by Ashwinigrl