project-chip/connectedhomeip #32653: [BUG] chip-tool unable to comission vendor lightbulb

pmatsibekker, 2024-03-20:
### Reproduction steps

I am using the metrics docker release. I successfully built chip-tool on amd x64 cpu ubuntu linux.
When I run chip-tool pairing code-wifi 1 mypersonalssid mypersonalssidpasswd 139xxxxxxxx, for a working vendor matter enabled lightbulb, I am not able to pair the device. It all works fine from Google Home with my android and nest2, but chip-tool is failing. I had attempted to use pariing ble-wifi but would need setup pin code and discriminator args which I am not sure how to pick up from a vendor matter lightbulb given that I only have the matter code visible.

1710958145.791386][114824:114824] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1710958145.791566][114824:114824] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1710958145.791639][114824:114824] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1710958145.791665][114824:114824] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1710958145.791988][114824:114824] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-xsmifE)
[1710958145.792204][114824:114824] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1710958145.792242][114824:114824] CHIP:DL: NVS set: chip-counters/reboot-count = 5 (0x5)
[1710958145.792798][114824:114824] CHIP:DL: Got Ethernet interface: eth0
[1710958145.792879][114824:114824] CHIP:DL: Found the primary Ethernet interface:eth0
[1710958145.792990][114824:114824] CHIP:DL: Failed to get WiFi interface
[1710958145.793014][114824:114824] CHIP:DL: Failed to reset WiFi statistic counts
[1710958145.793039][114824:114824] CHIP:IN: UDP::Init bind&listen port=0
[1710958145.793079][114824:114824] CHIP:IN: UDP::Init bound to port=59416
[1710958145.793105][114824:114824] CHIP:IN: UDP::Init bind&listen port=0
[1710958145.793122][114824:114824] CHIP:IN: UDP::Init bound to port=41404
[1710958145.793128][114824:114824] CHIP:IN: BLEBase::Init - setting/overriding transport
[1710958145.793152][114824:114824] CHIP:IN: TransportMgr initialized
[1710958145.793163][114824:114824] CHIP:FP: Initializing FabricTable from persistent storage
[1710958145.793219][114824:114824] CHIP:TS: Last Known Good Time: 2024-03-20T15:29:37
[1710958145.793370][114824:114824] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x5A319A2B2BA2D6FC, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1710958145.803497][114824:114824] CHIP:ZCL: Using ZAP configuration...
[1710958145.997547][114824:114824] CHIP:CTL: System State Initialized...
[1710958145.997607][114824:114824] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1710958145.997654][114824:114824] CHIP:CTL: Setting attestation nonce to random value
[1710958145.997691][114824:114824] CHIP:CTL: Setting CSR nonce to random value
[1710958145.997730][114824:114824] CHIP:IN: UDP::Init bind&listen port=5550
[1710958145.997763][114824:114824] CHIP:IN: UDP::Init bound to port=5550
[1710958145.997771][114824:114824] CHIP:IN: UDP::Init bind&listen port=5550
[1710958145.997856][114824:114824] CHIP:IN: UDP::Init bound to port=5550
[1710958145.997880][114824:114824] CHIP:IN: TransportMgr initialized
[1710958145.998054][114824:114832] CHIP:DL: CHIP task running
[1710958145.998198][114824:114832] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1710958145.998233][114824:114832] CHIP:CTL: Setting attestation nonce to random value
[1710958145.998250][114824:114832] CHIP:CTL: Setting CSR nonce to random value
[1710958145.999184][114824:114832] CHIP:CTL: Generating NOC
[1710958145.999484][114824:114832] CHIP:FP: Validating NOC chain
[1710958145.999939][114824:114832] CHIP:FP: NOC chain validation successful
[1710958146.000007][114824:114832] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1710958146.000036][114824:114832] CHIP:TS: Last Known Good Time: 2024-03-20T15:29:37
[1710958146.000059][114824:114832] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1710958146.000063][114824:114832] CHIP:TS: Retaining current Last Known Good Time
[1710958146.000927][114824:114832] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1710958146.001612][114824:114832] CHIP:TS: Committing Last Known Good Time to storage: 2024-03-20T15:29:37
[1710958146.002110][114824:114832] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1710958146.002125][114824:114832] CHIP:IN: UDP::Init bind&listen port=5550
[1710958146.002237][114824:114832] CHIP:IN: UDP::Init bound to port=5550
[1710958146.002266][114824:114832] CHIP:IN: UDP::Init bind&listen port=5550
[1710958146.002282][114824:114832] CHIP:IN: UDP::Init bound to port=5550
[1710958146.002285][114824:114832] CHIP:IN: TransportMgr initialized
[1710958146.004857][114824:114832] CHIP:CTL: Setting wifi credentials from parameters
[1710958146.004897][114824:114832] CHIP:CTL: Setting attestation nonce to random value
[1710958146.004914][114824:114832] CHIP:CTL: Setting CSR nonce to random value
[1710958146.004968][114824:114832] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1710958146.004974][114824:114832] CHIP:CTL: Starting commissioning discovery over BLE
[1710958146.004984][114824:114832] CHIP:CTL: Starting commissioning discovery over DNS-SD
[1710958146.005163][114824:114832] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on docker0:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:409: OS Error 0x02000065: Network is unreachable
[1710958146.005191][114824:114832] CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 1 failures.
[1710958146.005351][114824:114833] CHIP:DL: TRACE: Bluez mainloop starting Thread
[1710958146.005658][114824:114826] CHIP:DL: FAIL: get bus sync in StartupEndpointBindings, error: Could not connect: No such file or directory
[1710958146.005739][114824:114832] CHIP:DL: PlatformBlueZInit init success
[1710958146.005785][114824:114832] CHIP:CTL: Commissioning discovery over BLE failed: ../examples/chip-tool/third_party/connectedhomeip/src/platform/Linux/BLEManagerImpl.cpp:655: CHIP Error 0x00000003: Incorrect state
[1710958146.005791][114824:114832] CHIP:-: ../examples/chip-tool/third_party/connectedhomeip/src/platform/Linux/BLEManagerImpl.cpp:655: CHIP Error 0x00000003: Incorrect state at ../examples/chip-tool/third_party/connectedhomeip/src/controller/SetUpCodePairer.cpp:283
[1710958146.005796][114824:114832] CHIP:BLE: No adapter available for new connection establishment
[1710958147.006216][114824:114832] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on docker0:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:409: OS Error 0x02000065: Network is unreachable
[1710958147.006253][114824:114832] CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 1 failures.
[1710958149.007815][114824:114832] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on docker0:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:409: OS Error 0x02000065: Network is unreachable
[1710958149.007883][114824:114832] CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 1 failures.
[1710958153.012729][114824:114832] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on docker0:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:409: OS Error 0x02000065: Network is unreachable
[1710958153.012839][114824:114832] CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 1 failures.
[1710958161.017067][114824:114832] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on docker0:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:409: OS Error 0x02000065: Network is unreachable
[1710958161.017145][114824:114832] CHIP:DIS: mDNS broadcast had only partial success: 5 successes and 1 failures.

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

not sure.  came from us-docker.pkg.dev/nest-matter/docker-repo/virtual-device-image:latest

### Platform

windows

pmatsibekker, 2024-03-20:
CHIP:CTL: Discovery timed out
[1710960171.821269][133152:133156] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1710960171.821276][133152:133156] CHIP:TOO: Secure Pairing Failed
[1710960171.821547][133152:133152] CHIP:CTL: Shutting down the commissioner
[1710960171.821578][133152:133152] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1710960171.821605][133152:133152] CHIP:CTL: Shutting down the controller
[1710960171.821610][133152:133152] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1710960171.821613][133152:133152] CHIP:FP: Forgetting fabric 0x1
[1710960171.821620][133152:133152] CHIP:TS: Pending Last Known Good Time: 2024-03-20T15:29:37
[1710960171.821661][133152:133152] CHIP:TS: Previous Last Known Good Time: 2024-03-20T15:29:37
[1710960171.821665][133152:133152] CHIP:TS: Reverted Last Known Good Time to previous value
[1710960171.821676][133152:133152] CHIP:CTL: Shutting down the commissioner
[1710960171.821698][133152:133152] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1710960171.821711][133152:133152] CHIP:CTL: Shutting down the controller
[1710960171.821714][133152:133152] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1710960171.821815][133152:133152] CHIP:DMG: IM WH moving to [Uninitialized]
[1710960171.821838][133152:133152] CHIP:DMG: IM WH moving to [Uninitialized]
[1710960171.821842][133152:133152] CHIP:DMG: IM WH moving to [Uninitialized]
[1710960171.821843][133152:133152] CHIP:DMG: IM WH moving to [Uninitialized]
[1710960171.821864][133152:133152] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1710960171.821879][133152:133152] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1710960171.821989][133152:133152] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-yUDH6u)
[1710960171.822140][133152:133152] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1710960171.822170][133152:133152] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1710960171.822175][133152:133152] CHIP:DL: Inet Layer shutdown
[1710960171.822177][133152:133152] CHIP:DL: BLE shutdown
[1710960171.822179][133152:133152] CHIP:DL: System Layer shutdown
[1710960171.822193][133152:133152] CHIP:TOO: Run command failure: ../examples/chip-tool/commands/pairing/PairingCommand.cpp:164: CHIP Error 0x00000003: Incorrect state

bzbarsky-apple, 2024-03-21:
It's not clear to me from the above logs whether chip-tool is being run inside a docker container here, by the way, but if it is that could definitely be an issue; you would need to look into how to configure things so the docker container has BLE access...

closed 2024-03-21 by bzbarsky-apple