project-chip/connectedhomeip #32663: [CERT-TEST-FAILURE] -TC-CNET- 4.9 - Test failed

KishokG, 2024-03-21:
### Feature Area

Other

### Test Case

TC-CNET-4.9

### Reproduction steps

1. Built all-clusters-app and chiptool in two seperate RPI
2. Brought up DUT using commnad : `./chip-all-clusters-app --wifi`
3. Commission Controller using `./chip-tool pairing ble-wifi 1 SSID SSID-PWD  20202021 3840`
4. Armed the fail-safe :` ./chip-tool generalcommissioning arm-fail-safe 900 1 1 0`
5. Checked the 1st connected ACCESSPOINT : `./chip-tool networkcommissioning read networks 1 0`

> [1657733068.835240][7475:7480] CHIP:TOO:   Networks: 1 entries
> [1657733068.835354][7475:7480] CHIP:TOO:     [1]: {
> [1657733068.835434][7475:7480] CHIP:TOO:       NetworkID: 47524C50726976617465
> [1657733068.835487][7475:7480] CHIP:TOO:       Connected: TRUE
> [1657733068.835535][7475:7480] CHIP:TOO:      }

6. Removed the network : `./chip-tool networkcommissioning remove-network hex:47524C50726976617465 1  0 --Breadcrumb 1`
7. Checked whether the network is reverted to 1st ACCESS POINT : `./chip-tool networkcommissioning read networks 1 0`

**Expected Outcome:**

> [1684497703.596592][3941:3943] CHIP:TOO:   Networks: 1 entries
> [1684497703.596640][3941:3943] CHIP:TOO:     [1]: {
> [1684497703.596668][3941:3943] CHIP:TOO:       NetworkID: 47524C50726976617465
> [1684497703.596697][3941:3943] CHIP:TOO:       Connected: TRUE
> [1684497703.596721][3941:3943] CHIP:TOO:      }

**Actual Outcome:**

> [1711019121.312656][7411:7413] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0001 DataVersion: 3881988887
> [1711019121.312745][7411:7413] CHIP:TOO:   Networks: 0 entries
> [1711019121.313000][7411:7413] CHIP:EM: <<< [E:54187i S:41064 M:121351343 (Ack:118096550)] (S) Msg TX to 1:0000000000000001 [D8F4] [UDP:[fe80::e65f:1ff:fe0f:2878%eth0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)

### Bug prevalence

Whenever I do this 

### GitHub hash of the SDK that was being used

d2eab7f07ae9289fe14aa7e13dba7e64904ef80e

### Platform

raspi

### Anything else?

TH should be able to connect DUT using 1st ACCESSPOINT with Network Id and connected status as TRUE when the fail-safe is disarmed.

**DUT Logs:** [TC-CNET-4.9(DUT).txt](https://github.com/project-chip/connectedhomeip/files/14694305/TC-CNET-4.9.DUT.txt)
**Chip-tool Logs:** [TC-CNET-4.9.txt](https://github.com/project-chip/connectedhomeip/files/14694309/TC-CNET-4.9.txt)

cecille, 2024-04-12:
Does linux have a hook up to swap the wifi network from within the example app?

kagami555, 2024-04-22:
I would like to know if the problem has been solved? The same problem also occurred with the chip-lighting-app I compiled on the linux platform.

KishokG, 2024-04-29:
> Does linux have a hook up to swap the wifi network from within the example app?
> 
> ie, is this a NEW failure? Has this ever worked on the pi?

@cecille It's a new failure. It was worked fine in RPI platform

kagami555, 2024-04-29:
Hello, Kishok G do you want to ask, the steps of connecting to wifi in the matter program on Linux are done by other processes. In fact, this is indeed the case in my project. First of all, due to business needs, we need wifi. The management is placed in a process. Secondly, when I use this commit to compile, the action of connecting to wifi will fail. I think that the matter process will not actually be allowed to operate wifi, so I will skip it. Does this have any impact?

***@***.***
 
From: Kishok G
Date: 2024-04-29 14:01
To: project-chip/connectedhomeip
CC: kagami555; Comment
Subject: Re: [project-chip/connectedhomeip] [CERT-TEST-FAILURE] -TC-CNET- 4.9 - Test failed (Issue #32663)
Does linux have a hook up to swap the wifi network from within the example app?
ie, is this a NEW failure? Has this ever worked on the pi?
@cecille It's a new failure. It was worked fine in RPI platform

kagami555, 2024-04-29:
This has always been a problem, I never got it working correctly on my pi because of the commit I always used (437c5743ef4eac87ec5f80114d295f71e8a4f0e0)

***@***.***
 
From: Kishok G
Date: 2024-04-29 14:01
To: project-chip/connectedhomeip
CC: kagami555; Comment
Subject: Re: [project-chip/connectedhomeip] [CERT-TEST-FAILURE] -TC-CNET- 4.9 - Test failed (Issue #32663)
Does linux have a hook up to swap the wifi network from within the example app?
ie, is this a NEW failure? Has this ever worked on the pi?
@cecille It's a new failure. It was worked fine in RPI platform

cecille, 2024-05-01:
If your device is connected to the operational network BEFORE matter commissioning starts (ie, if the user has a way to set the network credentials outside of matter and this is required for proper functioning of the device) then you should not be including a network commissioning cluster and you should not be using BLE. See spec section 4.3.1 - Commissionable Node Discovery "A commissionable Node that is already connected to an IP-bearing network SHALL only make itself discoverable on the IP network and SHALL use the relevant DNS-SD service (_matterc._udp) described below." 

However, I'm not certain how you'd accomplish this in a reasonable way on a production device unless you had a custom commissioning step. ie, fresh out of the box, how would a user give your device wifi credentials? If it's through the matter commissioning process then you need BLE and network commissioning and the device cannot be connected to a wifi network before commissioning starts. If the device needs wifi credentials before matter commissioning, then it's up to whatever mechanism your product designed to get those allocated, and the discussion of the network commissioning cluster and its associated tests is moot.

Note also that 1) the PI os is not vanilla linux and the linux platform may not work for managing the network stack there and 2) I'm not sure we HAVE any production devices running on a pi base right now so you could be breaking new ground here.

cecille, 2024-05-09:
@KishokG - neither of the supplied logs show the final arm failsafe command or the network read. Can you please provide full logs.

andy31415, 2024-05-09:
> I would like to know if the problem has been solved? The same problem also occurred with the chip-lighting-app I compiled on the linux platform.

@kagami555 the examples included with the matter SDK are included as a starting point and built up sufficiently to help us CI and validation, however this is where typical maintenance stops. 

Building an actual product goes beyond that. If you need specific fixes, the fastest path is likely for you to investigate the root cause and propose a fix PR.

KishokG, 2024-05-10:
> @KishokG - neither of the supplied logs show the final arm failsafe command or the network read. Can you please provide full logs.

@cecille Here is the full log of TC-CNET-4.9.

**DUT Logs:** [TC-CNET-4.9(DUT) log.txt](https://github.com/project-chip/connectedhomeip/files/15270673/TC-CNET-4.9.DUT.log.txt)
**Chip-tool Logs:** [TC-CNET-4.9(Chip-tool) log.txt](https://github.com/project-chip/connectedhomeip/files/15270675/TC-CNET-4.9.Chip-tool.log.txt)

kagami555, 2024-05-10:
This is the completed log of my recent test, thank you very much
------------------------------------------------------------------

cecille, 2024-05-10:
@KishokG - can you also confirm whether this test works as expected for you on a normal wifi device? ex. esp32?

cecille, 2024-05-10:
Can you also confirm whether the PI is joined to the wifi network before you issue the pairing command?

KishokG, 2024-05-13:
> Can you also confirm whether the PI is joined to the wifi network before you issue the pairing command?

@cecille Yes. Both(Controller and DUT) RPI's are connected to the wifi network before commissioning.

KishokG, 2024-05-13:
> @KishokG - can you also confirm whether this test works as expected for you on a normal wifi device? ex. esp32?

I ran `TC-CNET-4.10` tc in Thread platform(NRF52840-DK) and the test is passing without any issues. 

**Note:** TC-CNET-4.9 and 4.10 both test cases has same functionality.

**Here is the TC-CNET-4.10 Log:** [TC-CNET-4.10(Chip-tool) log.txt](https://github.com/project-chip/connectedhomeip/files/15293586/TC-CNET-4.10.Chip-tool.log.txt)

cecille, 2024-05-13:
Does the PI HAVE an attached thread module? And is THAT connected prior to commissioning?

cecille, 2024-05-13:
4.10 is different here because you won't have the problem of having a pre-connected PAN before commissioning.

KishokG, 2024-05-30:
@cecille 
> @KishokG - can you also confirm whether this test works as expected for you on a normal wifi device? ex. esp32?

- Tested TC-CNET-4.9 in ESP32 platform and it's PASSING.

**Here is the logs:**

- [TC-CNET-4-9-DUT.log](https://github.com/project-chip/connectedhomeip/files/15499219/TC-CNET-4-9-DUT.log)
- [TC-CNET-4-9-chip-tool.log](https://github.com/project-chip/connectedhomeip/files/15499223/TC-CNET-4-9-chip-tool.log)

KishokG, 2024-06-19:
Now the test is PASSING in this commit: `51e4846460a6760800fc53206dadda26a255f932` (Tested in RPI platform).

**Here is the Logs:**

- [TC-CNET-4.9(Chip-tool) log_19_06_2024.txt](https://github.com/user-attachments/files/15899527/TC-CNET-4.9.Chip-tool.log_19_06_2024.txt)
- [TC-CNET-4.9(DUT) log_19_06_2024.txt](https://github.com/user-attachments/files/15899529/TC-CNET-4.9.DUT.log_19_06_2024.txt)

KishokG, 2024-08-21:
Tested in RPI platform and Again the test is FAILING in this commit: `d0d91272068f267cf880f9d56787ca28da885673`

**Here is the log:**  [TC-CNET_4.9_NonTE2.txt](https://github.com/user-attachments/files/16689412/TC-CNET_4.9_NonTE2.txt)

KishokG, 2025-05-13:
Now the test is PASSING in this commit: `a12250a4e4ffce5be5f809479fbe8d85e5acd230` (Tested in RPI platform).

**Here is the Logs:** 

[CNET-4.9.txt](https://github.com/user-attachments/files/20181429/CNET-4.9.txt)

closed 2025-05-13 by KishokG