project-chip/connectedhomeip #32669: [1.3] Update Media Clusters according to reported issues

lazarkov, 2024-03-21:
### Reproduction steps

Please update media clusters according to the report:

`Account Login Cluster:
LoggedOut event is marked as mandatory, but is actually optional

Application Basic Cluster:
Name attribute is char_string, but should be long_char_string
AllowedVendorList attribute has cap of 254 elements, but the spec has no cap

Audio Output Cluster:
Output List attribute has cap of 254 elements, but the spec has no cap

Channel Cluster:
Channel List attribute has cap of 254 elements, but the spec has no cap
RecordingFlagBitmap is a bitmap32, but it is a bitmap8 in the spec

GetProgramGuide Command:
StartTime field is marked as mandatory, but is optional in the spec
EndTime field is marked as mandatory, but is optional in the spec
ChannelList field has no cap, but has a cap of 255 elements in the spec
PageToken field is not nullable, but is nullable in the spec
ExternalIDList field has no cap, but has a cap of 255 elements in the spec
Data field has no cap, but has a cap of 8092 bytes in the spec

RecordProgram Command:
ProgramIdentifier field has no cap, but has a cap of 255 characters in the spec
ExternalIDList field has no cap, but has a cap of 255 elements in the spec
Data field has no cap, but has a cap of 8092 bytes in the spec

CancelRecordProgram Command:
ProgramIdentifier field has no cap, but has a cap of 255 characters in the spec
ExternalIDList field has no cap, but has a cap of 255 elements in the spec
Data field has no cap, but has a cap of 8092 bytes in the spec

ProgramStruct Struct:
Description field is char_string, but should be long_char_string
ThumbnailUrl field is char_string, but should be long_char_string
PosterArtUrl field is char_string, but should be long_char_string
DvbiUrl field is char_string, but should be long_char_string
ExternalIDList field is the wrong type(!): ProgramCastStruct but is AdditionalInfoStruct in the spec (working probably because they’re similar)

SeriesInfoStruct Struct:
Season field is char_string, but should be long_char_string
Episode field is char_string, but should be long_char_string

ProgramCategoryStruct Struct:
Category field is char_string, but should be long_char_string
SubCategory field is char_string, but should be long_char_string

ProgramCastStruct Struct:
Name field is char_string, but should be long_char_string
Role field is char_string, but should be long_char_string

PageTokenStruct Struct:
After field is char_string, but should be long_char_string
Before field is char_string, but should be long_char_string

Content App Observer Cluster:

ContentAppMessage Command:
Data field is char_string, but should be long_char_string
Data field is marked as optional, but is mandatory in the spec
Data field has no cap, but has a cap of 500 characters in the spec
EncodingHint field is marked as mandatory, but is optional in the spec
EncodingHint field has no cap, but has a cap of 100 characters in the spec

ContentAppMessageResponse Command:
Data field is char_string, but should be long_char_string
Data field has no cap, but has a cap of 500 characters in the spec
EncodingHint field has no cap, but has a cap of 100 characters in the spec

Content Control Cluster:
BlockUnrated attribute has the same define (“ENABLED”) as the Enabled attribute
BlockUnrated attribute is marked as mandatory, but is only mandatory with the BU feature in the spec
BlockChannelList attribute is missing      
BlockApplicationList attribute is missing  
BlockContentTimeWindow attribute is missing
AddBlockChannels command is missing            
RemoveBlockChannels command is missing        
AddBlockApplications command is missing        
RemoveBlockApplications command is missing    
SetBlockContentTimeWindow command is missing  
RemoveBlockContentTimeWindow command is missing
EnteringBlockContentTimeWindow event is missing
AppInfoStruct struct is missing
BlockChannelStruct struct is missing
TimePeriodStruct struct is missing
TimeWindowStruct struct is missing
DayOfWeekBitmap bitmap is missing (also specified as an enum8 in the spec)
BlockChannels feature is missing
BlockApplications feature is missing
BlockContentTimeWindow feature is missing
 
UpdatePIN Command:
Command is marked as not requiring timed invoke, but does require timed invoke in the spec
OldPIN field is marked as optional, but is mandatory in the spec

ResetPIN Command:
Command is marked as not requiring timed invoke, but does require timed invoke in the spec

Enable Command:
Command is marked as not requiring timed invoke, but does require timed invoke in the spec

Disable Command:
Command is marked as not requiring timed invoke, but does require timed invoke in the spec

AddBonusTime Command:
BonusTime field is marked as optional, but is mandatory in the spec

SetScreenDailyTime Command:
ScreenTime field has no cap, but has a cap of 86400  in the spec

RatingNameStruct Struct:
RatingName field has no cap, but has a cap of 8 characters in the spec
RatingNameDesc field has no cap, but has a cap of 64 characters in the spec

Content Launcher Cluster:
AcceptHeader attribute has a max of 254 elements, but a max of 100 elements in the spec

LaunchURL Command:
PlaybackPreferences field is missing

AdditionalInfoStruct Struct:
Name field is char_string, but should be long_char_string
Value field is char_string, but should be long_char_string

StyleInformationStruct Struct:
ImageURL field is char_string, but should be long_char_string
Color field is missing minLength attribute

BrandingInformationStruct Struct:
ProviderName field is char_string, but should be long_char_string

ParameterStruct Struct:
Value field is char_string, but should be long_char_string

PlaybackPreferencesStruct Struct:
PlaybackPosition field is marked as mandatory, but is only mandatory with the AS feature in the spec
PlaybackPosition field is not nullable, but is nullable in the spec
TextTrack field is marked as mandatory, but is only mandatory with the TT feature in the spec
TextTrack field is not nullable, but is nullable in the spec
AudioTracks field is marked as mandatory, but is only mandatory with the AT feature in the spec
AudioTracks field is not nullable, but is nullable in the spec

TrackPreferenceStruct Struct:
Characteristics field is not nullable, but is nullable in the spec
Characteristics field is marked as mandatory, but is optional in the spec
AudioOutputIndex field is marked as mandatory, but is only mandatory with the AT feature in the spec
AudioOutputIndex field is not nullable, but is nullable in the spec

Media Input Cluster:
InputList attribute has cap of 254 elements, but the spec has no cap

Media Playback Cluster:

ActivateAudioTrack Command:
AudioOutputIndex field is marked as mandatory, but is only mandatory with the AT feature in the spec
AudioOutputIndex field is not nullable, but is nullable in the spec

ActivateTextTrack Command:
TrackID field has no cap, but has a cap of 32 characters in the spec

StateChanged Event:
StartTime field is marked as mandatory, but is only mandatory with the AS feature in the spec
Duration field is marked as mandatory, but is only mandatory with the AS feature in the spec
SampledPosition field is marked as mandatory, but is only mandatory with the AS feature in the spec
PlaybackSpeed field is marked as mandatory, but is only mandatory with the AS feature in the spec
SeekRangeEnd field is marked as mandatory, but is only mandatory with the AS feature in the spec
SeekRangeStart field is marked as mandatory, but is only mandatory with the AS feature in the spec
AudioAdvanceUnmuted field is marked as mandatory, but is only mandatory with the AS feature in the spec
AudioAdvanceUnmuted field has a default of “true”, but the spec has a default of “false”

TrackStruct Struct:
TrackAttributes field is marked as optional, but is mandatory in the spec

TrackAttributesStruct Struct:
Characteristics field is missing
DisplayName field is char_string, but should be long_char_string
DisplayName field has no cap, but has a cap of 256 characters in the spec

Target Navigator Cluster:
TargetList attribute has cap of 254 elements, but the spec has no cap
TargetUpdated event is marked as mandatory, but is optional in the spec

TargetUpdated Event:
TargetList field is marked as mandatory, but is optional in the spec
CurrentTarget field is marked as mandatory, but is optional in the spec
Data field is marked as mandatory, but is optional in the spec`

### Bug prevalence

whenever report tool is started

### GitHub hash of the SDK that was being used

6b82061fa32f11f2fc4880c31793a5e027412bf2

### Platform

darwin

### Type

Spec Compliance Issue

lazarkov, 2024-03-22:
Repro the issue by executing the command:
`alchemy compare --sdkRoot=<path to connectedhomeip> --specRoot=<path to connectedhomeip-spec> --text`

lazarkov, 2024-03-22:
_Note: in order to get diffs for Control Cluster, you'll need to apply the fixes here: https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/9065 Otherwise it fails trying to figure out what a bitmap of type enum8 means._

chrisdecenzo, 2024-05-15:
please also address: https://github.com/project-chip/connectedhomeip/issues/33444