project-chip/connectedhomeip #32686: [Android] - Devices with android 12 and below have problem with mDNS discovery of commissionable devices

igor90ns, 2024-03-22:
### Reproduction steps

General idea is that in our application we want to scan all devices that have commissioning window open and present them to user to choose which one to configure. This works properly with android 13 and 14, but 12 and lower will find just one device.

1. Add two or more devices to apple home pod or google nest hub
2. Put both devices to pairing mode (open commissioning window)
3. Open Chip tool android app on devices with android 12 or lower
4. Go to PROVISION CHIP DEVICE WITH WIFI
5. Open Input device address fragment
6. Click Discover button
7. Keep clicking Discover button

Array of found devices is filled with same device, second device will be found only when first device close its commissioning mode

When I do exact same steps on android 13 and 14, discovery will add same device 2 or 3 times but it will find second device while both device are in commissioning mode

Issue is tested and mdns discovery **is not working properly on** 

- Huawei P smart 2019 (android 12)
- Samsung Tab S3 (android 9)
- Google Nexus (android 8.1)
- I know that those are old device, but we have to provide same experience to our users  

Devices that mdns discovery **is working properly** 

- Xiaomi Tab 5 (android 13)
- Pixel 7 (android 14)
- Samsung A52 (android 14)

Bellow are logs from chiptool android app from Huawei P smart. As attachment  I have added logs from xiaomi tab where everything works properly for comparation purpose.

Command used from ChipDeviceController is _**public void discoverCommissionableNodes()**_ 

```
linker                  com.google.chip.chiptool             W  Warning: "/data/app/com.google.chip.chiptool-tsReVNv1wGGk8EAAdmPEnQ==/base.apk!/lib/arm64-v8a/libc++_shared.so" unused DT entry: unknown processor-specific (type 0x70000001 arg 0x0) (ignoring)
DMG                     com.google.chip.chiptool             D  AccessControl: initializing
CTL                     com.google.chip.chiptool             D  JNI_OnLoad() called
CTL                     com.google.chip.chiptool             D  Loading Java class references.
CTL                     com.google.chip.chiptool             D  Java class references loaded.
DL                      com.google.chip.chiptool             D  AndroidChipPlatform JNI_OnLoad() called
SPT                     com.google.chip.chiptool             E  JavaVM is already set
DL                      com.google.chip.chiptool             D  Loading Java class references.
DL                      com.google.chip.chiptool             D  Java class references loaded.
DL                      com.google.chip.chiptool             D  BleConnectCallbackJNI_OnLoad
DL                      com.google.chip.chiptool             D  Initializing tracing for android
CTL                     com.google.chip.chiptool             D  newDeviceController() called
CTL                     com.google.chip.chiptool             D  Setting attestation nonce to random value
CTL                     com.google.chip.chiptool             D  Setting CSR nonce to random value
CTL                     com.google.chip.chiptool             D  Setting attestation nonce to random value
CTL                     com.google.chip.chiptool             D  Setting CSR nonce to random value
CTL                     com.google.chip.chiptool             D  Setting attestation nonce from parameters
CTL                     com.google.chip.chiptool             D  Setting CSR nonce from parameters
TOO                     com.google.chip.chiptool             D  KVS: Getting key AndroidDeviceControllerKey
CTL                     com.google.chip.chiptool             D  No existing credentials provided: generating ephemeral local NOC chain with OperationalCredentialsIssuer
TOO                     com.google.chip.chiptool             D  KVS: Getting key AndroidCARootCert1
CTL                     com.google.chip.chiptool             D  Generating NOC
IN                      com.google.chip.chiptool             D  UDP::Init bind&listen port=0
IN                      com.google.chip.chiptool             D  UDP::Init bound to port=51573
IN                      com.google.chip.chiptool             D  UDP::Init bind&listen port=0
IN                      com.google.chip.chiptool             D  UDP::Init bound to port=58345
IN                      com.google.chip.chiptool             D  BLEBase::Init - setting/overriding transport
IN                      com.google.chip.chiptool             D  TransportMgr initialized
FP                      com.google.chip.chiptool             D  Initializing FabricTable from persistent storage
TOO                     com.google.chip.chiptool             D  KVS: Getting key g/lkgt
TS                      com.google.chip.chiptool             D  Last Known Good Time: 2023-10-14T01:16:48
TOO                     com.google.chip.chiptool             D  KVS: Getting key g/fidx
TOO                     com.google.chip.chiptool             D  KVS: Getting key f/1/n
TOO                     com.google.chip.chiptool             D  KVS: Getting key f/1/r
TOO                     com.google.chip.chiptool             D  KVS: Getting key f/1/m
FP                      com.google.chip.chiptool             D  Fabric index 0x1 was retrieved from storage. Compressed FabricId 0xAF3109A1C7EEDFAD, FabricId 
[... 503 lines cut ...]
HwAppInnerBoostImpl     com.google.chip.chiptool             D  asyncReportData com.google.chip.chiptool,2,2,1,11 interval=288
OpenGLRenderer          com.google.chip.chiptool             D    HWUI Binary is  enabled
HwAppInnerBoostImpl     com.google.chip.chiptool             D  asyncReportData com.google.chip.chiptool,2,1,2,0 interval=400
ZrHung.AppEyeUiProbe    com.google.chip.chiptool             D  stop checker.
ViewRootIm...lActivity] com.google.chip.chiptool             D  surface should not be released
ViewRootIm...w:88eaddd] com.google.chip.chiptool             D  surface should not be released
libEGL                  com.google.chip.chiptool             W  EGLNativeWindowType 0x73a842b010 disconnect failed
ZrHung.AppEyeUiProbe    com.google.chip.chiptool             D  Current Activity:false
ZrHung.AppEyeUiProbe    com.google.chip.chiptool             D  not watching, wait.
<no-tag>                com.google.chip.chiptool             E  [ZeroHung]zrhung_get_config: Get config failed for wp[0x0008]
```
[huawei p smart.txt](https://github.com/project-chip/connectedhomeip/files/14721269/huawei.p.smart.txt)
[xiaomi.txt](https://github.com/project-chip/connectedhomeip/files/14721270/xiaomi.txt)

### Platform

android

### Platform Version(s)

v1.2.0.1 

### Type

Manually tested with SDK

yunhanw-google, 2024-03-27:
@igor90ns could you try this https://github.com/project-chip/connectedhomeip/pull/32459, and check whether it can fix your issue. thanks

yunhanw-google, 2024-05-01:
Close it now. Feel free to reopen if it is still problematic. thanks

PR #32807 (2024-07-10): [Android] Modify chiptool crash when skipCommissioningComplete flag enabled
Fix #32686 

When in CleanUp Stage, if CommissioneeDevice exists, it must be freed. However, this process must be final. This issue causes a crash because CommissioningStageComplete is called after Release and mDeviceBeingCommissioned is null.

Crash Point : 
https://github.com/project-chip/connectedhomeip/blob/824ba6dbd9a7e783353f4c714e7ca0ba03615c64/src/controller/CHIPDeviceController.cpp#L1903
closed 2024-05-01 by yunhanw-google