project-chip/connectedhomeip #32706: [BUG]  Unable to commission into new fabric using chip-tool api

pmatsibekker, 2024-03-25:
### Reproduction steps

I have commissioned a new matter device to my google android/nest device. Everything works fine. Now I am trying to commission the vendor device to a new fabric controlled by chip-tool. Its on the same wifi/ethernet network.  In my android Home view, I selected link services to link a matter-enabled app with pairing code (good for 2 minutes).
I tried using the chip-tool api to connect
./chip-tool pairing code 1 21104754926

[1711393350.571795][1625:1625] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_tool_kvs
[1711393350.572810][1625:1625] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1711393350.572845][1625:1625] CHIP:DL: ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_kvs
[1711393350.573004][1625:1625] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1711393350.573149][1625:1625] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1711393350.573221][1625:1625] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1711393350.574511][1625:1625] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-9sPmn6)
[1711393350.574652][1625:1625] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1711393350.574700][1625:1625] CHIP:DL: NVS set: chip-counters/reboot-count = 14 (0xE)
[1711393350.574833][1625:1625] CHIP:DL: Got Ethernet interface: eth0
[1711393350.574943][1625:1625] CHIP:DL: Found the primary Ethernet interface:eth0
[1711393350.575016][1625:1625] CHIP:DL: Failed to get WiFi interface
[1711393350.575039][1625:1625] CHIP:DL: Failed to reset WiFi statistic counts
[1711393350.575056][1625:1625] CHIP:IN: UDP::Init bind&listen port=0
[1711393350.575089][1625:1625] CHIP:IN: UDP::Init bound to port=36554
[1711393350.575092][1625:1625] CHIP:IN: UDP::Init bind&listen port=0
[1711393350.575119][1625:1625] CHIP:IN: UDP::Init bound to port=45496
[1711393350.575122][1625:1625] CHIP:IN: BLEBase::Init - setting/overriding transport
[1711393350.575124][1625:1625] CHIP:IN: TransportMgr initialized
[1711393350.575133][1625:1625] CHIP:FP: Initializing FabricTable from persistent storage
[1711393350.575184][1625:1625] CHIP:TS: Last Known Good Time: 2023-10-14T01:16:48
[1711393350.575527][1625:1625] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x24EA3F3A4B34EE0E, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1711393350.575752][1625:1625] CHIP:FP: Fabric index 0x2 was retrieved from storage. Compressed FabricId 0x9F4224575B081A6B, FabricId 0x0000000000000002, NodeId 0x000000000001B669, VendorId 0xFFF1
[1711393350.693186][1625:1625] CHIP:ZCL: Using ZAP configuration...
[1711393350.693790][1625:1625] CHIP:CTL: System State Initialized...
[1711393350.693902][1625:1625] CHIP:CTL: Setting attestation nonce to random value
[1711393350.693951][1625:1625] CHIP:CTL: Setting CSR nonce to random value
[1711393350.694001][1625:1625] CHIP:IN: UDP::Init bind&listen port=5550
[1711393350.694030][1625:1625] CHIP:IN: UDP::Init bound to port=5550
[1711393350.694035][1625:1625] CHIP:IN: UDP::Init bind&listen port=5550
[1711393350.694045][1625:1625] CHIP:IN: UDP::Init bound to port=5550
[1711393350.694048][1625:1625] CHIP:IN: TransportMgr initialized
[1711393350.694277][1625:1627] CHIP:DL: CHIP task running
[1711393350.694406][1625:1627] CHIP:DL: HandlePlatformSpecificBLEEvent 32786
[1711393350.694576][1625:1627] CHIP:CTL: Setting attestation nonce to random value
[1711393350.694617][1625:1627] CHIP:CTL: Setting CSR nonce to random value
[1711393350.695525][1625:1627] CHIP:CTL: Generating NOC
[1711393350.695996][1625:1627] CHIP:FP: Validating NOC chain
[1711393350.696465][1625:1627] CHIP:FP: NOC chain validation successful
[1711393350.696534][1625:1627] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1711393350.696565][1625:1627] CHIP:TS: Last Known Good Time: 2023-10-14T01:16:48
[1711393350.696589][1625:1627] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1711393350.696609][1625:1627] CHIP:TS: Retaining current Last Known Good Time
[1711393350.750938][1625:1627] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1711393350.751966][1625:1627] CHIP:TS: Committing Last Known Good Time to storage: 2023-10-14T01:16:48
[1711393350.752701][1625:1627] CHIP:CTL: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 24EA3F3A4B34EE0E)
[1711393350.752739][1625:1627] CHIP:IN: UDP::Init bind&listen port=5550
[1711393350.752758][1625:1627] CHIP:IN: UDP::Init bound to port=5550
[1711393350.752761][1625:1627] CHIP:IN: UDP::Init bind&listen port=5550
[1711393350.752823][1625:1627] CHIP:IN: UDP::Init bound to port=5550
[1711393350.752830][1625:1627] CHIP:IN: TransportMgr initialized
[1711393350.758585][1625:1627] CHIP:CTL: Setting attestation nonce to random value
[1711393350.758653][1625:1627] CHIP:CTL: Setting CSR nonce to random value
[1711393350.758660][1625:1627] CHIP:CTL: Checking ICD registration parameters
[1711393350.758691][1625:1627] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1711393350.758727][1625:1627] CHIP:CTL: Starting commissioning discovery over DNS-SD
[1711393380.775219][1625:1627] CHIP:CTL: Discovery timed out
[1711393380.775313][1625:1627] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1711393380.775332][1625:1627] CHIP:TOO: Secure Pairing Failed
[1711393380.775842][1625:1625] CHIP:CTL: Shutting down the commissioner
[1711393380.775935][1625:1625] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1711393380.776006][1625:1625] CHIP:CTL: Shutting down the controller
[1711393380.776112][1625:1625] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1711393380.776140][1625:1625] CHIP:FP: Forgetting fabric 0x1
[1711393380.776225][1625:1625] CHIP:TS: Pending Last Known Good Time: 2023-10-14T01:16:48
[1711393380.776420][1625:1625] CHIP:TS: Previous Last Known Good Time: 2023-10-14T01:16:48
[1711393380.776480][1625:1625] CHIP:TS: Reverted Last Known Good Time to previous value
[1711393380.776513][1625:1625] CHIP:CTL: Shutting down the commissioner
[1711393380.776595][1625:1625] CHIP:CTL: Shutting down the controller
[1711393380.776607][1625:1625] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1711393380.776688][1625:1625] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1711393380.776783][1625:1625] CHIP:FP: Shutting down FabricTable
[1711393380.776799][1625:1625] CHIP:TS: Pending Last Known Good Time: 2023-10-14T01:16:48
[1711393380.776885][1625:1625] CHIP:TS: Previous Last Known Good Time: 2023-10-14T01:16:48
[1711393380.776942][1625:1625] CHIP:TS: Reverted Last Known Good Time to previous value
[1711393380.777294][1625:1625] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-B0lXO4)
[1711393380.777771][1625:1625] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1711393380.777851][1625:1625] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1711393380.777864][1625:1625] CHIP:DL: Inet Layer shutdown
[1711393380.777869][1625:1625] CHIP:DL: BLE shutdown
[1711393380.777875][1625:1625] CHIP:DL: System Layer shutdown
[1711393380.778259][1625:1625] CHIP:TOO: Run command failure: ../examples/chip-tool/commands/pairing/PairingCommand.cpp:341: CHIP Error 0x00000003: Incorrect state

### Bug prevalence

Whenever i do this

### GitHub hash of the SDK that was being used

not sure.  loaded latest from github

### Platform

windows, other

### Platform Version(s)

version 11 with wsl Ubuntu 20.04

### Anything else?

I have tried various commands to get anything and its always failing but I can see network connection is good as I can ping servers.
./chip-tool pairing with code, code-wifi, onnetwork all fail
./chip-tool payload discover commissionables also fails.

bzbarsky-apple, 2024-03-25:
> I tried using the chip-tool api to connect
> `./chip-tool pairing code 1 21104754926`

Where did that code come from?  Is that the one the Android app provided?

> `./chip-tool payload discover commissionables` also fails.

There is no "payload discover" command in chip-tool... 

What is avahi, if you have it installed, showing you about the commissionable advertisement in question?

pmatsibekker, 2024-03-25:
Thank you for your response!
I built chip-tool from the connected homepage master branch on my windows
Ubuntu distribution. Everything builds fine and the executable can be run.
I only use my android to first commission the outlet matter enabled vendor
device and I would like to control it out of the new Ubuntu commissioned
fabric.

pmatsibekker, 2024-03-25:
Copied the wrong command.. it was
./chip-tool discover commissionables

bzbarsky-apple, 2024-03-26:
`chip-tool discover commissionables` should generally work.  So if that's not working, then chances are either a commissioning window is not being opened, or the device that is commissionable and chip-tool are not on the same network, or something on that network is messing with dns-sd.

avahi could be a way to at least cross-check that the issue is not with chip-tool itself, just in case.

pmatsibekker, 2024-03-26:
Hi Boris,
Thank you so much for trying to help me here. I ran avahi_browser on the
_matterd._udp.local and I see it getting triggered when i try to do a
pairing.

I found that running $chip-tool interactive start   got me further in the
discovery process but still not working.
$ pairing code 0x1 26058243772

[1711490840.968705][3123:3123] CHIP:TOO: Command: pairing code 0x1
26058243772
[1711490840.969214][3123:3125] CHIP:CTL: Setting attestation nonce to
random value
[1711490840.969272][3123:3125] CHIP:CTL: Setting CSR nonce to random value
[1711490840.970726][3123:3125] CHIP:CTL: Generating NOC
[1711490840.971142][3123:3125] CHIP:FP: Validating NOC chain
[1711490840.971615][3123:3125] CHIP:FP: NOC chain validation successful
[1711490840.971684][3123:3125] CHIP:FP: Updated fabric at index: 0x1, Node
ID: 0x000000000001B669
[1711490840.971713][3123:3125] CHIP:TS: Last Known Good Time:
2023-10-14T01:16:48
[1711490840.971716][3123:3125] CHIP:TS: New proposed Last Known Good Time:
2021-01-01T00:00:00
[1711490840.971720][3123:3125] CHIP:TS: Retaining current Last Known Good
Time
[1711490841.041703][3123:3125] CHIP:FP: Metadata for Fabric 0x1 persisted
to storage.
[1711490841.066230][3123:3125] CHIP:TS: Committing Last Known Good Time to
storage: 2023-10-14T01:16:48
[1711490841.068374][3123:3125] CHIP:CTL: Joined the fabric at index 1.
Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 24EA3F3A4B34EE0E)
[1711490841.068413][3123:3125] CHIP:IN: UDP::Init bind&listen port=5550
[1711490841.068435][3123:3125] CHIP:IN: UDP::Init bound to port=5550
[1711490841.068439][3123:3125] CHIP:IN: UDP::Init bind&listen port=5550
[1711490841.068448][3123:3125] CHIP:IN: UDP::Init bound to port=5550
[1711490841.068450][3123:3125] CHIP:IN: TransportMgr initialized
[1711490841.068503][3123:3125] CHIP:DIS: Updating services using
commissioning mode 0
[1711490841.096093][3123:3125] CHIP:DIS: CHIP minimal mDNS started
advertising.
[1711490841.096654][3123:3125] CHIP:DL: Using wifi MAC for hostname
[1711490841.096686][3123:3125] CHIP:DIS: Advertise operational node
24EA3F3A4B34EE0E-000000000001B669
[1711490841.096737][3123:3125] CHIP:DIS: Responding with _matter._tcp.local
[1711490841.096765][3123:3125] CHIP:DIS: Responding with
24EA3F3A4B34EE0E-000000000001B669._matter._tcp.local
[1711490841.096814][3123:3125] CHIP:DIS: Responding with
24EA3F3A4B34EE0E-000000000001B669._matter._tcp.local
[1711490841.096818][3123:3125] CHIP:DIS: Responding with
00155D8988E20000.local
[1711490841.096821][3123:3125] CHIP:DIS: Responding with
00155D8988E20000.local
[1711490841.096825][3123:3125] CHIP:DIS: Responding with
_I24EA3F3A4B34EE0E._sub._matter._tcp.local
[1711490841.096827][3123:3125] CHIP:DIS: CHIP minimal mDNS configured as
'Operational device'; instance name: 24EA3F3A4B34EE0E-000000000001B669.
[1711490841.097312][3123:3125] CHIP:DIS: mDNS service published:
_matter._tcp
[1711490841.097388][3123:3125] CHIP:DL: Using wifi MAC for hostname
[1711490841.097422][3123:3125] CHIP:DIS: Advertise operational node
9F4224575B081A6B-000000000001B669
[1711490841.097461][3123:3125] CHIP:DIS: Responding with _matter._tcp.local
[1711490841.097490][3123:3125] CHIP:DIS: Responding with
9F4224575B081A6B-000000000001B669._matter._tcp.local
[1711490841.097513][3123:3125] CHIP:DIS: Responding with
9F4224575B081A6B-000000000001B669._matter._tcp.local
[1711490841.097516][3123:3125] CHIP:DIS: Responding with
00155D8988E20000.local
[1711490841.097519][3123:3125] CHIP:DIS: Responding with
00155D8988E20000.local
[... 30 lines cut ...]
over DNS-SD
[1711490841.169680][3123:3125] CHIP:CTL: Starting commissioning discovery
over DNS-SD
[1711490841.169809][3123:3125] CHIP:DL: Long dispatch time: 201 ms, for
event type 2
[1711490871.171686][3123:3125] CHIP:CTL: Discovery timed out
[1711490871.171727][3123:3125] CHIP:CTL: Stopping commissioning discovery
over DNS-SD
[1711490871.171736][3123:3125] CHIP:TOO: Secure Pairing Failed
[1711490871.171743][3123:3125] CHIP:TOO: Run command failure:
../../examples/chip-tool/commands/pairing/PairingCommand.cpp:341: CHIP
Error 0x00000003: Incorrect state

i ran avahi-browser on _matterd._ud=
 $ avahi-browse _matterd._udp -r

   txt = ["VP=65521+32769"]
+   eth0 IPv6 0E60B071FA8AA9C3                              _matterd._udp
     local
=   eth0 IPv6 0E60B071FA8AA9C3                              _matterd._udp
     local
   hostname = [00155D8988E20000.local]
   address = [172.31---.--]
   port = [5550]
   txt = ["VP=65521+32769"]

pmatsibekker, 2024-03-28:
Ok, so I have it working now. Couple of things to note. 
1) This does not work with WSL for me. It seems to need the full ubuntu install desktop install.
2) You need to --bypass-attestation-verifier=true to complete the commissioning.

Commissioning then complete successfully and I could see the new fabric linking on my android.
However, one concern was that the logging didn't provide me with the discriminator code that I could understand. I was able to successfully toggle a matter plug by using discriminator set to 1. 
If you know how to distinguish it from the logs that would be very helpful going forward. 

Otherwise, the controller does as advertised. Thank you!

closed 2024-03-28 by pmatsibekker

bzbarsky-apple, 2024-03-30:
> You need to --bypass-attestation-verifier=true to complete the commissioning.

Probably because it's signed by a PAA that has not been imported into the SDK source tree yet....  I hope.

> the logging didn't provide me with the discriminator code that I could understand. I was able to successfully toggle a matter plug by using discriminator set to 1.

I'm not sure what you mean @pmatsibekker.  Discriminator is only used during commissioning, to narrow the set of things you might try to commission.

Where are you using the discriminator in any way?  Normally you "use" it just by using the pairing code (which encodes the discriminator, or part of it in the case of the numeric code).

pmatsibekker, 2024-03-30:
Hi,
If I understand the api  correctly, the controllers need the discriminator
to distinguish the individual devices on the fabric node once commissioned.
I don't use the commissioning code after that. I guessed that the
discriminator would be 1 because it was the first item in the fabric node
and it would go up with each device. The api doesn't say so, but it seemed
to work or else I had a lucky guess with my first add. I didn't see in the
commissioning logs how to find that identifier clearly marked out.

bzbarsky-apple, 2024-04-01:
> If I understand the api  correctly, the controllers need the descriptor id to distinguish the individual devices on the fabric node once commissioned.

Once the device is commissioned, all you need is the node ID.

The node ID is assigned by the commissioner, and can be anything the commissioner wants, as long as it's unique within the commissioner's fabric.  The commissioning command used above:
```
pairing code 0x1 26058243772
```
is using `0x1` as the node ID to assign to the commissionee.  After that, you need to use the same node ID to sent it commands.  You can use any other node ID you want here, as desired....  It's totally up to you, as the commissioner.  

I'm not sure whether that actually answers your question/concern, though.  If it doesn't, can you please clearly explain what problem you are trying to address here?

pmatsibekker, 2024-04-01:
Yes, very helpful.