project-chip/connectedhomeip #32722: [Feature] Better internationalization

jonsmirl, 2024-03-26:
### Feature description

The existing localization cluster has a fixed set of locales. That doesn't work if you can download images with new locales in them. The problem is that the list of locales is fixed and it survives an update. There is no way for the new image to add new locales or remove old ones from the available locale list.

This locale list is also just a language list, not a locale list. For example it does not indicate the date order, units, etc 

The calendar option also doesn't do what you need. All of the US and EU are on the Gregorian calendar, but the date format varies all over. Is it is month first, day first or year first? 12 or 24 hour clock, metric or imperial units? The node can be "English" but then all of these attributes still vary with the region.

Again the phone can be used to set all of this in during node setup, but it needs to be set in a cross-fabric manner.

### Platform

all

bzbarsky-apple, 2024-03-27:
> The problem is that the list of locales is fixed and it survives an update. 

Fixed things can change across updates, by definition, no?

This seems like a spec request, not an SDK request.

jonsmirl, 2024-03-27:
Right now persistent state is not altered by OTA in over to preserve the state the user has set. There is no mechanism to change it.