project-chip/connectedhomeip #32727: [1.2] If SRP registration for any service fails, it may never be readvertised until power cycle or SRP server change

ndyck14, 2024-03-26:
### Reproduction steps

Steps originally reported in https://github.com/openthread/openthread/issues/9909

Code snippet from repo:
```
CHIP_ERROR DnssdServer::AdvertiseOperational()
{
    VerifyOrDie(mFabricTable != nullptr);

    for (const FabricInfo & fabricInfo : *mFabricTable)
    {
...
// Should we keep trying to advertise the other operational
        // identities on failure?
        ReturnErrorOnFailure(mdnsAdvertiser.Advertise(advertiseParameters));
```
If Advertise(..) fails, the service in question will not be registered until the node restarts, or the SRP server restarts (change in reachability).

We've otherwise worked around this, so unclear if others will run into it.

### Bug prevalence

rarely

### GitHub hash of the SDK that was being used

1.2

### Platform

efr32

### Type

Platform Issue

Damian-Nordic, 2024-03-27:
But what's the reason that `Advertise` fails other than SRP server is not available in which case the service refresh will be triggered when the SRP server shows up? Could you elaborate what exactly is wrong with the current behavior and what's the proposal?

ndyck14, 2024-03-27:
in our scenario, unfortunately we were never able to observe it directly on a device with logging. But the observation is consistent with the code: the `mdnsAdvertiser.Advertise(advertiseParameters)` call fails for some reason (for some, but not all services), and when it does, it has no recovery mechanism.

Its possible that our scenario is not valid (we've refactored away from how we originally added our own SRP registration). But the code is written as "we assume this call will never fail", which may be a problem in the future.