project-chip/connectedhomeip #32746: Is there any sdk for android app? Do you still use these? CHIPController.jar, SetupPayloadParser.jar, AndroidPlatform.jar

JasonClark2, 2024-03-27:
### Build issue(s)

1.
2.
3.
...

### Platform

android

yunhanw-google, 2024-03-27:
Yes, you can try android chip-tool, and exercise CHIPController.jar, SetupPayloadParser.jar, AndroidPlatform.jar. You can mimic the dependancy/configuration from android chip-tool and create your app.

in ubuntu/debian, you can do the below.
git clone https://github.com/project-chip/connectedhomeip.git
cd connectedhomeip
git pull
git submodule init
git submodule update
docker run -it -v ~/workspace/connectedhomeip:/connectedhomeip ghcr.io/project-chip/chip-build-android:latest
git config --global --add "*"
source scripts/bootstrap.sh
./scripts/build/build_examples.py --target android-arm64-chip-tool build

closed 2024-03-27 by yunhanw-google

JasonClark2, 2024-03-29:
After I follow your steps, I always get an error. Do I need to update something?

/scripts/build/build_examples.py --target android-arm64-chip-tool build
2024-03-29 05:07:25 INFO    Building targets: android-arm64-chip-tool
2024-03-29 05:07:25 INFO    Preparing builder 'android-arm64-chip-tool'
2024-03-29 05:07:25 INFO    Generating /connectedhomeip/out/android-arm64-chip-tool
2024-03-29 05:07:25 INFO    Setting up Android deps through Gradle
2024-03-29 05:07:25 INFO    Downloading https://services.gradle.org/distributions/gradle-7.2-bin.zip
2024-03-29 05:07:49 INFO    
2024-03-29 05:07:49 WARNING Exception in thread "main" java.net.SocketException: Unexpected end of file from server
2024-03-29 05:07:49 WARNING 	at sun.net.www.http.HttpClient.parseHTTPHeader(HttpClient.java:855)
2024-03-29 05:07:49 WARNING 	at sun.net.www.http.HttpClient.parseHTTP(HttpClient.java:678)
2024-03-29 05:07:49 WARNING 	at sun.net.www.http.HttpClient.parseHTTPHeader(HttpClient.java:852)
2024-03-29 05:07:49 WARNING 	at sun.net.www.http.HttpClient.parseHTTP(HttpClient.java:678)
2024-03-29 05:07:49 WARNING 	at sun.net.www.protocol.http.HttpURLConnection.getInputStream0(HttpURLConnection.java:1595)
2024-03-29 05:07:49 WARNING 	at sun.net.www.protocol.http.HttpURLConnection.getInputStream(HttpURLConnection.java:1500)
2024-03-29 05:07:49 WARNING 	at sun.net.www.protocol.https.HttpsURLConnectionImpl.getInputStream(HttpsURLConnectionImpl.java:268)
2024-03-29 05:07:49 WARNING 	at org.gradle.wrapper.Download.downloadInternal(Download.java:100)
2024-03-29 05:07:49 WARNING 	at org.gradle.wrapper.Download.download(Download.java:80)
2024-03-29 05:07:49 WARNING 	at org.gradle.wrapper.Install$1.call(Install.java:68)
2024-03-29 05:07:49 WARNING 	at org.gradle.wrapper.Install$1.call(Install.java:48)
2024-03-29 05:07:49 WARNING 	at org.gradle.wrapper.ExclusiveFileAccessManager.access(ExclusiveFileAccessManager.java:69)
2024-03-29 05:07:49 WARNING 	at org.gradle.wrapper.Install.createDist(Install.java:48)
2024-03-29 05:07:49 WARNING 	at org.gradle.wrapper.WrapperExecutor.execute(WrapperExecutor.java:107)
2024-03-29 05:07:49 WARNING 	at org.gradle.wrapper.GradleWrapperMain.main(GradleWrapperMain.java:63)
2024-03-29 05:07:49 WARNING Traceback (most recent call last):
2024-03-29 05:07:49 WARNING   File "/connectedhomeip/third_party/android_deps/set_up_android_deps.py", line 37, in <module>
2024-03-29 05:07:49 WARNING     sys.exit(main())
2024-03-29 05:07:49 WARNING   File "/connectedhomeip/third_party/android_deps/set_up_android_deps.py", line 30, in main
2024-03-29 05:07:49 WARNING     subprocess.check_call(
2024-03-29 05:07:49 WARNING   File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
2024-03-29 05:07:49 WARNING     raise CalledProcessError(retcode, cmd)
2024-03-29 05:07:49 WARNING subprocess.CalledProcessError: Command '['/connectedhomeip/third_party/android_deps/gradlew', '-p', '/connectedhomeip/third_party/android_deps', 'setUpAndroidDeps']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/connectedhomeip/./scripts/build/build_examples.py", line 230, in <module>
    main(auto_envvar_prefix='CHIP')
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1719, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/dist-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/connectedhomeip/./scripts/build/build_examples.py", line 220, in cmd_build
    context.obj.Build()
  File "/connectedhomeip/scripts/build/build/__init__.py", line 67, in Build
    self.Generate()
  File "/connectedhomeip/scripts/build/build/__init__.py", line 62, in Generate
    builder.generate()
  File "/connectedhomeip/scripts/build/builders/android.py", line 333, in generate
    self._Execute(
  File "/connectedhomeip/scripts/build/builders/builder.py", line 104, in _Execute
    self._runner.Run(cmdarray, title=title)
  File "/connectedhomeip/scripts/build/runner/shell.py", line 73, in Run
    raise Exception('Command %r failed: %d' % (cmd, code))
Exception: Command ['python3', 'third_party/android_deps/set_up_android_deps.py'] failed: 1