project-chip/connectedhomeip #32766: [Test Failed] DA-1.5 Python test execution failure

hoseokkim1, 2024-03-28:
### Test issue(s)

Matter v 1.2 certification is in progress, and the following error occurs when running the DA-1.5 python item.

1. cd chip-cer fica on-tool ./scripts/ubuntu/update-paa-certs.sh 
   -> After running the above command, I ensured that PAA is available locally in /var/paaroot-certs.

2. docker run -v /var/paa-root-certs:/paa_roots -v /var/run/dbus/system_bus_socket:/var/run/dbus/system_bus_socket -v /launch_dir --privileged --network host -it connectedhomeip/chip-cert-bins:5b4f8004662d00bdb111367fec7d3ea978c23372 bash
  -> I ran the above command and downloaded the docker image.

3. rm -f admin_storage.json && python3 TC_DA_1_5.py --discriminator 20202021 --passcode 3840 --commissioning-method ble-thread --thread-dataset-hex 0e08000000000001000035060004001fffe00708fd57c818dbc93ad80410124e06bc123c366d01396c77780dfd030c0402a0f7f8000300001901025b3502085b35dead5b35beef030435623335051000112233445566778899aabbccddeeff --ble-interface-id 0 --paa-trust-store-path /paa_roots --storage-path admin_storage.json  --PICS /home/ubuntu/connectedhomeip/src/python_testing/ci-pics-values 
   -> An error occurs when executing the above command on TH for commissioning with DUT.

Please refer to the error log below.

ubuntu@ubuntu:~/connectedhomeip/src/python_testing$ 
ubuntu@ubuntu:~/connectedhomeip/src/python_testing$ ls
TC_ACE_1_2.py   TC_DRLK_2_12.py               TC_ICDM_2_1.py       TC_RVCCLEANM_3_2.py   TC_TIMESYNC_2_1.py   TC_TIMESYNC_2_5.py      TestCommissioningTimeSync.py      drlk_2_x_common.py                    test_testing
TC_ACE_1_3.py   TC_DRLK_2_2.py                TC_IDM_1_2.py        TC_RVCOPSTATE_2_1.py  TC_TIMESYNC_2_10.py  TC_TIMESYNC_2_6.py      TestConformanceSupport.py         global_attribute_ids.py
TC_ACE_1_4.py   TC_DRLK_2_3.py                TC_OPSTATE_2_1.py    TC_RVCOPSTATE_2_3.py  TC_TIMESYNC_2_11.py  TC_TIMESYNC_2_7.py      TestMatterTestingSupport.py       hello_test.py
TC_CGEN_2_4.py  TC_DeviceBasicComposition.py  TC_OPSTATE_2_3.py    TC_RVCRUNM_1_2.py     TC_TIMESYNC_2_12.py  TC_TIMESYNC_2_8.py      TestTimeSyncTrustedTimeSource.py  matter_testing_support.py
TC_DA_1_2.py    TC_FAN_3_3.py                 TC_RR_1_1.py         TC_RVCRUNM_2_1.py     TC_TIMESYNC_2_13.py  TC_TIMESYNC_2_9.py      basic_composition_support.py      requirements.txt
TC_DA_1_5.py    TC_FAN_3_4.py                 TC_RVCCLEANM_1_2.py  TC_RVCRUNM_3_2.py     TC_TIMESYNC_2_2.py   TC_TIMESYNC_3_1.py      ci-pics-values                    spec_parsing_support.py
TC_DA_1_7.py    TC_FAN_3_5.py                 TC_RVCCLEANM_2_1.py  TC_SC_3_6.py          TC_TIMESYNC_2_4.py   TC_TestEventTrigger.py  conformance_support.py            taglist_and_topology_test_support.py
ubuntu@ubuntu:~/connectedhomeip/src/python_testing$ rm -f admin_storage.json && python3 TC_DA_1_5.py --discriminator 20202021 --passcode 3840 --commissioning-method ble-thread --thread-dataset-hex 0e08000000000001000035060004001fffe00708fd57c818dbc93ad80410124e06bc123c366d01396c77780dfd030c0402a0f7f8000300001901025b3502085b35dead5b35beef030435623335051000112233445566778899aabbccddeeff --ble-interface-id 0 --paa-trust-store-path /paa_roots --storage-path admin_storage.json  --PICS /home/ubuntu/connectedhomeip/src/python_testing/ci-pics-values
Traceback (most recent call last):
  File "/home/ubuntu/connectedhomeip/src/python_testing/TC_DA_1_5.py", line 19, in <module>
    import chip.clusters as Clusters
ModuleNotFoundError: No module named 'chip'
ubuntu@ubuntu:~/connectedhomeip/src/python_testing$

Can I get help resolving the error in this Python test item?

### Platform

python, raspi

hoseokkim1, 2024-03-28:
[image]

bzbarsky-apple, 2024-03-28:
@hoseokkim1 Shouldn't you be using `scripts/run_in_python_env.sh` to run the test?  So:
```
scripts/run_in_python_env.sh 'python3 TC_DA_1_5.py --args-go-here'
```