project-chip/connectedhomeip #32783: [Documentation] Clarification of use of Extra Clusters on an Endpoint

jvmahon, 2024-03-29:
### Documentation issues

1. Section 7.15.4 of the Matter 1.2 Core Specification  allows for the addition of "extra" clusters to an endpoint. See text below.
2. I believe this should be clarified to state that where a more complete / comprehensive device type exist that already include the additional clusters, that more comprehensive device type should be favored. For example, a Color Control cluster should not be added to the Dimmable Light type - instead, the type should be changed to Extended Color Light.  Perhaps the text to be added is something simple like "Implementations SHOULD use the device type with defined clusters and function that most closely corresponds to the included clusters and functions of an endpoint."
3. I am specifically working through this with a developer that is implementing a device using a Dimmable Light type with an added color control cluster, rather than just using Enhanced Color Light. The standard seems to allow this, but a more rigorous ap0plication of device types would be preferred.
...
[image]

jonsmirl, 2024-03-29:
You can have multiple device types on an endpoint. I have all of OnOff, Dimmable, ColorTemp and Extended Color device types set on my endpoint. This is especially important for binding since an OnOff switch is going to look to bind to an OnOff device, etc.

What annoys me more is an app like Google Home which doesn't respect the Device Type.  Please don't ignore the Device Type and then go search in the clusters trying to 'correct' my mistake.  The problem here, Google Home, is that you won't implement the Generic Switch device type. Please stop turning my Generic Switch endpoints into something else based on the clusters you discovered. My Generic Switch has an indicator LED; that does no mean you should turn it into a light bulb simply because you won't implement Generic Switch. Implement Generic Switch and then you'd know to ignore the cluster for the LED.

jvmahon, 2024-03-29:
> The problem here, Google Home, is that you won't implement the Generic Switch device type. Please stop turning my Generic Switch endpoints into something else based on the clusters you discovered.

Google isn't even doing that!  The Switch Cluster 0x003B is only used for Generic Switch. Google Home seems to be just creating a junk endpoint device with no function! 

On a related note, wouldn't you be better off setting the indicator light to its own endpoint where its could be given the unique and clear device type of On/Off light.

jonsmirl, 2024-03-29:
In my specific use case I don't want to separate the Generic Switch and indicator light onto different endpoints. My device has multiple buttons and that's how I can tell which LED goes with which button. I am fine with Google/Apple simply ignoring my indicator LEDs and letting my own app control them.

Technically should be constructed as 'composite' devices (composed of light+generic sw ep) but Google/Apple support for that is even more messed up. Right now I'd be happy if Google simple ignored my indicator LED.  If they make composite devices work I can always change my devices to match.