project-chip/connectedhomeip #32794: [Build] Chef rootnode_roomairconditioner_9cf3607804 Linux failure

lboue, 2024-03-29:
### Build issue(s)

Hello,

I am trying to build roomairconditioner example with chef and it fails:
1. git clean -Xdf
2. source ./scripts/bootstrap.sh
3. ./examples/chef/chef.py -b -d rootnode_roomairconditioner_9cf3607804 -t linux 

commit id: 6ed5dd31200554fb381e6f292337a86b761601c4

```
user@laptop:~/connectedhomeip$ ./examples/chef/chef.py -b -d rootnode_roomairconditioner_9cf3607804 -t linux 
Running for the first time and configuring config.yaml. Change this configuration file to include correct configuration for the vendor's SDK
ameba:
  AMEBA_SDK: null
  MATTER_SDK: null
  MODEL: D
  TTY: null
esp32:
  IDF_PATH: null
  TTY: null
nrfconnect:
  TTY: null
  ZEPHYR_BASE: null
  ZEPHYR_SDK_INSTALL_DIR: null
silabs-thread:
  CU: null
  GECKO_SDK: /home/user/connectedhomeip/examples/chef/../../third_party/efr32_sdk/repo
  SILABS_BOARD: null
  TTY: null
telink:
  TTY: null
  ZEPHYR_BASE: null
  ZEPHYR_SDK_INSTALL_DIR: null

  ______  __    __   _______  _______
 /      ||  |  |  | |   ____||   ____|
|  ,----'|  |__|  | |  |__   |  |__
|  |     |   __   | |   __|  |   __|
|  `----.|  |  |  | |  |____ |  |
 \______||__|  |__| |_______||__|

Target is set to rootnode_roomairconditioner_9cf3607804
Setting up environment...

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  ACTIVATOR! This sets your shell environment variables.

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Project actions........skipped
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20240329 23:11:34 INF Environment passes all checks!

Environment looks good, you are ready to go!

RPC PW disabled
Software Version String: ""
Product ID 0x8000 / Vendor ID 0xFFF1
Building...
x86_64
Done. Made 3640 targets from 290 files in 802ms
ninja: Entering directory `out'
ninja: error: '../third_party/connectedhomeip/third_party/jsoncpp/repo/src/lib_json/json_reader.cpp', needed by 'obj/third_party/connectedhomeip/third_party/jsoncpp/repo/src/lib_json/jsoncpp.json_reader.cpp.o', missing and no known rule to make it
Traceback (most recent call last):
  File "/home/user/connectedhomeip/./examples/chef/chef.py", line 1011, in <module>
    sys.exit(main())
  File "/home/user/connectedhomeip/./examples/chef/chef.py", line 872, in main
    shell.run_cmd("ninja -C out")
  File "/home/user/connectedhomeip/examples/chef/stateful_shell.py", line 139, in run_cmd
    raise RuntimeError(
RuntimeError: Error. Nonzero return code.
Returncode: 1
Cmd: ninja -C out

```

### Platform

other

jinbiaochenstayhungry, 2024-04-01:
looks like u json repo  didnot clone well，suggest check it first

lboue, 2024-04-01:
@jinbiaochenstayhungry 
I deleted the entire onnectedhomeip directory and cloned again from scratch. However, I still get the same error.
How can I fix it?

lboue, 2024-04-01:
Fixed with `git submodule update --init`.

closed 2024-04-01 by lboue