project-chip/connectedhomeip #32795: Error occurs while building CHIP-TOOL and sample apps in new connectedhomeip

Survensa, 2024-04-01:
### Feature Area

Other

### Test Case

q

### Reproduction steps

1. [git clone ](https://github.com/project-chip/connectedhomeip.git) https://github.com/project-chip/connectedhomeip.git
2. cd connectedhomeip
3. git pull
4. git checkout v1.3-branch
5. source scripts/bootstrap.sh
6. source scripts/activate.sh
7. scripts/examples/gn_build_example.sh      examples/all-clusters-app/linux/      examples/all-clusters-app/linux/out/all-clusters-app   chip_inet_config_enable_ipv4=false 

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

5bb5c9e23d532cea40476fc0bd1d3008522792ba

### Platform

raspi

jinbiaochenstayhungry, 2024-04-01:
seems like you clone of perfetto and nlunit-test had not successed .suggest you check it first.

Survensa, 2024-04-08:
Issue resolved with the `git submodule sync && git submodule update --init` before bootstrap

closed 2024-04-08 by Survensa