project-chip/connectedhomeip #32835: [1.3] TC-TMP-2.1 uses invalid YAML and has PICS set incorrectly.

cecille, 2024-04-03:
### Reproduction steps

```
Step 3:
    - label: "Step 3: TH reads the MaxMeasuredValue attribute from the DUT"
      PICS: TMP.S.A0002
      command: "readAttribute"
      attribute: "MaxMeasuredValue"
      response:
          saveAs: CurrentMaxMeasured
          constraints:
              type: int16s
              minValue: CurrentMinMeasured+1
              maxValue: 32767
   ```
Arithmetic expressions are not supported in yaml, which means "CurrentMinMeasured+1" is invalid. It also means that the CI is not running this step. Probably because the PICS is set incorrectly, even though this is a mandatory attribute.
          

### Bug prevalence

always

### GitHub hash of the SDK that was being used

e53dca89c1533b79a6ff8da8e3e9ee14e449770a

### Platform

other

### Platform Version(s)

yaml;

### Type

Test Improvement

Apollon77, 2024-04-03:
Thank you for validating

cecille, 2024-04-03:
@raul-marquez-csa FYI

cecille, 2024-04-03:
From ci-pics-values:

# Thermostat User Configuration cluster
TMP.S=1
TMP.S.A0000=1
TMP.S.A0001=1
TMP.S.A0002=1
TMP.S.A0003=1
TMP.M.ManuallyControlled=1

So why is this step not failing in the CI?

cecille, 2024-04-04:
OK, it looks like what's going on here is that the all clusters app returns null for these two attributes, and the yaml check therefore does not hit the affected line. There are a few things that need to happen as a result of this:

- we need a spec issue for the constraint on that cluster because it is underdescribed
- this test needs to be re-implemeted in python. This needs to happen somehow either BEFORE this lands as a part of the 1.3 package or with a CCB to back port to wherever that addition change landed
- We need to implement a linter to ensure that we're not trying to do arithmetic operations in yaml files. YAML is not the appropriate place to do math. While we're at it, let's also lint for the unit testing cluster in lieu of this because we've seen that workaround before too
- We need to understand what's going on with the constraints checker around null values, because this seems incorrect to just skip all the checks here.

cecille, 2024-04-04:
spec issue: https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/9091
CI issue: https://github.com/project-chip/matter-test-scripts/issues/218
tests with arithmetic operations: https://github.com/project-chip/matter-test-scripts/issues/217
test with unit testing cluster: https://github.com/project-chip/matter-test-scripts/issues/216
tracking YAML runner tests when returned value is null: https://github.com/project-chip/connectedhomeip/issues/32851

cecille, 2024-04-15:
https://github.com/project-chip/connectedhomeip/pull/32881 is the fix for this issue. Other issues will be tracked as above.

closed 2024-04-15 by cecille