project-chip/connectedhomeip #32851: [BUG] YAML runner appears to skip all checks if the returned value is null

cecille, 2024-04-04:
### Reproduction steps

See TC-TMP-2.1. Current all clusters returns null for MinMeasuredValue and MaxMeasuredValue, but the test passes despite having constraint checks for type AND for min/max.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

e53dca89c1533b79a6ff8da8e3e9ee14e449770a

### Platform

core

bzbarsky-apple, 2024-04-05:
And yes, I just confirmed that `notValue: null` does not fail if the value is in fact null.  So we need to fix that.

bzbarsky-apple, 2024-04-05:
https://github.com/project-chip/connectedhomeip/pull/32871 should fix `notValue: null`.

cecille, 2024-04-05:
Thanks, Boris. Ok, sounds like the runner is correct, and the test just isn't properly accounting for this case. I'll close this issue and handle it in the test itself.

closed 2024-04-05 by cecille