project-chip/connectedhomeip #32875: [1.3] NetworkCommissioning Cluster reads SSID field for Thread server instances

s-mcclain, 2024-04-05:
### Reproduction steps

There are some Matter Admins which may populate an empty SSID for the ScanNetworks command in NetworkCommissioning cluster in the field. The recent changes to the SDK now always checks if the SSID field exists regardless if the field is empty or not.

This will result in a Matter Accessory with these changes to always fail the ScanNetworks request.

The Matter specification states the following about the SSID FIeld:

```
11.9.7.1

SSID Field
This field, if present, SHALL contain the SSID for a directed scan of that particular Wi-Fi SSID. Otherwise, if the field is absent, or if it is null, this SHALL indicate scanning of all BSSID in range. This field SHALL be ignored for ScanNetworks invocations on non-WiFi server instances.
```

The last sentence indicates a requirement for the SSID field to be ignored for non-WiFi servers.

This was introduced in #32019 

Recommendation: 
Add check `if ssid.HasValue() && !ssid.Value().empty()`

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

660c928 

### Platform

core

### Platform Version(s)

1.3

### Type

Spec Compliance Issue

bzbarsky-apple, 2024-04-05:
> The last sentence indicates a requirement for the SSID field to be ignored for non-WiFi servers.

The problem is that this sentence contradicts the normative requirement in the Conformance column, which says that this field is only allowed to be present when the WI feature is supported.

But yes, the fact that the spec says self-contradictory things is not good, and we need to change one part or the other here... @tcarmelveilleux thoughts?

tcarmelveilleux, 2024-04-09:
The issue is that the test plan for TC-CNET-4.22 took the SHALL:

```
This field SHALL be ignored for ScanNetworks invocations on non-WiFi server instances
```

and misinterpreted that it had to be allowed for TH. I think we have to clean-up the conformance here. Since 1.0, TC-CNET-4.22 has been enforcing (in theory) that a TH server tolerates the NULL. It's just that this is not checked in CI since Thread based commissioning is not in CI, and the SDK auto-commissioner anyway doesn't make sure of ScanNetworks.

PR #32884 (2024-04-09): Temp fix for #32875 (network commissioning cluster check for SSID on Thread feature)
Fixes  #32875

As part of network commissioning cluster cleanup in 1.3, a check was added to the scan networks handler which checks for the SSID argument on a Thread device and returns an error when found.

There is confusing text in the spec which states that non-WiFi implementations should ignore this argument when null, and so the new check appears to not be conformant to that. But regardless, a popular commissioner was found to be passing a null SSID argument and this change breaks Thread device commission [... cut]
closed 2024-04-09 by mergify[bot]