project-chip/connectedhomeip #32900: [BUG] Commissioning failure with randomized hostname + reset

caipiblack, 2024-04-08:
### Reproduction steps

Hello,

I found an issue with the `Nanoleaf NL67E200` (Certified E27 Color Light), the issue appear at 100%.

**Initial steps:**
1. You need an OpenThread border router
2. Configure these variables according to your setup:
```
export NODE_ID=11
export DATASET=0e08000000000001000035060004001fffe00208641cfa7a52f52d110708fd5a64614c71f9260510f85e77f89beaa664492f3149c6376b3c030f4f70656e5468726561642d3961643801029ad80410bbe309ce7b54d8397be938782550779a0c0402a0f7f8000300000d
export QR_CODE=0214-451-7862
```
3. Add these folders:
```
mkdir -p /tmp/chiptool1
mkdir -p /tmp/chiptool2
```

**Testing steps:**
1. You commission the device using chip-tool (on the computer A):
```
./chip-tool pairing code-thread ${NODE_ID} hex:${DATASET} ${QR_CODE} --bypass-attestation-verifier 1 --storage-directory /tmp/chiptool1
```
2. With chiptool (on the computer A), you turn on the OpenCommissioningWindow on the device:
```
./chip-tool pairing open-commissioning-window ${NODE_ID} 0 300 10000 0 --storage-directory /tmp/chiptool1
```
3. With chiptool (on the computer B), you commission the device (it will be using DNS-SD to search the device):
```
./chip-tool pairing code ${NODE_ID} ${QR_CODE} --bypass-attestation-verifier 1 --storage-directory /tmp/chiptool2
```
4. You reset the device: (Off/On sequence x5, then the light blinks)
5. You do the steps 1 to 3 again :D, you will get this result:
```
Discovered device does not have an open commissioning window.
```

_Note: To run the tests, you can use two computers or the same computer but you use --storage-directory arguements to make sure chip-tool has two instances. As we don't want to disturb the test_

I made some debug and this is what I see:
- First it seems that this device continue to advertise on `_matterc._udp` after being commissioning with CM=0, this is probably normal (not sure). (It appears to be "Extended Discovery" in the specifications, but I don't know if it's normal to do it when the device is commissioned).
- This device seems use a randomized hostname (this is the most important here)

So when we are at the step 5, we have this DNS-SD records on the device:

```
caipiblack@caipiblack-computer:~/connectedhomeip-v1.2.0.1/out/host$ avahi-browse -r _matterc._udp
+ wlp0s20f3 IPv6 3337AB8EF48791D9                              _matterc._udp        local
+ wlp0s20f3 IPv6 2723F2DDF0373E6F                              _matterc._udp        local
+ wlp0s20f3 IPv6 977E393BA9DDD02F                              _matterc._udp        local
+ wlp0s20f3 IPv4 3337AB8EF48791D9                              _matterc._udp        local
+ wlp0s20f3 IPv4 2723F2DDF0373E6F                              _matterc._udp        local
+ enp1s0 IPv6 3337AB8EF48791D9                              _matterc._udp        local
+ enp1s0 IPv6 2723F2DDF0373E6F                              _matterc._udp        local
+ enp1s0 IPv4 3337AB8EF48791D9                              _matterc._udp        local
+ enp1s0 IPv4 2723F2DDF0373E6F                              _matterc._udp        local
= wlp0s20f3 IPv6 3337AB8EF48791D9                              _matterc._udp        local
   hostname = [E6F1EA1B788A60FC.local]
   address = [fd6c:b7b1:c8d0:1:a013:36cb:832f:d6ff]
   port = [5540]
   txt = ["PI=" "PH=36" "RI=0301553349C2F37A81D4FADF04B3B4465A71" "CM=1" "D=262" "T=0" "SAI=300" "SII=5000" "VP=4442+67"]
= wlp0s20f3 IPv6 2723F2DDF0373E6F                              _matterc._udp        local
   hostname = [864AD1AEDBD649C1.local]
   address = [fd6c:b7b1:c8d0:1:3816:6398:35ba:1c70]
   port = [5540]
   txt = ["PI=" "PH=36" "RI=010111B09FF44019AD352120EBAC616564F8" "CM=0" "D=262" "T=0" "SAI=300" "SII=5000" "VP=4442+67"]
= wlp0s20f3 IPv4 3337AB8EF48791D9                              _matterc._udp        local
   hostname = [E6F1EA1B788A60FC.local]
   address = [fd6c:b7b1:c8d0:1:a013:36cb:832f:d6ff]
   port = [5540]
   txt = ["PI=" "PH=36" "RI=0301553349C2F37A81D4FADF04B3B4465A71" "CM=1" "D=262" "T=0" "SAI=300" "SII=5000" "VP=4442+67"]
= wlp0s20f3 IPv4 2723F2DDF0373E6F                              _matterc._udp        local
   hostname = [864AD1AEDBD649C1.local]
   address = [fd6c:b7b1:c8d0:1:3816:6398:35ba:1c70]
   port = [5540]
   txt = ["PI=" "PH=36" "RI=010111B09FF44019AD352120EBAC616564F8" "CM=0" "D=262" "T=0" "SAI=300" "SII=5000" "VP=4442+67"]
= enp1s0 IPv6 3337AB8EF48791D9                              _matterc._udp        local
   hostname = [E6F1EA1B788A60FC.local]
   address = [fd6c:b7b1:c8d0:1:a013:36cb:832f:d6ff]
   port = [5540]
   txt = ["PI=" "PH=36" "RI=0301553349C2F37A81D4FADF04B3B4465A71" "CM=1" "D=262" "T=0" "SAI=300" "SII=5000" "VP=4442+67"]
= enp1s0 IPv6 2723F2DDF0373E6F                              _matterc._udp        local
   hostname = [864AD1AEDBD649C1.local]
   address = [fd6c:b7b1:c8d0:1:3816:6398:35ba:1c70]
   port = [5540]
   txt = ["PI=" "PH=36" "RI=010111B09FF44019AD352120EBAC616564F8" "CM=0" "D=262" "T=0" "SAI=300" "SII=5000" "VP=4442+67"]
= enp1s0 IPv4 3337AB8EF48791D9                              _matterc._udp        local
   hostname = [E6F1EA1B788A60FC.local]
   address = [fd6c:b7b1:c8d0:1:a013:36cb:832f:d6ff]
   port = [5540]
   txt = ["PI=" "PH=36" "RI=0301553349C2F37A81D4FADF04B3B4465A71" "CM=1" "D=262" "T=0" "SAI=300" "SII=5000" "VP=4442+67"]
= enp1s0 IPv4 2723F2DDF0373E6F                              _matterc._udp        local
   hostname = [864AD1AEDBD649C1.local]
   address = [fd6c:b7b1:c8d0:1:3816:6398:35ba:1c70]
   port = [5540]
   txt = ["PI=" "PH=36" "RI=010111B09FF44019AD352120EBAC616564F8" "CM=0" "D=262" "T=0" "SAI=300" "SII=5000" "VP=4442+67"]
```

Important things in these records:
- We have 4 records per device (IPv4, IPv6 on two interfaces)
- So we have 8 records because the device appear twice
- The device does not have the same hostname each time :)

So as I understand this is what happens:
- After the step 3, the device continue to advertise with CM=0
- The device is then reseted
- On the next boot, the device hostname change as it is randomized :)
- Then the matter-stack try to remove all SRP records for this "new" hostname (one of the first step when entering in the thread network). But the old records for the old hostname stay present
- When the device is put in OCW again, it advertise with new hostname with CM=1
- PROBLEM: When we try to commission using chip-tool, we only find "one" device, and it take the older one with CM=0 so: commissioning failure.

Now we have to find where the problem is: (specifications ? device stack ? controller stack ?),

**The specifications says:**

```
For DNS‑SD a target host name is required, in addition to the instance name. The target host name
SHALL be constructed using one of the available link-layer addresses, such as a 48-bit device MAC
address (for Ethernet and Wi‑Fi) or a 64-bit MAC Extended Address (for Thread) expressed as a
fixed-length twelve-character (or sixteen-character) hexadecimal string, encoded as ASCII (UTF-8)
text using capital letters, e.g., B75AFB458ECD.<domain>. In the event that a device performs MAC
address randomization for privacy, then the target host name SHALL use the privacy-preserving
randomized version and the hostname SHALL be updated in the record every time the underlying
link-layer address rotates. Note that it is legal to reuse the same hostname on more than one inter­
face, even if the underlying link-layer address does not match the hostname on that interface, since
the goal of using a link-layer address is to ensure local uniqueness of the generated hostname. If
future link layers are supported by Matter that do not use 48-bit MAC addresses or 64-bit MAC
Extended Address identifiers, then a similar rule will be defined for those technologies.
```

**There is something not clear:**

```
In the event that a device performs MAC
address randomization for privacy, then the target host name SHALL use the privacy-preserving
randomized version and the hostname SHALL be updated in the record every time the underlying
link-layer address rotates.
```

Question:
- Does the hostname have to be persistent at each reset? :1st_place_medal: 

### Bug prevalence

100%

### GitHub hash of the SDK that was being used

v1.2.0.1

### Platform

other

bzbarsky-apple, 2024-04-09:
>     there is a problem in Linux mDNS resolver implementation (only return one record)

This is the problem, if it's happening, and it could cause other issues too (e.g. if you have a discriminator collision).  

The Darwin (Mac/iOS) browse implementation returns all the records that match the discriminator, and then SetUpCodePairer can try them all one by one...

For the other questions:

> but I don't know if it's normal to do it when the device is commissioned

That's the only time you _can_ advertise with CM=0 usefully, really.

> Did the hostname have to be persistent after reset? 

No.

caipiblack, 2024-04-16:
My collegues tell me that after updating the nanoleaf bulb from 3.5.10 to 3.6.173 "it works better".

https://helpdesk.nanoleaf.me/en-US/nanoleaf-essentials-matter-release-notes-255125

I will check at the end of the week what has changed. (For exemple, it's possible that they remove MAC randomization)

But in any cases:

- Only ONE mDNS record was transmit to higher level during the commissioning and it was the wrong one, so the commissioning failed because the IPv6 was not the good one or the CM field of the mDNS packet was 0 (i had the two cases)

It can also be "viewed" by registering a discovery delegate: `CurrentCommissioner().RegisterDeviceDiscoveryDelegate(this)`, the function `OnDiscoveredDevice(const chip::Dnssd::DiscoveredNodeData & nodeData)` was only called one time, with the wrong data in nodeData.

Can someone check why the Linux implementation of mDNS is only returning one record in some situations ?