project-chip/connectedhomeip #32908: [1.2] Matter Denial of Service attack by replaying CASE Sigma1 messages

lxj199722, 2024-04-09:
### Reproduction steps

Matter device have joined wifi and commissioning into Apple’s fabric、Google’s fabric、CHIP’s fabric.
1. On the CHIP controller’s host start a packet capture (e.g., via tcpdump, Wireshark)
2. For each target device issue the opening of a CASE session by using the chip-tool; for this experiment we used the command to read the device’s vendor name:
    $ ./chip-tool basicinformation read vendor-name <node_id> <endpoint_id>
3. Open the capture file and export as HEX dump the CASE Sigma1 Matter header, payload header and application payload; as shown in the figure below, the export should include the entire Matter application layer
4. Import the HEX dump in the application that is used to replay the CASE Sigma1 message; for this purpose a Python application was used (attached to this report, file name: case-sigma1-replay.py)
5. Replay the CASE Sigma1 message (tested with 2 packets / second,  up to 15 packets / second), while also increasing the value of the Message Counter for each message
6. Run the attack for aprox. 20 minutes
7. Stop the attack
8. Wait a few hours / days – the devices should be displayed by Google / Apple Home applications as off-line until reboot

How to deny the repeat CASE sigma1 and resume the CASE session?

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

181b0cb

### Platform

ameba

### Platform Version(s)

1.2.0.1

### Type

Security Issue

### Anything else?

[case-sigma1-replay.txt](https://github.com/project-chip/connectedhomeip/files/14915716/case-sigma1-replay.txt)

bzbarsky-apple, 2024-04-09:
> Wait a few hours / days – the devices should be displayed by Google / Apple Home applications as off-line until reboot

This part should not be happening.

What I would expect to happen is that the device gets one Sigma1 message, responds with Sigma2.

If it now gets more Sigma1 messages before its wait for the Sigma2 times out, it just immediately responds BUSY to those.

At some point, if the other side is just not acking anything, the device will run out of exchanges and start dropping the Sigma1 messages without responding to them.

But once the message flood stops, I would expect the device to time out whatever Sigma2 it's waiting for a response for, then be back to normal operation the next time Apple Home or Google Home tries to establish CASE to it.

lxj199722, 2024-04-10:
> This part should not be happening.
>
> What I would expect to happen is that the device gets one Sigma1 message, responds with Sigma2.
>
> If it now gets more Sigma1 messages before its wait for the [...]

In the one figure below an example sequence of a CASE Sigma1 message is replayed against a Matter Smart Plug. As shown in Figure, the protocol opcode is 0x30 (CASE Sigma1), while the protocol opcode in Figure is 0x31 (CASE Sigma2). The packet sequence shown in the figures clearly displays the payload lengths. It is observable that there are multiple packets with a payload length of 166 bytes (i.e., the payload length of CASE Sigma1 messages), for which the target device generates responses (CASE Sigma2 messages) with payload length of 499 bytes.
[image]

It was observed that a CASE session cannot be established while the attack is running. For this purpose the chip-tool was used to issue attribute interrogations – timeout was reported. However, existing CASE sessions seem not to be affected and devices accept commands (e.g., turn ON / OFF).

In the case of the target device (Matter Smart Plug) a second behavior was also observed:
* After stopping the attack a CASE session could NOT be established even after waiting for several days (the chip-tool was used, as well as restarting border routers)
* The only solution seems to be to unplug and plug in the devices

bzbarsky-apple, 2024-04-10:
> It was observed that a CASE session cannot be established while the attack is running.

Yes, that's expected given the current "can only support one CASE establishment at a time as a server" setup.

> After stopping the attack a CASE session could NOT be established even after waiting for several day

This part is very much not expected.  Is this reproducible with the example apps in the SDK, or only with the specific Matter Smart Plug being tested?

lxj199722, 2024-04-11:
> Yes, that's expected given the current "can only support one CASE establishment at a time as a server" setup.
>
>
> This part is very much not expected. Is this reproducible with the example apps in the [...]

A iot security labs find the security issue in our matter products. We have contacted the matter sdk platform provider ameba, they believe we need to request help from matter. Maybe,  the iot security lab's test device matter version is too old, we ask them to update the newest firmware version use matter version v1.2.0.1.

closed 2024-04-22 by lxj199722