project-chip/connectedhomeip #32926: [Test Failed] Test_TC_LVL_2_1.yaml - step 9b [OnLevel] fails

xvortep, 2024-04-10:
### Test issue(s)

Greetings,

I've encountered a problem while running tests on LevelControlCluster

```
INFO       | 2024-04-08 12:56:05.407419 | Executing Test Step: Step 9b: TH reads the OnLevel attribute from the DUT

ERROR      | 2024-04-08 12:56:05.417099 | Test Step Error: Error occurred during execution of test case TC-LVL-2.1. '>=' not supported between instances of 'int' and 'NoneType'

INFO       | 2024-04-08 12:56:05.418558 | Test Step Completed [ERROR]: Step 9b: TH reads the OnLevel attribute from the DUT
```

As I understand, since our setup has a FeatureMap set to 0 (no additional features are implemented), after reading an OnLevel Attribute, it is compared with **MinLevelFeatureMapNotSupportedValue** and **MaxLevelFeatureMapNotSupportedValue** which are not defined in our case, since **MinLevelFeatureMapNotSupportedValue is defined only when LVL.S.A0002 && !LVL.S.F01 is fullfilled, and for MaxLevelFeatureMapNotSupportedValue - LVL.S.A0003 && !LVL.S.F01.**
This results in comparison error indicated above ( '>=' not supported between instances of 'int' and 'NoneType' ).

The easiest fix i see is to add a new step e.g: 
```
    - label: "Step 9c: Reads the OnLevel attribute "
      PICS: LVL.S.A0011 && !LVL.S.F01 && !LVL.S.A0002 && !LVL.S.A0003
      command: "readAttribute"
      attribute: "OnLevel"
      response:
          constraints:
              type: int8u
              minValue: 0
              maxValue: 254

```

Thanks in advance.

### Platform

all

bzbarsky-apple, 2024-04-10:
@xvortep This looks like an issue for https://github.com/CHIP-Specifications/chip-test-scripts/issues if it's a problem in the test script.

bzbarsky-apple, 2024-04-10:
@cjandhyala

cjandhyala, 2024-04-10:
correct. transferring the issue to scripts repo. BTW is this blocking your certification testing ?