project-chip/connectedhomeip #32941: [Feature] XML Device Types define Endpoint Composition

paulr34, 2024-04-11:
### Feature description

ZAP utilizes the data model XML and has recently incorporated support for Parent Endpoints.

For ZAP to store Endpoint Composition into the database, it requires the Endpoint Composition to be specified in the Device Type XML.

For instance, the XML should accurately represent the specifications regarding Endpoint Composition. Notably, the current XML lacks any mention of the Temperature Controlled Cabinet being mandatory for a Refrigerator Device Type.

[image]

Referencing the "old XML" [here](https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/matter-devices.xml#L2017-L2031) and the "new XML" [here](https://github.com/project-chip/connectedhomeip/blob/7bf764ab05eb4ef52b007666e822ba963f643922/data_model/device_types/Refrigerator.xml), it's evident that we need the Device Type to be defined as either "Tree" or "FullFamily" if the Device Type includes Endpoint Composition.

### Platform

all

paulr34, 2024-04-11:
@bzbarsky-apple it makes sense based on previous discussion to do something like following:

Example 1

```
<endpointComposition>
         <type>fullFamily</type>
         <endpoint>
                    <deviceType> name of device type </deviceType>
         </endpoint>
</endpointComposition>
```
OR

Example 2

```
<endpointComposition>
         <endpoint>
                    <deviceType> name of device type </deviceType>
         </endpoint>
         <endpoint>
                    <deviceType> name of device type </deviceType>
         </endpoint>
</endpointComposition>
```

ZAP can parse the XML in this case and know 
1. This Device Type has Endpoint Composition
2. This Device Type is fullFamily or not by the "type"
3. If the "type" is not "fullFamily" then it is "tree" like in Example 2

Alternatively we could put "type" for tree as well like:

Example 3

```
<endpointComposition>
         <type> tree </type>
         <endpoint>
                    <deviceType> name of device type </deviceType>
         </endpoint>
         <endpoint>
                    <deviceType> name of device type </deviceType>
         </endpoint>
</endpointComposition>
```

I might be out of date or missing some context, but If something along these lines is added to the XML, ZAP can parse and add to the database per Device Type

paulr34, 2024-04-16:
It's logical to assign an attribute to an endpoint XML tag to indicate whether it's optional. In the example below, the first endpoint is marked as mandatory (true), while the second one is not (false), making it optional. Despite not being mandatory, the presence of the second endpoint in the table implies that it's allowed.

Something along these lines makes sense to me:

```
<endpointComposition>
         <endpoint mandatory=true>
                    <deviceType> name of device type </deviceType>
         </endpoint>
         <endpoint mandatory=false>
                    <deviceType> name of device type </deviceType>
         </endpoint>
</endpointComposition>
```

paulr34, 2024-04-16:
If there are Conformance rules like how there are with "features" we could add something like
`      <optionalConform>
        <notTerm>
          <orTerm>
            <feature name="LT"/>
            <feature name="DF"/>
          </orTerm>
        </notTerm>
      </optionalConform>`

Regarding Device Type or Id, from a ZAP perspective either *could* work but let me double check and get back to you tomorrow.

paulr34, 2024-04-17:
It makes most sense for ZAP to consume the Device Type ID rather than the name

bzbarsky-apple, 2024-04-17:
That seems pretty reasonable.

PR #34818 (2024-08-21): adding mandatory endpointComposition data to XML for ZAP to consume
Fixes #32941 Adds mandatory endpointComposition data to XML for ZAP to consume
closed 2024-08-21 by mergify[bot]