project-chip/connectedhomeip #32966: [Build] gn_bootstrap.sh issue on grpc

KannanIdris, 2024-04-12:
### Build issue(s)

(.venv) user@matter-light:~/connectedhomeip $ sudo ./scripts/build/gn_bootstrap.sh

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (2.2s)
  Setting up Python environment.....[-]
ninja: Entering directory `/home/user/connectedhomeip/.environment/gn_out'
[1/498] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/linux:linux_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/matter_build_venv._install_3p_deps.pw_pystamp --capture-output --module pip --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._install_3p_deps_metadata_path_list.txt -- install --constraint ../../scripts/constraints.txt --requirement ../../scripts/requirements.txt --requirement python/gen/matter_build_venv/generated_requirements.txt
FAILED: python/gen/matter_build_venv._install_3p_deps.pw_pystamp
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//build/toolchain/linux:linux_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/matter_build_venv._install_3p_deps.pw_pystamp --capture-output --module pip --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/matter_build_venv._install_3p_deps_metadata_path_list.txt -- install --constraint ../../scripts/constraints.txt --requirement ../../scripts/requirements.txt --requirement python/gen/matter_build_venv/generated_requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Ignoring click: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring future: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring pyparsing: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring idf-component-manager: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring gdbgui: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring pygdbmi: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring reedsolo: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring bitstring: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring ecdsa: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring kconfiglib: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring construct: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring python-socketio: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring mbed-tools: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring pytest: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring mbed-ls: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring pdoc3: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring gitpython: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring python-dotenv: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring pyudev: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring typing-extensions: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring bflb-iot-tool: markers 'platform_machine != "aarch64" and sys_platform == "linux"' don't match your environment
Ignoring pyobjc-core: markers 'sys_platform == "darwin"' don't match your environment
Ignoring pyobjc-framework-cocoa: markers 'sys_platform == "darwin"' don't match your environment
Ignoring pyobjc-framework-corebluetooth: markers 'sys_platform == "darwin"' don't match your environment
Ignoring pandas: markers 'platform_machine != "aarch64" and platform_machine != "arm64"' don't match your environment
Ignoring graphlib-backport: markers 'python_version < "3.9"' don't match your environment
Collecting pip-tools
  Using cached https://www.piwheels.org/simple/pip-tools/pip_tools-6.1.0-py3-none-any.whl (44 kB)
Collecting virtualenv
  Using cached https://www.piwheels.org/simple/virtualenv/virtualenv-20.4.7-py2.py3-none-any.whl (7.2 MB)
Collecting pytest-json-report>=1.5.0
  Using cached https://www.piwheels.org/simple/pytest-json-report/pytest_json_report-1.5.0-py3-none-any.whl (18 kB)
Collecting leb128
  Using cached https://www.piwheels.org/simple/leb128/leb128-1.0.7-py3-none-any.whl (4.0 kB)
Collecting requests>=2.24.0
  Using cached https://www.piwheels.org/simple/requests/requests-2.25.1-py2.py3-none-any.whl (61 kB)
Requirement already satisfied: wheel in ./python-venv/lib/python3.11/site-packages (from -r ../../scripts/requirements.txt (line 24)) (0.36.2)
Collecting pgi
  Using cached https://www.piwheels.org/simple/pgi/pgi-0.0.11.2-py3-none-any.whl (173 kB)
Collecting portpicker
  Using cached https://www.piwheels.org/simple/portpicker/portpicker-1.4.0-py3-none-any.whl (13 kB)
Collecting mobly
  Using cached https://www.piwheels.org/simple/mobly/mobly-1.11.1-py3-none-any.whl (133 kB)
Collecting west>=0.12.0
[... 418 lines cut ...]
    subprocess.check_call(ninja_cmd, stdout=outs, stderr=outs)
  File "/usr/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['ninja', '-C', '/home/user/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/user/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 795, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/user/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 787, in main
    return EnvSetup(**vars(parse())).setup()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 457, in setup
    result = step(spin)
             ^^^^^^^^^^
  File "/home/user/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 606, in virtualenv
    if not virtualenv_setup.install(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 342, in install
    install_packages(gn_target)
  File "/home/user/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 330, in install_packages
    raise subprocess.CalledProcessError(err.returncode, err.cmd,
subprocess.CalledProcessError: Command '['ninja', '-C', '/home/user/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

adidasPR, 2024-08-07:
Was this issue ever resolved @KannanIdris ? I'm having the exact same issue so looking for some guidance here. Please help!!

andy31415, 2024-10-24:
what is your O/S configuration here to reproduce? We seem to successfully bootstrap on master currently in CI and on our devel machines, so wondering what your environment contains (and generally trying to figure out if this is #32687 or something else)

andy31415, 2024-10-24:
Error seems to say:

```
Collecting grpc>=1.0.0
  Using cached grpc-1.0.0.tar.gz (5.2 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-hkg6rpyu/grpc_1c13cd9095744c60a8f3b225141f5204/setup.py", line 33, in <module>
          raise RuntimeError(HINT)
      RuntimeError: Please install the official package with: pip install grpcio
      [end of output]
```

yunhaod, 2024-11-04:
I found this: https://stackoverflow.com/questions/78419357/why-wont-the-grpc-package-install-on-my-system
Is there a particular script that calls for: pip install grpc?

andy31415, 2024-11-05:
GRPC is probably pulled in from some python dependencies. We generally have constraints set in https://github.com/project-chip/connectedhomeip/blob/v1.1-branch/scripts/setup/constraints.txt however even though we try to be hermetic, package versions still seem to ocassionally break over time.

As far as I am aware, the v1.0-branch is not maintained anymore by the code SDK maintainers as too many branches add up over time. I would recomment to work with the latest version (or at least the latest stable branch) instead.

woody-apple, 2025-04-02:
Closing this.