project-chip/connectedhomeip #32979: Issue rpc support for all-cluster-app cannot compile

Chapoly1305, 2024-04-13:
From https://github.com/project-chip/connectedhomeip/issues/31002

Compile process will fail if I enable the rpc. This is the command I have used on Ubuntu 22.04, v1.3-branch.

```
./scripts/examples/gn_build_example.sh examples/all-clusters-app/linux/ app-cluster-app chip_enable_pw_rpc=true 'import("//with_pw_rpc.gni")'
```

This is the error message from console,

```
ACTIVATOR! This sets your shell environment variables.

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Project actions........skipped
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20240412 23:01:15 INF Environment passes all checks!

Environment looks good, you are ready to go!

+ env
SHELL=/bin/bash
SESSION_MANAGER=local/chen-Z690-AERO-G-DDR4:@/tmp/.ICE-unix/2026,unix/chen-Z690-AERO-G-DDR4:/tmp/.ICE-unix/2026
QT_ACCESSIBILITY=1
COLORTERM=truecolor
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg
SSH_AGENT_LAUNCHER=gnome-keyring
XDG_MENU_PREFIX=gnome-
TERM_PROGRAM_VERSION=1.88.1
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
GTK_IM_MODULE=ibus
GNOME_SHELL_SESSION_MODE=ubuntu
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
PW_PROJECT_ROOT=/home/chen/connectedhomeip
XMODIFIERS=@im=ibus
DESKTOP_SESSION=ubuntu
PW_ZAP_CIPD_INSTALL_DIR=/home/chen/connectedhomeip/.environment/cipd/packages/zap
GTK_MODULES=gail:atk-bridge
PWD=/home/chen/connectedhomeip
LOGNAME=chen
XDG_SESSION_DESKTOP=ubuntu
XDG_SESSION_TYPE=x11
GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1
SYSTEMD_EXEC_PID=2049
XAUTHORITY=/run/user/1000/gdm/Xauthority
VSCODE_GIT_ASKPASS_NODE=/usr/share/code/code
CIPD_CACHE_DIR=/home/chen/.cipd-cache-dir
GJS_DEBUG_TOPICS=JS ERROR;JS LOG
WINDOWPATH=2
PW_ROOT=/home/chen/connectedhomeip/third_party/pigweed/repo
HOME=/home/chen
USERNAME=chen
LANG=en_US.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
XDG_CURRENT_DESKTOP=Unity
VIRTUAL_ENV=/home/chen/connectedhomeip/.environment/pigweed-venv
GIT_ASKPASS=/usr/share/code/resources/app/extensions/git/dist/askpass.sh
INVOCATION_ID=1e499775fa374811b339a285a27ed321
MANAGERPID=1771
PW_PYTHON_CIPD_INSTALL_DIR=/home/chen/connectedhomeip/.environment/cipd/packages/python
CHROME_DESKTOP=code-url-handler.desktop
_PW_ROSETTA=0
GJS_DEBUG_OUTPUT=stderr
VSCODE_GIT_ASKPASS_EXTRA_ARGS=
LESSCLOSE=/usr/bin/lesspipe %s %s
XDG_SESSION_CLASS=user
TERM=xterm-256color
LESSOPEN=| /usr/bin/lesspipe %s
USER=chen
VSCODE_GIT_IPC_HANDLE=/run/user/1000/vscode-git-b3ddad4e52.sock
DISPLAY=:1
SHLVL=2
QT_IM_MODULE=ibus
PW_PACKAGE_ROOT=/home/chen/connectedhomeip/.environment/packages
_PW_ENVIRONMENT_CONFIG_FILE=/home/chen/connectedhomeip/scripts/setup/environment.json
XDG_RUNTIME_DIR=/run/user/1000
_PW_ACTUAL_ENVIRONMENT_ROOT=/home/chen/connectedhomeip/.environment
PW_BRANDING_BANNER=/home/chen/connectedhomeip/scripts/setup/banner.txt
PW_ARM_CIPD_INSTALL_DIR=/home/chen/connectedhomeip/.environment/cipd/packages/arm
VSCODE_GIT_ASKPASS_MAIN=/usr/share/code/resources/app/extensions/git/dist/askpass-main.js
JOURNAL_STREAM=8:42358
XDG_DATA_DIRS=/usr/share/ubuntu:/usr/share/gnome:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop
GDK_BACKEND=x11
PATH=/home/chen/connectedhomeip/third_party/pigweed/repo/out/host/host_tools:/home/chen/connectedhomeip/.environment/pigweed-venv/bin:/home/chen/connectedhomeip/.environment/cipd/packages/arm/bin:/home/chen/connectedhomeip/.environment/cipd/packages/arm:/home/chen/connectedhomeip/.environment/cipd/packages/zap:/home/chen/connectedhomeip/.environment/cipd/packages/pigweed/bin:/home/chen/connectedhomeip/.environment/cipd/packages/pigweed:/home/chen/connectedhomeip/.environment/cipd:/home/chen/.local/bin:/home/chen/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
GDMSESSION=ubuntu
ORIGINAL_XDG_CURRENT_DESKTOP=ubuntu:GNOME
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
GIO_LAUNCHED_DESKTOP_FILE_PID=4048
PW_PIGWEED_CIPD_INSTALL_DIR=/home/chen/connectedhomeip/.environment/cipd/packages/pigweed
GIO_LAUNCHED_DESKTOP_FILE=/usr/share/applications/code.desktop
TERM_PROGRAM=vscode
_=/usr/bin/env
+ gn gen --check --fail-on-unused-args --root=examples/all-clusters-app/linux/ app-cluster-app '--args=chip_enable_pw_rpc=true import("//with_pw_rpc.gni")'
ERROR Unresolved dependencies.
//:chip-all-clusters-common(//third_party/connectedhomeip/build/toolchain/linux:linux_x64_gcc)
  needs //third_party/connectedhomeip/third_party/pigweed/repo/pw_hdlc:pw_rpc(//third_party/connectedhomeip/build/toolchain/linux:linux_x64_gcc)
```

I have checked the [upstream](https://github.com/google/pigweed/tree/1f12d06f5133a9a471e0551e5f915dff80ee908c/pw_hdlc/public/pw_hdlc), there was no such a file named pw_rpc.h, neither a pw_source_set("pw_rpc") exist in third_party/pigweed/repo/pw_hdlc/BUILD.gn, which leads to a speculation the previous commit included a bug.

bzbarsky-apple, 2024-04-15:
@adis-ikea @tehampson @andy31415 could you take a look, please?

Chapoly1305, 2024-04-15:
> @adis-ikea @tehampson @andy31415 could you take a look, please?

Please review my PR, I have found the cause and fix. 
https://github.com/project-chip/connectedhomeip/pull/32980

closed 2024-04-16 by mergify[bot]