project-chip/connectedhomeip #32981: lighting-app for linux supported clusters & restart after commissioning

dggonz, 2024-04-14:
Hello,

I did not find any other way to ask this, hence an issue. I am trying to play with the lighting-app demo for linux, reading the source code it seems it should support the OnOff cluster, which it does not. On the other hand it does support the Switch cluster.

I am a bit confused as when I search on the internet and read the code in lighting-common everything seems to indicate it should support the OnOff cluster.

Do I need to compile the example in a certain way? or everything is as it should.

Also, after commssioning, if I will the demo and then restart it the chip-lighting-app it is advertised in mDNS as if it was not commissioned. The messages in avahi does not show "T=1" in the txt fields. It shows all attributes (   txt = ["PI=" "PH=33" "CM=1" "D=3840" "T=1" "DN=Test Bulb" "DT=257" "VP=65521+32769"]) as if it was waiting to be commissioned for the first time.

bzbarsky-apple, 2024-04-15:
>  On the other hand it does support the Switch cluster.

@dggonz Which exact files are you looking at here?  `examples/lighting-app/lighting-common/lighting-app.zap` very much supports OnOff and does not support Switch....

dggonz, 2024-04-15:
Hi it does seem to be supported and hence my surprise. If I read the ServerList attribute from the Descriptor Cluster What I get back is:

0x1d 0x1f 0x28 0x2a 0x2b 0x2c 0x30 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x3b 0x3c 0x3e 0x3f 0x40 0x41 

which does not include 0x06 (the OnOff cluster) but it does include 0x3b (Switch cluster) ... which is weird.

bzbarsky-apple, 2024-04-15:
Ah!  That's on endpoint 0 (I have no idea why this thing has a Switch server on endpoint 0).

The actual lighting bits are on endpoint 1, no?

dggonz, 2024-04-17:
Correct, thanks!

closed 2024-04-17 by dggonz