project-chip/connectedhomeip #32989: [BUG] Chip-Cert-Bins docker image not building

raju-apple, 2024-04-15:
### Reproduction steps

Trying to build the chip-cert-bins container : 

docker buildx build --build-arg COMMITHASH=28b2a030f321519cd26e95f2706cbce2890fa5d4 --platform linux/arm64 --tag connectedhomeip/chip-cert-bins:28b2a030f321519cd26e95f2706cbce2890fa5d4 --push .

Looks like its failing at the py wheels section : 

Logs : 
Logs : 

raju@Rajus-MacBook-Pro chip-cert-bins % docker buildx build --build-arg COMMITHASH=28b2a030f321519cd26e95f2706cbce2890fa5d4 --platform linux/arm64 --tag connectedhomeip/chip-cert-bins:28b2a030f321519cd26e95f2706cbce2890fa5d4 --push .

[+] Building 3756.8s (53/53) FINISHED                                                          
 => [internal] booting buildkit                                                           0.5s
 => => starting container buildx_buildkit_mybuild0                                        0.5s
 => [internal] load .dockerignore                                                         0.0s
 => => transferring context: 2B                                                           0.0s
 => [internal] load build definition from Dockerfile                                      0.0s
 => => transferring dockerfile: 13.04kB                                                   0.0s
 => [internal] load metadata for docker.io/library/ubuntu:22.04                           3.5s
 => [auth] library/ubuntu:pull token for registry-1.docker.io                             0.0s
 => [stage-2  1/32] FROM docker.io/library/ubuntu:22.04@sha256:77906da86b60585ce12215807  6.7s
 => => resolve docker.io/library/ubuntu:22.04@sha256:77906da86b60585ce12215807090eb327e7  0.0s
 => => sha256:f4bb4e8dca02be491b4f72d2ef2127a64ce49c48d0d9c0a0fcaffa62 27.36MB / 27.36MB  6.2s
 => => extracting sha256:f4bb4e8dca02be491b4f72d2ef2127a64ce49c48d0d9c0a0fcaffa625067679  0.5s
 => [stage-2  2/32] RUN ln -snf /usr/share/zoneinfo/Etc/UTC /etc/localtime && echo Etc/U  0.1s
 => [chip-build-cert  2/14] RUN case linux/arm64 in     "linux/amd64")     echo "Buildin  0.1s
 => [chip-build-cert  3/14] RUN set -x     && apt-get update     && apt-get upgrade -y  227.9s
 => [stage-2  3/32] RUN apt-get update -y                                                16.7s
 => [stage-2  4/32] RUN apt-get install -y libssl-dev libdbus-1-dev libglib2.0-dev lib  116.7s
 => [stage-2  5/32] WORKDIR /root/                                                        0.1s
 => [chip-build-cert  4/14] RUN case linux/arm64 in     "linux/amd64")     set -x     &&  8.8s 
 => [chip-build-cert  5/14] RUN set -x     && DEBIAN_FRONTEND=noninteractive  apt-get u  42.3s 
 => [chip-build-cert  6/14] RUN set -x     && pip3 install attrs coloredlogs PyGithub p  19.5s 
 => [chip-build-cert  7/14] RUN set -x     && git clone https://gn.googlesource.com/gn   65.6s
 => [chip-build-cert  8/14] RUN set -x     && git clone https://github.com/google/bloa  178.9s
 => [chip-build-cert  9/14] RUN mkdir /root/connectedhomeip                               0.1s
 => [chip-build-cert 10/14] RUN git clone https://github.com/project-chip/connectedhom  163.9s
 => [chip-build-cert 11/14] WORKDIR /root/connectedhomeip/                                0.1s
 => [chip-build-cert 12/14] RUN git checkout 28b2a030f321519cd26e95f2706cbce2890fa5d4     0.8s
 => [chip-build-cert 13/14] RUN ./scripts/checkout_submodules.py --allow-changing-globa  80.5s
 => [chip-build-cert 14/14] RUN bash scripts/bootstrap.sh                               340.7s
 => [chip-build-cert-bins 1/3] RUN git rev-parse HEAD > /root/.sdk-sha-version            0.4s
 => [chip-build-cert-bins 2/3] RUN case linux/arm64 in     "linux/amd64")     set -x   2245.0s
 => [chip-build-cert-bins 3/3] RUN source scripts/activate.sh && scripts/build_python.  237.9s
 => [stage-2  6/32] COPY --from=chip-build-cert-bins /root/.sdk-sha-version .sdk-sha-ver  0.0s
 => [stage-2  7/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-tool  0.4s
 => [stage-2  8/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-shel  0.1s
 => [stage-2  9/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-cert  0.0s
 => [stage-2 10/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-all-  0.3s
 => [stage-2 11/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-all-  0.3s
 => [stage-2 12/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-all-  0.2s
 => [stage-2 13/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-ligh  0.2s
 => [stage-2 14/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-tv-c  0.3s
 => [stage-2 15/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-tv-a  0.3s
 => [stage-2 16/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-brid  0.2s
 => [stage-2 17/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/thermosta  0.2s
 => [stage-2 18/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-ota-  0.2s
 => [stage-2 19/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-ota-  0.2s
 => [stage-2 20/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-lock  0.2s
 => [stage-2 21/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-app1  0.2s
 => [stage-2 22/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/lit-icd-a  0.1s
 => [stage-2 23/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-ener  0.1s
 => [stage-2 24/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/chip-micr  0.1s
 => [stage-2 25/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/python_li  3.1s
 => [stage-2 26/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/out/python_en  1.0s
 => [stage-2 27/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/src/python_te  0.1s
 => [stage-2 28/32] COPY --from=chip-build-cert-bins /root/connectedhomeip/scripts/tests  0.0s
 => [stage-2 29/32] RUN pip install -r /tmp/requirements.txt && rm /tmp/requirements.txt  1.7s
[... 261 lines cut ...]
#0 13.93         File "/usr/lib/python3.10/subprocess.py", line 971, in __init__
#0 13.93           self._execute_child(args, executable, preexec_fn, close_fds,
#0 13.93         File "/usr/lib/python3.10/subprocess.py", line 1863, in _execute_child
#0 13.93           raise child_exception_type(errno_num, err_msg, err_filename)
#0 13.93       FileNotFoundError: [Errno 2] No such file or directory: 'meson'
#0 13.93       [end of output]
#0 13.93   
#0 13.93   note: This error originates from a subprocess, and is likely not a problem with pip.
#0 13.93 error: metadata-generation-failed
#0 13.93 
#0 13.93 × Encountered error while generating package metadata.
#0 13.93 ╰─> See above for output.
#0 13.93 
#0 13.93 note: This is an issue with the package mentioned above, not pip.
#0 13.93 hint: See above for details.
------
Dockerfile:302
--------------------
 300 |     RUN pip install -r /tmp/requirements.txt && rm /tmp/requirements.txt
 301 |     
 302 | >>> RUN pip install --no-cache-dir python_lib/controller/python/chip*.whl
 303 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c pip install --no-cache-dir python_lib/controller/python/chip*.whl" did not complete successfully: exit code: 1
raju@Rajus-MacBook-Pro chip-cert-bins %

raju-apple, 2024-04-16:
@andy31415 @cecille Saw this while trying to search the web for similar issues - https://github.com/mesonbuild/meson/discussions/11205#discussioncomment-4468758

raju-apple, 2024-04-16:
Branch information - 1.3-SVE
SHA on the branch - 28b2a030f321519cd26e95f2706cbce2890fa5d4
Attached the logs to the issue 
[Logs_chip-cert-bins-failure.zip](https://github.com/project-chip/connectedhomeip/files/14999888/Logs_chip-cert-bins-failure.zip)

Error snippet 
```

 => ERROR [stage-2 32/32] RUN pip install --no-cache-dir python_lib/controller/python/chip*.whl                                                                               14.9s 
------                                                                                                                                                                              
 > [stage-2 32/32] RUN pip install --no-cache-dir python_lib/controller/python/chip*.whl:                                                                                           
#0 0.298 Processing ./python_lib/controller/python/chip_clusters-0.0-py3-none-any.whl                                                                                               
#0 0.307 Processing ./python_lib/controller/python/chip_core-0.0-cp37-abi3-linux_aarch64.whl                                                                                        
#0 0.310 Processing ./python_lib/controller/python/chip_repl-0.0-py3-none-any.whl                                                                                                   
#0 0.715 Collecting aenum                                                                                                                                                           
#0 1.410   Downloading aenum-3.1.15-py3-none-any.whl (137 kB)
#0 1.733      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 137.6/137.6 KB 688.7 kB/s eta 0:00:00
#0 1.921 Collecting dacite
#0 2.090   Downloading dacite-1.8.1-py3-none-any.whl (14 kB)
#0 2.282 Collecting ipdb
#0 2.379   Downloading ipdb-0.13.13-py3-none-any.whl (12 kB)
#0 2.548 Collecting ecdsa
#0 2.707   Downloading ecdsa-0.19.0-py2.py3-none-any.whl (149 kB)
#0 2.901      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 149.3/149.3 KB 756.9 kB/s eta 0:00:00
#0 3.023 Collecting coloredlogs
#0 3.046   Downloading coloredlogs-15.0.1-py2.py3-none-any.whl (46 kB)
#0 3.055      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 46.0/46.0 KB 5.9 MB/s eta 0:00:00
#0 3.091 Collecting construct
#0 3.120   Downloading construct-2.10.70-py3-none-any.whl (63 kB)
#0 3.126      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 63.0/63.0 KB 10.8 MB/s eta 0:00:00
#0 3.374 Collecting cryptography
#0 3.396   Downloading cryptography-42.0.5-cp39-abi3-manylinux_2_28_aarch64.whl (4.4 MB)
#0 3.717      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.4/4.4 MB 13.7 MB/s eta 0:00:00
#0 3.748 Collecting deprecation
#0 3.770   Downloading deprecation-2.1.0-py2.py3-none-any.whl (11 kB)
#0 3.840 Collecting rich
#0 3.861   Downloading rich-13.7.1-py3-none-any.whl (240 kB)
#0 3.875      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 240.7/240.7 KB 18.0 MB/s eta 0:00:00
#0 3.877 Requirement already satisfied: pyyaml in /usr/lib/python3/dist-packages (from chip-core==0.0) (5.4.1)
#0 3.935 Collecting ipython!=8.1.0
#0 3.959   Downloading ipython-8.23.0-py3-none-any.whl (814 kB)
#0 4.009      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 814.2/814.2 KB 16.8 MB/s eta 0:00:00
#0 4.075 Collecting ipykernel
#0 4.102   Downloading ipykernel-6.29.4-py3-none-any.whl (117 kB)
#0 4.112      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 117.1/117.1 KB 12.5 MB/s eta 0:00:00
#0 4.133 Collecting dbus-python==1.2.18
#0 4.160   Downloading dbus-python-1.2.18.tar.gz (578 kB)
#0 4.193      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 578.2/578.2 KB 18.1 MB/s eta 0:00:00
#0 4.230   Preparing metadata (setup.py): started
#0 4.330   Preparing metadata (setup.py): finished with status 'done'
#0 4.375 Collecting pygobject
#0 4.401   Downloading pygobject-3.48.2.tar.gz (715 kB)
#0 4.446      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 715.2/715.2 KB 17.5 MB/s eta 0:00:00
#0 4.628   Installing build dependencies: started
#0 13.02   Installing build dependencies: finished with status 'done'
#0 13.03   Getting requirements to build wheel: started
#0 13.09   Getting requirements to build wheel: finished with status 'done'
#0 13.17   Installing backend dependencies: started
#0 14.75   Installing backend dependencies: finished with status 'done'
#0 14.75   Preparing metadata (pyproject.toml): started
#0 14.82   Preparing metadata (pyproject.toml): finished with status 'error'
#0 14.82   error: subprocess-exited-with-error
#0 14.82   
#0 14.82   × Preparing metadata (pyproject.toml) did not run successfully.
#0 14.82   │ exit code: 1
#0 14.82   ╰─> [33 lines of output]
#0 14.82       Traceback (most recent call last):
#0 14.82         File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 156, in prepare_metadata_for_build_wheel
#0 14.82           hook = backend.prepare_metadata_for_build_wheel
#0 14.82       AttributeError: module 'mesonpy' has no attribute 'prepare_metadata_for_build_wheel'
[... 25 lines cut ...]
#0 14.82           self._execute_child(args, executable, preexec_fn, close_fds,
#0 14.82         File "/usr/lib/python3.10/subprocess.py", line 1863, in _execute_child
#0 14.82           raise child_exception_type(errno_num, err_msg, err_filename)
#0 14.82       FileNotFoundError: [Errno 2] No such file or directory: 'meson'
#0 14.82       [end of output]
#0 14.82   
#0 14.82   note: This error originates from a subprocess, and is likely not a problem with pip.
#0 14.82 error: metadata-generation-failed
#0 14.82 
[... cut ...]
#0 14.82 note: This is an issue with the package mentioned above, not pip.
#0 14.82 hint: See above for details.
------
Dockerfile:302
--------------------
 300 |     RUN pip install -r /tmp/requirements.txt && rm /tmp/requirements.txt
 301 |     
 302 | >>> RUN pip install --no-cache-dir python_lib/controller/python/chip*.whl
 303 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c pip install --no-cache-dir python_lib/controller/python/chip*.whl" did not complete successfully: exit code: 1

```

raju-apple, 2024-04-16:
Re-tried the same command on the TOT master : 

Branch - Master
SHA - 90732b256ee1b405d7461f69eb8e25abe700c805

Uploaded the logs from the failure - Chip-cert-bins-failure-master.txt

Failing at a different place but the logs are much more verbose
[Chip-cert-bins-failure-master.txt](https://github.com/project-chip/connectedhomeip/files/15000535/Chip-cert-bins-failure-master.txt)

andy31415, 2024-04-16:
Tested a bit, it seems installing `openjdk-jdk-8` fixes the pip install. Somehow installing the repl whl seems to pull a meson build system which requires java.

Will create a fix PR, however only once I test if we can use ccache to speed up the build of this image as it is super slow right now.

raju-apple, 2024-04-16:
Thanks @andy31415 !

cecille, 2024-04-16:
> Tested a bit, it seems installing `openjdk-8-jdk` fixes the pip install. Somehow installing the repl whl seems to pull a meson build system which requires java.
> 
> Will create a fix PR, however only once I test if we can use ccache to speed up the build of this image as it is super slow right now.

Any idea why that doesn't get pulled in on master?

andy31415, 2024-04-16:
> Any idea why that doesn't get pulled in on master?

I believe these chert-bins images are only built on-demand, so at some point things got broken and we never noticed. We do not directly ask for mason, so it is probably some sub-dependency. Master does build/install python however jdk (and many many other packages) are installed as compile dependencies and all is well. this is how I found it: installed all build prerequisites and saw it work, then started trimming them down to bare minimum.

PR #33015 (2024-04-16): Fix up chip-cert-bins Dockerfile to make it actually build (and slightly faster)
Fixes #32989 

### Changes

- Do not needlesly clear apt update results, as it is reused and a build image will not copy over their contents anyway
- Update compile of bloaty to use ninja, so we do not hardcode -j8
- Install openjdk8 before installing chip whl, since apparently we depend on mason which depends on the jdk
closed 2024-05-14 by andy31415