project-chip/connectedhomeip #33011: [Tooling] CI: Build relevant docker images on a regular basis

cecille, 2024-04-16:
### Reproduction steps / Feature

Definitely not as part of the CI that's run on PR pushes, but if there's a daily run, we should do this there. Or weekly. Any frequency better than once-per-release would be good.

Relevant docker images:
chip-cert-bins
others?

### Platform

other

### Type

Manually tested with SDK

raju-apple, 2024-04-16:
Ideally , if we can create this for every commit and upload it to the docker registry , it would be ideal :)

cecille, 2025-02-11:
Duplicate of #30750