project-chip/connectedhomeip #33026: [BUG] Python controller build is limited with 254 fabrics

Saravana-kr22, 2024-04-17:
### Reproduction steps

I am python as controller to stress test the dut for multi-admin scenarios , Where i found that with current sdk  only 254 fabrics (controllers) can be build with the certificate_authority_manager function in the [matter_testing_support](https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/matter_testing_support.py) file 

if i try to build the fabric beyond 254 i got the error as below:

> Traceback (most recent call last):
  File "/home/ubuntu/connectedhomeip/src/python_testing/Matter_QA/Scripts/ReliabilityScripts/TC_Multi_admin.py", line 69, in build_controller
    th = th_fabric_admin.NewController(thNodeId)
  File "/home/ubuntu/connectedhomeip/no/lib/python3.10/site-packages/chip/FabricAdmin.py", line 101, in NewController
    controller = ChipDeviceCtrl.ChipDeviceController(
  File "/home/ubuntu/connectedhomeip/no/lib/python3.10/site-packages/chip/ChipDeviceCtrl.py", line 1774, in __init__
    ).raise_on_error()
  File "/home/ubuntu/connectedhomeip/no/lib/python3.10/site-packages/chip/native/__init__.py", line 87, in raise_on_error
    raise self.to_exception()
chip.exceptions.ChipStackError: src/credentials/PersistentStorageOpCertStore.cpp:220: CHIP Error 0x00000003: Incorrect state

### Bug prevalence

Whenever i do this 

### GitHub hash of the SDK that was being used

3cb83827fc6d46d66fc34ca0a39b54dff2170c00

### Platform

python

### Anything else?

**Reference log:** 
[cadmin_build_failure_log.txt](https://github.com/project-chip/connectedhomeip/files/15012604/cadmin_build_failure_log.txt)

bzbarsky-apple, 2024-04-17:
Each fabric on a given device requires a unique fabric index, and those are allowed to be in the range 1-254 per spec.

cecille, 2024-04-19:
I think they're removing them from the device, but you can only mint 254 fabrics on the controller because they never get deleted, just shut down.

bzbarsky-apple, 2024-04-26:
Well, why are they not being deleted?  That's broken, clearly, and needs to be fixed.