project-chip/connectedhomeip #33039: [1.3]  OTA speed is very slow, how to get better speed

GnarFang, 2024-04-18:
### Reproduction steps

The OTA-provider sample can work fine, And using the wifi/Ethernet to transmit data. but the speed of it is sadly slow, I have a 77mb *.ota file, and it made alomst 3 hours how can I resolve this situation, thank you guys very much!!

My platform is Linux platform.  OTA Provider and request are running on different Linux platform (Soc i.MX93 or i.MX8MM)

#provider
chip-ota-provider-app --discriminator 22 --secured-device-port 5565 --KVS /tmp/chip_kvs_provider --filepath /home/root/thermostat-app.ota &

#request
chip-ota-requestor-app --discriminator 18 --secured-device-port 5560 --KVS /tmp/chip_kvs_requestor --periodicQueryTimeout 60 --otaDownloadPath /home/root/download.bin

Even if I add  -P 5 to decrease the polling time, but the speed didn't improve.
chip-ota-provider-app --discriminator 22 --secured-device-port 5565 --KVS /tmp/chip_kvs_provider --filepath /home/root/thermostat-app.ota  -P 5&

[1713415243.648261][1468:1468] CHIP:ATM: ReceiveInit
[1713415243.648271][1468:1468] CHIP:ATM:   Proposed Transfer Control: 0x20
[1713415243.648281][1468:1468] CHIP:ATM:   Range Control: 0x0
[1713415243.648291][1468:1468] CHIP:ATM:   Proposed Max Block Size: 10240
[1713415243.648301][1468:1468] CHIP:ATM:   Start Offset: 0x0000000000000000
[1713415243.648311][1468:1468] CHIP:ATM:   Proposed Max Length: 0x0000000000000000
[1713415243.648321][1468:1468] CHIP:ATM:   File Designator Length: 29
[1713415243.648331][1468:1468] CHIP:ATM:   File Designator: /home/root/thermostat-app.ota
[1713415243.650419][1468:1468] CHIP:BDX: OutputEvent type: InitReceived
[1713415243.650440][1468:1468] CHIP:ATM: Sending BDX Message
[1713415243.650451][1468:1468] CHIP:ATM: ReceiveAccept
[1713415243.650460][1468:1468] CHIP:ATM:   Transfer Control: 0x20
[1713415243.650470][1468:1468] CHIP:ATM:   Range Control: 0x0
[1713415243.982741][1468:1468] CHIP:ATM:   Max Block Size: 1024
[1713415243.982754][1468:1468] CHIP:ATM:   Length: 0x0000000000000000

The Max Block Size is limited to 1024.   only 10KB data can be transferred in one second.

### Bug prevalence

Whenvever I do this.

### GitHub hash of the SDK that was being used

1.2 /1.3

### Platform

IMX8

### Type

Core SDK Performance Improvement

KEEPHEADU, 2024-08-28:
I have the same problem. I use efr32mg24.

Trishia, 2024-10-11:
I also have same problem to transfer chip-ota-requestor-app-v2.ota(About 92M) that I made via ota_image_tool.py for testing TC-SU-3.2   as a OTA Provider/Controller DUT. It takes long time over 2hours to download now. 

How can I reduce downloading time? 
Can we change the Max Block Size in chip-ota-requestor-app or Any thing else?

Advise me if you have any solution plz.

bzbarsky-apple, 2024-10-15:
No idea for Linux, sorry.  You could try using `nm` and the like to try to investigate why the binary is as big as it is....

andy31415, 2024-10-21:
Added NXP tag due to usage of `ota_image_tool.py`. @doru91

andy31415, 2024-10-21:
The original bug seems to be for iMX ... so assuming this is general NXP rather than linux?

Trishia, 2024-11-06:
> @Trishia , @GnarFang for the large binary size, could you confirm if you ran `strip` on it?
> 
> I ran a short test on my machine and the size went from 95MB to 4.5MB for otarequestor (although that is probably not the image that should be OTAed)
> 
> ![image](https://private-user-images.githubusercontent.com/1832280/378476258-6dfe118f-abee-4201-8ad3-754142a7b5c2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA4NzU1MTMsIm5iZiI6MTczMDg3NTIxMywicGF0aCI6Ii8xODMyMjgwLzM3ODQ3NjI1OC02ZGZlMTE4Zi1hYmVlLTQyMDEtOGFkMy03NTQxNDJhN2I1YzIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMDZUMDY0MDEzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzBlZmRjZDlmOWYxMzY5NzUxNTdjMTQ1MzUwODUzYWNhNGRhYmE2MTJjNTRlODczMzY5ZWQyZjI0MjI4N2ZiMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.NCjZjVWzA-CbXwh_yNMMR56Skv7fx6O-xBV1FqAfbgc)

Thanks!!! @andy31415 

I didn't know about `strip` before. As a result, I could reduce the size of chip-ota-requestor-app to 1.5M with Optimization options that I mentioned above (https://github.com/project-chip/connectedhomeip/issues/33039#issuecomment-2406611066).
```
ubuntu@raspberrypi:~/connectedhomeip/out$ ls -alh
total 4.3M
drwxrwxr-x  3 ubuntu ubuntu 4.0K Oct 14 16:21 .
drwxrwxr-x 21 ubuntu ubuntu 4.0K Oct 11 15:16 ..
-rwxrwxr-x  1 ubuntu ubuntu 1.5M Oct 14 16:03 chip-ota-requestor-app-v2
-rw-rw-r--  1 ubuntu ubuntu 1.5M Oct 14 16:16 chip-ota-requestor-th-v2.ota
```