project-chip/connectedhomeip #33040: [1.2][chip-tool]chip-tool BLE scan error on the OpenWrt platform

Semi-Bin-Blain, 2024-04-18:
### Reproduction steps

Hi there,
I'm trying to run the `chip-tool` on the OpenWrt， I successfully cross-compiled the Matter1.2 `chip-tool` for the OpenWrt, but it can not work correctly, during the ble-scan step, it always fails with the `BLE scan error `, sometimes it will also show a `Segmentation fault`；
I think the Bluetooth device is ok, I can scan the ble-device with the command: `hcitool lescan`;
I tried to upgrade the `bluez` version to the 5.66 and change my Bluetooth dongle, but both of these do not work;
I also tried to downgrade the `chip-tool` to Matter1.1,  but it can only work correctly with the Matter1.0 device；
Is there any suggestion for this problem?
Thanks in advanced for all the reply!

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

181b0cb14ff007ec912f2ba6627e05dfb066c008

### Platform

other

### Platform Version(s)

OpenWrt 21.02.0+musl 1.2.3

### Type

Platform Issue

### Anything else?

Here is the log:
```log
[root@Hoo_Max_H5EA:/root]#chip-tool pairing ble-thread 2 hex:0e080000000000010000000300001a35060004001fffe0020836e406545bceb6570708fd80339979240dee0510
208b0caaaf8d7db763abaf1311c72879030f4f70656e5468726561642d643235330102d2530410f823b0f8039b2b1e1c5cebcfbd9ef7420c0402a0f7f8  92008135 228 --paa-trust-st
ore-path ./paa-root-certs/
[1713418413.303678][11104:11104] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_tool_kvs
[1713418413.313574][11104:11104] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1713418413.314374][11104:11104] CHIP:DL: ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_kvs
[1713418413.316346][11104:11104] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1713418413.317782][11104:11104] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1713418413.319124][11104:11104] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1713418413.321139][11104:11104] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-CnPjkm)
[1713418413.325152][11104:11104] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1713418413.326016][11104:11104] CHIP:DL: NVS set: chip-counters/reboot-count = 3 (0x3)
[1713418413.334203][11104:11104] CHIP:DL: Failed to get Ethernet interface
[1713418413.337993][11104:11104] CHIP:DL: Failed to reset Ethernet statistic counts
[1713418413.345374][11104:11104] CHIP:DL: Got WiFi interface: wlan0
[1713418413.346437][11104:11104] CHIP:DL: Failed to reset WiFi statistic counts
[1713418413.347232][11104:11104] CHIP:IN: UDP::Init bind&listen port=0
[1713418413.348251][11104:11104] CHIP:IN: UDP::Init bound to port=33302
[1713418413.350767][11104:11104] CHIP:IN: UDP::Init bind&listen port=0
[1713418413.351657][11104:11104] CHIP:IN: UDP::Init bound to port=47976
[1713418413.352368][11104:11104] CHIP:IN: BLEBase::Init - setting/overriding transport
[1713418413.353350][11104:11104] CHIP:IN: TransportMgr initialized
[1713418413.354213][11104:11104] CHIP:FP: Initializing FabricTable from persistent storage
[1713418413.354927][11104:11104] CHIP:TS: Last Known Good Time: 2023-10-14T01:16:48
[1713418413.361317][11104:11104] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x983941C9CCFDCF76, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1713418413.369282][11104:11104] CHIP:ZCL: Using ZAP configuration...
[1713418413.425654][11104:11104] CHIP:CTL: System State Initialized...
[1713418414.049066][11104:11104] CHIP:CTL: Setting attestation nonce to random value
[1713418414.050027][11104:11104] CHIP:CTL: Setting CSR nonce to random value
[1713418414.050931][11104:11104] CHIP:IN: UDP::Init bind&listen port=5550
[1713418414.051994][11104:11104] CHIP:IN: UDP::Init bound to port=5550
[1713418414.052783][11104:11104] CHIP:IN: UDP::Init bind&listen port=5550
[1713418414.053818][11104:11104] CHIP:IN: UDP::Init bound to port=5550
[1713418414.054527][11104:11104] CHIP:IN: TransportMgr initialized
[1713418414.055961][11104:11118] CHIP:DL: CHIP task running
[1713418414.056922][11104:11118] CHIP:DL: HandlePlatformSpecificBLEEvent 32784
[... 223 lines cut ...]
[1713418457.821930][11651:11651] CHIP:CTL: Shutting down the commissioner
[1713418457.822210][11651:11651] CHIP:CTL: Shutting down the controller
[1713418457.822267][11651:11651] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1713418457.822311][11651:11651] CHIP:FP: Forgetting fabric 0x1
[1713418457.822383][11651:11651] CHIP:TS: Pending Last Known Good Time: 2023-10-14T01:16:48
[1713418457.822707][11651:11651] CHIP:TS: Previous Last Known Good Time: 2023-10-14T01:16:48
[1713418457.822757][11651:11651] CHIP:TS: Reverted Last Known Good Time to previous value
[1713418457.822878][11651:11651] CHIP:CTL: Shutting down the commissioner
[1713418457.823006][11651:11651] CHIP:CTL: Shutting down the controller
[1713418457.823055][11651:11651] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1713418457.823475][11651:11651] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1713418457.823634][11651:11651] CHIP:FP: Shutting down FabricTable
[1713418457.823711][11651:11651] CHIP:TS: Pending Last Known Good Time: 2023-10-14T01:16:48
[1713418457.824022][11651:11651] CHIP:TS: Previous Last Known Good Time: 2023-10-14T01:16:48
[1713418457.824070][11651:11651] CHIP:TS: Reverted Last Known Good Time to previous value
[1713418457.828524][11651:11651] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-ElPANN)
[1713418457.831313][11651:11651] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1713418457.832206][11651:11651] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1713418457.833028][11651:11651] CHIP:DL: Inet Layer shutdown
[1713418457.833903][11651:11651] CHIP:DL: BLE shutdown
[1713418457.835878][11651:11660] CHIP:DL: Bluez disconnected
[1713418457.836050][11651:11660] CHIP:DL: Bluez notify CHIPoBluez connection disconnected
[1713418457.885395][11651:11651] CHIP:DL: System Layer shutdown
[1713418457.887119][11651:11651] CHIP:TOO: Run command failure: src/platform/Linux/bluez/ChipDeviceScanner.cpp:173: CHIP Error 0x00000032: Timeout
```

andy31415, 2024-10-24:
I do not believe OpenWRT platform has an official maintainer within the SDK team.

ksperling-apple, 2024-10-24:
> I do not believe OpenWRT platform has an official maintainer within the SDK team.

That's correct, there's currently no direct support for building the full SDK on / for OpenWrt.
We do have a repository that provides an OpenWrt feed with a package definition for building the network-manager example app: https://github.com/project-chip/matter-openwrt This example app should build and work on OpenWrt 23.x. It doesn't build with Bluetooth support currently.

@Semi-Bin-Blain You could try using the Makefiles from the feed repo as a starting point; you also didn't say what version of Matter you're currently trying to build and what kind of configuration you're using. I would start by working off the latest version of the Matter SDK.