project-chip/connectedhomeip #33100: TC-CNET- 4.9 step7,10,12

kagami555, 2024-04-23:
### Feature Area

Other

### Test Case

TC-CNET- 4.9

### Reproduction steps

1.Built chip-lighting-app int linux and chiptool in seperate RPI
2.Brought up DUT using commnad : ./chip-lighting-app --ble-device 0 --wifi
3.Commission the Controller using: ./chip-tool pairing ble-wifi 1 SSID SSID-PWD  20202021 3840 --paa-trust-store-path   /var/paa-root-certs
4.Armed the fail-safe : ./chip-tool generalcommissioning arm-fail-safe 900 1 1 0
5.Remove Network:./chip-tool networkcommissioning remove-network hex:54502D36303030 1 0 --Breadcrumb 1
6.Read LastNetworkId(Failed)./chip-tool networkcommissioning read last-network-id 1 0
7.Reconnect nerwork./chip-tool networkcommissioning connect-network hex:54502D36303030 1 0 --Breadcrumb 2
8.Setting fail-safe ./chip-tool generalcommissioning arm-fail-safe 0 0 1 0
9.Read NerworkId(Failed): ./chip-tool networkcommissioning read networks 1 0
10. Armed the fail-safe : ./chip-tool generalcommissioning arm-fail-safe 900 1 1  0
11. Remove Network(Failed): ./chip-tool generalcommissioning arm-fail-safe 900 1 1 0
[TC-CNET-4.9_TH-Fail.txt](https://github.com/project-chip/connectedhomeip/files/15070744/TC-CNET-4.9_TH-Fail.txt)

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

437c5743ef4eac87ec5f80114d295f71e8a4f0e0

### Platform

raspi

### Anything else?

I seem to have seen a similar problem on the issue(#32663). I don’t know if it has been solved.

lessonzhang321, 2024-05-16:
[image]

lessonzhang321, 2024-05-16:
[image]

s-jain2022, 2024-05-16:
The issue seems to be due to clearing out of the mlastnetworkid, and mlastnetworkingstatus here: https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/network-commissioning/network-commissioning.cpp#L784 and line 786,
Line 785 could still be ok.
These values should not be cleared out on remove network command.
Spec section 11.8.6.6. and 11.8.6.7., provide description for these attributes

Wondering if removing those two lines out will help resolve and not impact other aspects.

s-jain2022, 2024-05-16:
Note This impacts TC-CNET- 4.10 as well

kagami555, 2024-05-17:
> The issue seems to be due to clearing out of the mlastnetworkid, and mlastnetworkingstatus here: https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/network-commissioning/network-commissioning.cpp#L784 and line 786, Line 785 could still be ok.
> 
> SetLastNetworkId(ByteSpan{}); and SetLastNetworkingStatusValue(NullNullable);
> 
> These values should not be cleared out on remove network command. Spec section 11.8.6.6. and 11.8.6.7., provide description for these attributes
> 
> Wondering if removing those two lines out will help resolve and not impact other aspects.

@s-jain2022  Thank you very much for your suggestion. However, I have tried the above method before. It only solves the problem of being unable to read the LastNetworkId in step 6, but there are still problems in subsequent steps 9 and 11.

cjandhyala, 2024-08-19:
@kagami555  is this still an issue ?

kagami555, 2024-08-20:
Hi:
     Since there is no plan to update in the near future, I will use the old version for the function for the time being. Your update will be added in the subsequent version iteration. At that time, I will test it again and synchronize the results to you.

***@***.***
 
From: cjandhyala
Date: 2024-08-20 06:01
To: project-chip/connectedhomeip
CC: kagami555; Mention
Subject: Re: [project-chip/connectedhomeip] TC-CNET- 4.9 step7,10,12 (Issue #33100)
@kagami555 is this still an issue ?