project-chip/connectedhomeip #33111: [1.3] All-Cluster-App Linux Binding is not working

Chapoly1305, 2024-04-23:
### Reproduction steps

We are currently trying to write automation to test the feature of binding. We have used example/all-cluster-app/linux, and referring to the articles listed below,
- https://github.com/project-chip/connectedhomeip/tree/master/examples/bridge-app/asr
- https://docs.silabs.com/d/matter-wifi-getting-started/0.1/04-light-switch-step-by-step-example

We were able to write acl to device act as the light, bind the switch and light (chip-tool displayed success), but when the switch turned on or off, but the light status has not changed accordingly. 

The command we used to write ACL and Binding, 1111 is the light role, 2222 is the switch role. 
```
accesscontrol write acl '[{"fabricIndex": 0, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 0, "privilege": 1, "authMode": 2, "subjects": [2222], "targets": null}]' 1111 0

binding write binding '[ { "fabricIndex" : 1 , "node" :1111,"endpoint" :1,"cluster" :6} ]' 2222 1
```

The command used to trigger the status of onoff, 
```
onoff on 2222 1
onoff off 2222 1  
```

We would like to know if the feature on linux version of all-cluster-app has been implemented, if there are other apps that on linux is in knowledge working, thanks. 

[chiptool.log](https://github.com/project-chip/connectedhomeip/files/15079087/chiptool.log)

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

5bb5c9e23d532cea40476fc0bd1d3008522792ba

### Platform

core

### Type

Common Cluster Logic, Core SDK Interopability Issue

bzbarsky-apple, 2024-04-24:
all-clusters-app does not have a client switch cluster, yes?  So why would this be expected to work?

lighting-app does have a switch client, I believe.

Chapoly1305, 2024-04-24:
@bzbarsky-apple 
Thank you for your response. As code indicated the all-cluster-all does have the client role of one-off cluster. 

https://github.com/project-chip/connectedhomeip/blob/e1dd2798bdd7d6c9c0f1cb7f9e3d319b907f21c3/examples/all-clusters-app/all-clusters-common/all-clusters-app.zap#L6603 

As well as the decoded in zap, 
[image]

We expect this app can act a a light/switch that forwards the action to another lightbulb, it should not have to be a switch client to one-off server, should it? And lighting-app does not have a one-off or switch client. We respectfully ask for more information.

bzbarsky-apple, 2024-04-25:
Ah, interesting.  It does have a client, and there's some binding handling code.  I have no idea what it's supposed to be able to do, though...

agatah2333, 2024-05-03:
And one more question about the binding: If the binding function succeeds, like when the switch is on and the light turns on, is the 'light on' message sent by the switch directly or through the controller?

bzbarsky-apple, 2024-05-03:
Binding is for direct communication.

bhargavnagotha, 2025-01-31:
facing the same issue.
i have run my bridge app and commissioned with nodeId : 1
Then run my chip-all-cluster-app and commissioned it with nodeId:2 

After that I write ACL and bind the cluster-app device with the bridge app:
ACL:
accesscontrol write acl '[{"fabricIndex": 0, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 0, "privilege": 1, "authMode": 2, "subjects": [2], "targets": null}]' 1 0

binding write binding '[ { "fabricIndex" : 1 , "node" :1,"endpoint" :1,"cluster" :6} ]' 21

Binding:
./chip-tool binding write binding '[{"fabricIndex": 1, "node":1, "endpoint":3, "cluster":6}]' 2 1

Toggle:
./chip-tool onoff toggle 1 3

Now, when I toggle to endpoint 3 of the bridge app, it should also trigger the endpoint of my all-cluster app, right?
But the endpoint of my all-cluster app is not getting triggered when I trigger endpoint 3 of the bridge app