project-chip/connectedhomeip #33131: [1.1] Error Deprecated40 When use Bridge Matter with Cluster ColorControl

ngocuong86, 2024-04-24:
### Reproduction steps

I want to use Bridge Matter to use for device nonMatter. When I fake a device with cluster ColorControl to connect to Google Home and chip_tool but don't control device. The device can be uploaded to Google but cannot control LevelControl and ColorControl functions. Still use the On/Off feature. When I put it into chip_tool I see the Attribute data has an error Deprecated40 in all Attribute of ColorControl and Attribute ClusterRevision and FeatureMap of LevelControl.
**I define ATTRIBUTE:**
`DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(colorExtendAttrs)
DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::CurrentHue::Id, INT8U, 1, 0), /* Current Hue */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::CurrentSaturation::Id, INT8U, 1, 0), /* Current Saturation */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::RemainingTime::Id, INT16U, 2, 0), /* Remaining Time */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::CurrentX::Id, INT16U, 2, 0), /* Current X */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::CurrentY::Id, INT16U, 2, 0), /* Current Y */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::ColorTemperatureMireds::Id, INT16U, 2, 0), /* Color Temperature */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::ColorMode::Id, ENUM8, 1, 0), /* Color Mode */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::Options::Id, BITMAP8, 1, 0), /* Options */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::NumberOfPrimaries::Id, INT8U, 1, 0), /* Number Of Primaries */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::EnhancedCurrentHue::Id, INT16U, 2, 0), /* Enhanced Current Hue */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::EnhancedColorMode::Id, ENUM8, 1, 0), /* Enhanced Color Mode */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::ColorLoopActive::Id, INT8U, 1, 0), /* Color Loop Active */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::ColorLoopDirection::Id, INT8U, 1, 0), /* Color Loop Direction */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::ColorLoopTime::Id, INT16U, 2, 0), /* Color Loop Time */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::ColorCapabilities::Id, BITMAP16, 2, 0), /* Color Capabilities */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::ColorTempPhysicalMinMireds::Id, INT16U, 2, 0), /* Color Temp Physical Min */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::ColorTempPhysicalMaxMireds::Id, INT16U, 2, 0), /* Color Temp Physical Max */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::CoupleColorTempToLevelMinMireds::Id, INT16U, 2, 0), /* Couple Color Temp To Level Min Mireds */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::StartUpColorTemperatureMireds::Id, INT16U, 2, 0), /* Start Up Color Temperature */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::ClusterRevision::Id, INT16U, 2, 0), /* Cluster Revision */
    DECLARE_DYNAMIC_ATTRIBUTE(ColorControl::Attributes::FeatureMap::Id, BITMAP32, 4, 0), /* Feature Map */
DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();
DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(levelControlAttrs)
DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::CurrentLevel::Id, INT8U, 1, 0), /* Current Level */
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::RemainingTime::Id, INT16U, 2, 0), /* Remaining Time */
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::MinLevel::Id, INT8U, 1, 0), /* Min Level */
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::Options::Id, BITMAP8, 1, 0), /* Options */
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::StartUpCurrentLevel::Id, INT8U, 1, 0), /* Start Up Current Level */
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::CurrentFrequency::Id, INT16U, 2, 0), /* Current Frequency */
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::MinFrequency::Id, INT16U, 2, 0), /* Min Frequency */
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::MaxFrequency::Id, INT16U, 2, 0), /* Max Frequency */
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::OnOffTransitionTime::Id, INT16U, 2, 0), /* On/Off Transition Time */
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::OnLevel::Id, INT8U, 1, 0), /* On Level */
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::OnTransitionTime::Id, INT16U, 2, 0), /* On Transition Time */
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::OffTransitionTime::Id, INT16U, 2, 0), /* Off Transition Time */
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::DefaultMoveRate::Id, INT16U, 2, 0), /* Default Move Rate */
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::ClusterRevision::Id, INT16U, 2, 0), /* Cluster Revision */
    DECLARE_DYNAMIC_ATTRIBUTE(LevelControl::Attributes::FeatureMap::Id, BITMAP32, 4, 0), /* Feature Map */
DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(scenesManagerAttrs)
DECLARE_DYNAMIC_ATTRIBUTE(Scenes::Attributes::SceneCount::Id, INT16U, 2, 0), /* Scene Table Size */
    DECLARE_DYNAMIC_ATTRIBUTE(Scenes::Attributes::FeatureMap::Id, BITMAP32, 4, 0), /* Feature Map */
    DECLARE_DYNAMIC_ATTRIBUTE(Scenes::Attributes::ClusterRevision::Id, INT16U, 2, 0), /* Cluster Revision */
    DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(groupsManagerAttrs)
DECLARE_DYNAMIC_ATTRIBUTE(Groups::Attributes::NameSupport::Id, BITMAP8, 1, 0), /* Name Support */
    DECLARE_DYNAMIC_ATTRIBUTE(Groups::Attributes::ClusterRevision::Id, INT16U, 2, 0), /* Cluster Revision */
    DECLARE_DYNAMIC_ATTRIBUTE(Groups::Attributes::FeatureMap::Id, BITMAP32, 4, 0), /* Feature Map */
    DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();

DECLARE_DYNAMIC_ATTRIBUTE_LIST_BEGIN(IdentifyAttrs)
DECLARE_DYNAMIC_ATTRIBUTE(Identify::Attributes::IdentifyTime::Id, INT16U, 2, 0), /* Identify Time */
    DECLARE_DYNAMIC_ATTRIBUTE(Identify::Attributes::IdentifyType::Id, ENUM8, 1, 0), /* Identify Type */
    DECLARE_DYNAMIC_ATTRIBUTE(Identify::Attributes::FeatureMap::Id, BITMAP32, 4, 0), /* Feature Map */
    DECLARE_DYNAMIC_ATTRIBUTE(Identify::Attributes::ClusterRevision::Id, INT16U, 2, 0), /* Cluster Revision */
    DECLARE_DYNAMIC_ATTRIBUTE_LIST_END();`
 
**I define Command:**
`constexpr CommandId onOffIncomingCommands[] = {
    app::Clusters::OnOff::Commands::Off::Id,
    app::Clusters::OnOff::Commands::On::Id,
    app::Clusters::OnOff::Commands::Toggle::Id,
    app::Clusters::OnOff::Commands::OffWithEffect::Id,
[... 326 lines cut ...]
    else
    {
        return EMBER_ZCL_STATUS_FAILURE;
    }
}`

[
[main.txt](https://github.com/project-chip/connectedhomeip/files/15087954/main.txt)
](url)

### Bug prevalence

every

### GitHub hash of the SDK that was being used

v1.1 branch

### Platform

vscode, other

### Type

Test Improvement

bzbarsky-apple, 2024-04-24:
@ngocuong86 That comment is entirely unreadable, sorry.  Can you please attach, not paste, large chunks of code, and attach, not paste, the logs you are getting?

ngocuong86, 2024-04-24:
> @ngocuong86 That comment is entirely unreadable, sorry. Can you please attach, not paste, large chunks of code, and attach, not paste, the logs you are getting?

Sorry, I have attached the main.txt file above. You can read the code in there.
[main.txt](https://github.com/project-chip/connectedhomeip/files/15096014/main.txt)
 And The log I getting that: "Status IB: 0x40 "

ngocuong86, 2024-04-25:
@bzbarsky-apple After I update to v1.3-branch. I have the error: status = 0x4c (Unallocated).
That is my log of chiptool: 
[logChiptool.txt](https://github.com/project-chip/connectedhomeip/files/15108252/logChiptool.txt)
Do you have any updates for that error?

bzbarsky-apple, 2024-04-25:
@ngocuong86 So looking at your `HandleReadColorControlAttribute` function... if `(attributeId == ColorControl::Attributes::CurrentHue::Id) && (maxReadLength == 1)` then it never hits a return statement.  Presumably your compiler warns about this but does not error, and it's returning whatever value happens to be on the stack (which in your case seems to be 0x4C).