project-chip/connectedhomeip #33147: [Android] Fail to commission due to PASESession timed out

franmontiel, 2024-04-24:
### Reproduction steps

In Android, when commissioning Matter devices we are facing a recurrent issue that make the commissioning to fail due to a timeout when sending PBKDF param request.

When calling `chipDeviceController.establishPaseConnection` method we see the following error happening randomly (but quite frequently). 
```
Sent PBKDF param request
Retransmitting MessageCounter:147474955 on exchange 11302i Send Cnt 1
Retransmitting MessageCounter:147474955 on exchange 11302i Send Cnt 2
Retransmitting MessageCounter:147474955 on exchange 11302i Send Cnt 3
Retransmitting MessageCounter:147474955 on exchange 11302i Send Cnt 4
Failed to Send CHIP MessageCounter:147474955 on exchange 11302i sendCount: 4 max retries: 4
PASESession timed out while waiting for a response from the peer. Expected message type was 33
```
This is happening in all Android versions we tested and with different builds of the CHIP library (v1.2.0.1, and multiple builds taking current HEAD at the moment on main branch).

### Platform

android, core (please add to version below)

### Type

Manually tested with SDK

law-ko, 2024-07-07:
We are facing this issue as well. Following [Google's sample app for Matter ](https://github.com/google-home/sample-apps-for-matter-android)

KhushbuShah25, 2025-03-21:
Still facing same issue with CHIP library 1.4.0.0

With 1.4.0.0 CHIP libs, there is another issue ICD Client storage so I have taken latest commit from master branch (commit id : `607fdcad2c784acb7f64912ed156f6bcba9ac233`). But still facing PASESession timed out issue with latest lib.

Also observed that, app works fine with some network and not working with some network.
Seems some issue related to network / mDNS.

Did you find any solution for this issue ?