project-chip/connectedhomeip #33187: [1.2] Android - OpenBasicCommissioning Fail ErrorCode : 1413

swap9391, 2024-04-26:
### Reproduction steps

I am trying to open commissioning window in Android app by calling - deviceController.openPairingWindowCallback. 

getting error  as - 

Received Command Response Status for Endpoint=0 Cluster=0x0000_003C Command=0x0000_0001 Status=0x85
Failed to open pairing window on the device. Status IM Error 0x00000585: General error: 0x85
OpenBasicCommissioning Fail! 
Device ID : 1
ErrorCode : 1413
Delete AndroidCommissioningWindowOpener

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

1.2

### Platform

android

### Platform Version(s)

1.2

### Type

Common Cluster Logic

bzbarsky-apple, 2024-04-26:
0x85 is INVALID_COMMAND.

You didn't say what arguments you are passing to `openPairingWindowCallback`, but chances are your duration is either too long or too short (spec requires it to be 3-15 minutes).

swap9391, 2024-04-28:
> 0x85 is INVALID_COMMAND.
> 
> You didn't say what arguments you are passing to `openPairingWindowCallback`, but chances are your duration is either too long or too short (spec requires it to be 3-15 minutes).

Ohh I think duration could be the issue I kept 6000

```
fun sendBasicCommissioningCommandClick(devicePointer: Long) {
        val deviceController= ChipClient.getDeviceController(mContext)
        deviceController.openPairingWindowCallback(
            devicePointer,
            6000,
            object : OpenCommissioningCallback {
                override fun onError(status: Int, deviceId: Long) {
                    LoggerUtils.debug("OpenBasicCommissioning Fail! \nDevice ID : $deviceId\nErrorCode : $status")
                }

                override fun onSuccess(deviceId: Long, manualPairingCode: String?, qrCode: String?) {
                    LoggerUtils.debug("OpenBasicCommissioning Success! \n Node ID: $deviceId")
                }
            }
        )
    }

```

swap9391, 2024-04-28:
What is the meaning of duration, is it timeout error limit?

swap9391, 2024-04-29:
> 0x85 is INVALID_COMMAND.
> 
> You didn't say what arguments you are passing to `openPairingWindowCallback`, but chances are your duration is either too long or too short (spec requires it to be 3-15 minutes).

Same issue even if duration is increased , is there any add on step before it. I have just commissioned the device and trying to open commissioning window. 

```

scope.launch(Dispatchers.IO) {
            val deviceController = ChipClient.getDeviceController(mContext)
            deviceController.openPairingWindowCallback(
                devicePointer,
                420000,
                object : OpenCommissioningCallback {
                    override fun onError(status: Int, deviceId: Long) {
                        LoggerUtils.debug("OpenBasicCommissioning Fail! \nDevice ID : $deviceId\nErrorCode : $status")
                    }

                    override fun onSuccess(
                        deviceId: Long,
                        manualPairingCode: String?,
                        qrCode: String?
                    ) {
                        LoggerUtils.debug("OpenBasicCommissioning Success! \n Node ID: $deviceId")
                    }
                }
            )
        }
```

bzbarsky-apple, 2024-04-30:
duration is the maximum length of time the commissioning window will be open.  

I suspect duration is in seconds here, so valid values are 180 to 900.  But you should check with the API documentation...

AlbertLarry, 2024-05-06:
@swap9391 
May I ask how this problem was ultimately resolved
I also encountered the same problem

my code
`ChipClient.getDeviceController(context)
            .openPairingWindowWithPINCallback(
                devicePtr,
                5*60*1000,
                1001L,
                502,
                3196352,
                object : OpenCommissioningCallback {
                    override fun onError(status: Int, deviceId: Long) {
                        Log.i(TAG, "onError: $status  deviceId:$deviceId")
                    }

                    override fun onSuccess(
                        deviceId: Long,
                        manualPairingCode: String?,
                        qrCode: String?
                    ) {
                        Log.i(
                            TAG,
                            "onSuccess: $deviceId  manualPairingCode:$manualPairingCode  qrCode:$qrCode"
                        )
                    }
                })`

swap9391, 2024-05-17:
@AlbertLarry Yes it is resolved, Problem was same as you I was passing wrong duration... 
Make it 
`ChipClient.getDeviceController(context) .openPairingWindowWithPINCallback( devicePtr, 180, 1000L, 502, 3196352L, object : OpenCommissioningCallback { `

closed 2024-10-08 by yunhanw-google