project-chip/connectedhomeip #33212: [Build] Building Android chip-tools from scripts fails

KhushbuShah25, 2024-04-29:
### Build issue(s)

1. Clone latest code from the repo. Updated submodules recursively.
2. Followed below steps as mentioned in [android_building.md](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md)

```
export ANDROID_HOME=/Users/myuser/Library/Android/sdk
export ANDROID_NDK_HOME=/Users/myuser/Library/Android/sdk/ndk/27.0.11718014
export JAVA_HOME=/Applications/Android\ Studio.app/Contents/jbr/Contents/Home

source scripts/bootstrap.sh
```

[image]

3. After that, run the following command
`./scripts/build/build_examples.py --target android-arm64-chip-tool build`

Facing below error while running above command.

```
2024-04-29 12:46:48 INFO    Building targets: android-arm64-chip-tool
2024-04-29 12:46:48 INFO    Preparing builder 'android-arm64-chip-tool'
2024-04-29 12:46:48 INFO    Generating /Users/myuser/connectedhomeip/out/android-arm64-chip-tool
2024-04-29 12:46:48 INFO    Setting up Android deps through Gradle
2024-04-29 12:46:48 INFO    Starting a Gradle Daemon (subsequent builds will be faster)
2024-04-29 12:46:50 INFO    > Task :buildSrc:compileJava NO-SOURCE
2024-04-29 12:46:50 INFO    > Task :buildSrc:compileGroovy
2024-04-29 12:46:50 INFO    > Task :buildSrc:processResources NO-SOURCE
2024-04-29 12:46:50 INFO    > Task :buildSrc:classes
2024-04-29 12:46:50 INFO    > Task :buildSrc:jar
2024-04-29 12:46:50 INFO    > Task :buildSrc:assemble
2024-04-29 12:46:50 INFO    > Task :buildSrc:compileTestJava NO-SOURCE
2024-04-29 12:46:50 INFO    > Task :buildSrc:compileTestGroovy NO-SOURCE
2024-04-29 12:46:50 INFO    > Task :buildSrc:processTestResources NO-SOURCE
2024-04-29 12:46:50 INFO    > Task :buildSrc:testClasses UP-TO-DATE
2024-04-29 12:46:50 INFO    > Task :buildSrc:test NO-SOURCE
2024-04-29 12:46:50 INFO    > Task :buildSrc:check UP-TO-DATE
2024-04-29 12:46:50 INFO    > Task :buildSrc:build
2024-04-29 12:46:51 INFO    > Task :copyArtifacts
2024-04-29 12:46:51 INFO    > Task :setUpAndroidDeps
2024-04-29 12:46:51 INFO    
2024-04-29 12:46:51 INFO    BUILD SUCCESSFUL in 2s
2024-04-29 12:46:51 INFO    4 actionable tasks: 4 executed
2024-04-29 12:46:51 INFO    Command ['python3', 'third_party/android_deps/set_up_android_deps.py'] completed
2024-04-29 12:46:51 INFO    Setting up Java deps
2024-04-29 12:47:06 INFO    Command ['third_party/java_deps/set_up_java_deps.sh'] completed
2024-04-29 12:47:06 INFO    Generating android-arm64-chip-tool
2024-04-29 12:47:06 INFO    Done. Made 6744 targets from 360 files in 352ms
2024-04-29 12:47:06 INFO    Command ['gn', 'gen', '--check', '--fail-on-unused-args', '/Users/myuser/connectedhomeip/out/android-arm64-chip-tool', '--args=target_os="android" target_cpu="arm64" android_ndk_root="/Users/myuser/Library/Android/sdk/ndk/27.0.11718014" android_sdk_root="/Users/myuser/Library/Android/sdk" chip_build_controller_dynamic_server=true '] completed
2024-04-29 12:47:06 INFO    Accepting NDK licenses @ cmdline-tools
2024-04-29 12:47:08 WARNING Warning: Errors during XML parse:
2024-04-29 12:47:08 WARNING Warning: Additionally, the fallback loader failed to parse the XML.
2024-04-29 12:47:08 INFO    Command ['bash', '-c', 'yes | /Users/myuser/Library/Android/sdk/cmdline-tools/latest/bin/sdkmanager --licenses >/dev/null'] completed
2024-04-29 12:47:08 INFO    Building /Users/myuser/connectedhomeip/out/android-arm64-chip-tool
2024-04-29 12:47:08 INFO    Building JNI android-arm64-chip-tool
2024-04-29 12:47:08 INFO    ninja: Entering directory `/Users/myuser/connectedhomeip/out/android-arm64-chip-tool'
2024-04-29 12:47:08 WARNING ninja: error: '/Users/myuser/Library/Android/sdk/ndk/27.0.11718014/sources/cxx-stl/llvm-libc++/libs/arm64-v8a/libc++_shared.so', needed by 'lib/jni/arm64-v8a/libc++_shared.so', missing and no known rule to make it
Traceback (most recent call last):
  File "/Users/myuser/connectedhomeip/./scripts/build/build_examples.py", line 230, in <module>
    main(auto_envvar_prefix='CHIP')
  File "/Users/myuser/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/Users/myuser/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/Users/myuser/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1688, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
  File "/Users/myuser/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/myuser/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/Users/myuser/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/myuser/connectedhomeip/./scripts/build/build_examples.py", line 220, in cmd_build
    context.obj.Build()
  File "/Users/myuser/connectedhomeip/scripts/build/build/__init__.py", line 71, in Build
    builder.build()
  File "/Users/myuser/connectedhomeip/scripts/build/builders/builder.py", line 99, in build
    self._build()
  File "/Users/myuser/connectedhomeip/scripts/build/builders/android.py", line 466, in _build
    self._Execute(
  File "/Users/myuser/connectedhomeip/scripts/build/builders/builder.py", line 104, in _Execute
    self._runner.Run(cmdarray, title=title)
  File "/Users/myuser/connectedhomeip/scripts/build/runner/shell.py", line 73, in Run
    raise Exception('Command %r failed: %d' % (cmd, code))
Exception: Command ['ninja', '-C', '/Users/myuser/connectedhomeip/out/android-arm64-chip-tool'] failed: 1

```

Seems facing error due to latest NDK as per this line. But new there are no STL libs are available in latest NDK.
**ninja: error: '/Users/myuser/Library/Android/sdk/ndk/27.0.11718014/sources/cxx-stl/llvm-libc++/libs/arm64-v8a/libc++_shared.so', needed by 'lib/jni/arm64-v8a/libc++_shared.so', missing and no known rule to make it**

### Platform

android

joonhaengHeo, 2024-04-29:
`error: '/Users/myuser/Library/Android/sdk/ndk/27.0.11718014/sources/cxx-stl/llvm-libc++/libs/arm64-v8a/libc++_shared.so', needed by 'lib/jni/arm64-v8a/libc++_shared.so', missing and no known rule to make it`

Android build is required SDK 26 & NDK **23.2.8568313**.
Please check your build environment.

KhushbuShah25, 2024-04-30:
Android SDK 26 is installed. But when I installed NDK from latest Android studio, it installs latest ndk 27.0.11718014.
There is no option in it to get older ndk version. 
I have downloaded older ndk version from [this link](https://github.com/android/ndk/wiki/Unsupported-Downloads) and tried  again.

Facing below error for 1.2.0.1:

```
2024-04-30 15:24:53 INFO    Building targets: android-arm64-chip-tool
2024-04-30 15:24:53 INFO    Preparing builder 'android-arm64-chip-tool'
2024-04-30 15:24:53 INFO    Generating /Users/myuser/connectedhomeip/out/android-arm64-chip-tool
2024-04-30 15:24:53 INFO    Setting up Android deps through Gradle
2024-04-30 15:24:54 INFO    > Task :buildSrc:compileJava NO-SOURCE
2024-04-30 15:24:55 INFO    > Task :buildSrc:compileGroovy
2024-04-30 15:24:55 INFO    > Task :buildSrc:processResources NO-SOURCE
2024-04-30 15:24:55 INFO    > Task :buildSrc:classes
2024-04-30 15:24:55 INFO    > Task :buildSrc:jar
2024-04-30 15:24:55 INFO    > Task :buildSrc:assemble
2024-04-30 15:24:55 INFO    > Task :buildSrc:compileTestJava NO-SOURCE
2024-04-30 15:24:55 INFO    > Task :buildSrc:compileTestGroovy NO-SOURCE
2024-04-30 15:24:55 INFO    > Task :buildSrc:processTestResources NO-SOURCE
2024-04-30 15:24:55 INFO    > Task :buildSrc:testClasses UP-TO-DATE
2024-04-30 15:24:55 INFO    > Task :buildSrc:test NO-SOURCE
2024-04-30 15:24:55 INFO    > Task :buildSrc:check UP-TO-DATE
2024-04-30 15:24:55 INFO    > Task :buildSrc:build
2024-04-30 15:24:55 INFO    > Task :copyArtifacts
2024-04-30 15:24:55 INFO    > Task :setUpAndroidDeps
2024-04-30 15:24:55 INFO    
2024-04-30 15:24:55 INFO    BUILD SUCCESSFUL in 1s
2024-04-30 15:24:55 INFO    4 actionable tasks: 4 executed
2024-04-30 15:24:55 INFO    Command ['python3', 'third_party/android_deps/set_up_android_deps.py'] completed
2024-04-30 15:24:55 INFO    Setting up Java deps
2024-04-30 15:25:11 INFO    Command ['third_party/java_deps/set_up_java_deps.sh'] completed
2024-04-30 15:25:11 INFO    Generating android-arm64-chip-tool
2024-04-30 15:25:11 INFO    Done. Made 6574 targets from 343 files in 280ms
2024-04-30 15:25:11 INFO    Command ['gn', 'gen', '--check', '--fail-on-unused-args', '/Users/myuser/connectedhomeip/out/android-arm64-chip-tool', '--args=target_os="android" target_cpu="arm64" android_ndk_root="/Users/myuser/Library/Android/sdk/ndk/23.2.8568313" android_sdk_root="/Users/myuser/Library/Android/sdk" chip_build_controller_dynamic_server=true '] completed
2024-04-30 15:25:11 INFO    Accepting NDK licenses @ cmdline-tools
2024-04-30 15:25:12 WARNING Warning: Errors during XML parse:
2024-04-30 15:25:12 WARNING Warning: Additionally, the fallback loader failed to parse the XML.
2024-04-30 15:25:12 INFO    Command ['bash', '-c', 'yes | /Users/myuser/Library/Android/sdk/cmdline-tools/latest/bin/sdkmanager --licenses >/dev/null'] completed
2024-04-30 15:25:12 INFO    Building /Users/myuser/connectedhomeip/out/android-arm64-chip-tool
2024-04-30 15:25:12 INFO    Building JNI android-arm64-chip-tool
2024-04-30 15:25:12 INFO    ninja: Entering directory `/Users/myuser/connectedhomeip/out/android-arm64-chip-tool'
2024-04-30 15:25:12 INFO    [1/1571] stamp obj/build/chip/java/tests/java_build_test._metadata_path_list.txt.stamp
2024-04-30 15:25:12 INFO    [2/1571] cp /Users/myuser/Library/Android/sdk/ndk/23.2.8568313/sources/cxx-stl/llvm-libc++/libs/arm64-v8a/libc++_shared.so lib/jni/arm64-v8a/libc++_shared.so
2024-04-30 15:25:12 INFO    [3/1571] stamp obj/src/controller/data_model/data_model_codegen_codegen._metadata_path_list.txt.stamp
2024-04-30 15:25:12 INFO    [4/1571] stamp obj/src/controller/data_model/cluster-tlv-metadata_codegen._metadata_path_list.txt.stamp
2024-04-30 15:25:12 INFO    [5/1571] stamp obj/src/controller/data_model/data_model_zapgen_files._metadata_path_list.txt.stamp
2024-04-30 15:25:12 INFO    [6/1571] stamp obj/src/controller/data_model/java-jni-generate_codegen._metadata_path_list.txt.stamp
2024-04-30 15:25:12 INFO    [7/1571] stamp obj/src/app/icd/observer.stamp
2024-04-30 15:25:12 INFO    [8/1571] stamp obj/src/controller/data_model/data_model_zapgen_zap_pregen._metadata_path_list.txt.stamp
2024-04-30 15:25:12 INFO    [9/1571] stamp obj/build/chip/java/shared_cpplib.stamp
2024-04-30 15:25:12 INFO    [10/1571] ACTION //build/chip/java/tests:child_library_config(//build/toolchain/android:android_arm64)
2024-04-30 15:25:12 INFO    [11/1571] ACTION //build/chip/java/tests:child_prebuilt_config(//build/toolchain/android:android_arm64)
2024-04-30 15:25:12 INFO    [12/1571] ACTION //src/app:gen_app_buildconfig(//build/toolchain/android:android_arm64)
2024-04-30 15:25:12 INFO    [13/1571] stamp obj/build/chip/java/tests/child_library_config.stamp
2024-04-30 15:25:12 INFO    [14/1571] ACTION //src/app/server/java:android_config(//build/toolchain/android:android_arm64)
2024-04-30 15:25:12 INFO    [15/1571] ACTION //src/ble:gen_ble_buildconfig(//build/toolchain/android:android_arm64)
2024-04-30 15:25:12 INFO    [16/1571] stamp obj/build/chip/java/tests/child_prebuilt_config.stamp
2024-04-30 15:25:12 INFO    [17/1571] ACTION //src/controller/java:android_config(//build/toolchain/android:android_arm64)
2024-04-30 15:25:12 INFO    [18/1571] stamp obj/src/app/gen_app_buildconfig.stamp
2024-04-30 15:25:12 INFO    [19/1571] ACTION //build/chip/java/tests:grandchild_library_config(//build/toolchain/android:android_arm64)
2024-04-30 15:25:12 INFO    [20/1571] stamp obj/src/ble/gen_ble_buildconfig.stamp
[... cut ...]
    builder.generate()
  File "/Users/myuser/connectedhomeip/scripts/build/builders/android.py", line 342, in generate
    self._Execute(
  File "/Users/myuser/connectedhomeip/scripts/build/builders/builder.py", line 104, in _Execute
    self._runner.Run(cmdarray, title=title)
  File "/Users/myuser/connectedhomeip/scripts/build/runner/shell.py", line 73, in Run
    raise Exception('Command %r failed: %d' % (cmd, code))
Exception: Command ['python3', 'third_party/android_deps/set_up_android_deps.py'] failed: 1

```

KhushbuShah25, 2024-04-30:
Resolved error by installing kotlin. 
`brew install kotlin`
Generated chip tool libs successfully. 
But actually when I have checkout for older commit (3c6bed3bb28f66cc2e1ebbe3561fa0838f6b7c51), it starts failing again.

KhushbuShah25, 2024-05-01:
Btw, I am trying to build in Macbook pro which is having Apple M3 Pro chip. Anyone has faced python environment issue for this ?

VinhBH1, 2024-06-04:
Hi @KhushbuShah25  
when you Generated chip tool libs successfully. Where is the location ? . i am still stuck in erro Unresolved reference: chip 
.i have run 
TARGET_CPU=arm64 ./scripts/examples/android_app_ide.sh succesfully but it still have error Unresolved reference: chip 
[image]

joonhaengHeo, 2024-06-11:
I have the same issue and I can't figure out how to solve it.
https://github.com/project-chip/connectedhomeip/issues/31271

NagendraMudadla, 2024-07-12:
> Resolved error by installing kotlin. `brew install kotlin` Generated chip tool libs successfully. But actually when I have checkout for older commit ([3c6bed3](https://github.com/project-chip/connectedhomeip/commit/3c6bed3bb28f66cc2e1ebbe3561fa0838f6b7c51)), it starts failing again.

Can you share the latest build libraries with me, because my system is hanging while generating the chip tool, libraries.

vulpes2, 2024-07-21:
@KhushbuShah25 You can install older versions of the NDK with the SDK Manager if you check the "Show Package Details" option at the bottom right corner.

A quick guide for people who want to build the chip-tool Android app on Linux with the scripts, with solutions to all the common problems I've encountered so far collected in one place. Adjust the paths as required.

1. Follow the docs and install the correct version of Android SDK and NDK, and make sure your java version is **old enough**, 11 worked for me.

```
export ANDROID_HOME=~/Android/Sdk
export ANDROID_NDK_HOME=~/Android/Sdk/ndk/23.2.8568313
export JAVA_HOME=/usr/lib/jvm/java-11-openjdk-11.0.23.0.9-1.fc40.x86_64
```
2. **Install Kotlin** from https://github.com/JetBrains/kotlin/releases and make sure `kotlinc` is available in your `$PATH`, version 1.9.25 worked for me.
3. Run `source scripts/bootstrap.sh`, or `source scripts/activate.sh` if you have already ran the bootstrap script before.
4. Run `./scripts/build/build_examples.py --target android-arm64-chip-tool build`

Common issues:

1. `ninja: error: loading 'build.ninja': No such file or directory`: run `gn gen out/android-arm64-chip-tool` to fix it and try again.
2. `FileNotFoundError: [Errno 2] No such file or directory: 'gn'`: you are not in the pigweed venv, run `source scripts/activate.sh` and try again.
3. `kotlinc: command not found`: you did not install kotlin correctly or it's not available in your `$PATH`, symlink `kotlinc` to a folder that is in `$PATH`.

closed 2024-10-28 by yunhanw-google