project-chip/connectedhomeip #33237: [1.2] When the ColorTemperatureMireds in the lighting-app is adjusted to 0xFEFF, the Google Home UI displays green.

liujie1f23, 2024-04-30:
### Reproduction steps

1、Add test device(ClassificationID:0x010D) of **lighting-app** to the Google Home
2、Select the device and turn on the device of lighting-app 
3、Turn on the UI of "Light colour" and select  the item of "Temperature"
4、Adjust the bar of the UI to the top of **Temperature**(**ColorTemperatureMireds**=0xFEFF) and the color of the UI which is represented the lighting level  will be **green**,and the device worked as expected to keep **warm** color.
     Test result reveals that when the Mireds value corresponding to the color temperature exceeds **0x595**, the UI indicating the lighting level turns **green**, whereas it should display  **warm**, reddish tint at this point. Adjusting the level value **does not change** the color from green, and it remains so until the Mireds value, corresponding to a higher color temperature, is set to be less than or equal to **0x595**. then the UI for the lighting level correctly displays the intended colors.
5、Reproduced the phenomenon using connectedhomeip/examples/lighting-app/**linux** routines 
ps:
example.lighting-app.linux.txt：the log of chip-lighting-app("sudo out/debug/chip-lighting-app --ble-device 0)
GoogleHomeMiredFeffGreen.mp4:Phenomenon video

### Bug prevalence

Whenever i do this,it will occur

### GitHub hash of the SDK that was being used

52e4a6435f74b20fb1bf355cc67587c3a3853488

### Platform

raspi, other

### Platform Version(s)

1.2

### Type

Platform Issue, Core SDK Performance Improvement

liujie1f23, 2024-04-30:
[example.lighting-app.linux.txt](https://github.com/project-chip/connectedhomeip/files/15161045/example.lighting-app.linux.txt)

https://github.com/project-chip/connectedhomeip/assets/28770475/68095df3-93d8-4737-991f-1791e7311213

liujie1f23, 2024-04-30:
GoogleHomeVersion:3.15.1.4 
Android Version:12
Phone Model:Galaxy Note 10
[image]

bzbarsky-apple, 2024-04-30:
@liujie1f23 This sounds like an issue to be reported to Google about their UI, not a core Matter problem....

liujie1f23, 2024-05-06:
@bzbarsky-apple
Can the issue be circumvented by modifying the Matter code without altering the Google UI?
Will a member from Google step in to help analyze this issue, or should we report it on a Google forum instead?

Best regards

bzbarsky-apple, 2024-05-07:
@liujie1f23 You should report it to Google through their reporting channels.

andy31415, 2024-10-24:
Please report google integration issues through google reporting channels. See https://project-chip.github.io/connectedhomeip-doc/issue_triage.html#google-integration

closed 2024-10-24 by andy31415