project-chip/connectedhomeip #33244: openCommissionWindow does not work in iOS[BUG]

aryanarayantiwari, 2024-04-30:
### Reproduction steps

Simply setup a matter controller and add an accesory using MatterSupport framework in swift and then use the openCommissionWindow function. I have tried both setups one that generates a random passcode and the one with a discriminator. However both times I receive errors even when trying with MTRBasicAdministratorCluster() I still receive errors.
The errors are as follows :- 
Timeout error
Mdns Error
Malformed Values in Cluster
etc.

I would like to have a good example of how to implement this functionality such that sharing a controller works. Pointing to a example or a good documentation would help too.

### Bug prevalence

each time

### GitHub hash of the SDK that was being used

Darwin

### Platform

darwin

### Platform Version(s)

1.0

bzbarsky-apple, 2024-04-30:
@aryanarayantiwari This is literally something we have automated tests for, and it works fine in those.

If it's not working for you, and you want help, you probably need to provide more information about what _exactly_ you are doing (calling what APIs, with what arguments).  And possibly logs, but let's start with the exact API calls you are making.

aryanarayantiwari, 2024-04-30:
@bzbarsky-apple I am using [openCommissioningWindow(withDiscriminator:duration:queue:completion:)](https://developer.apple.com/documentation/matter/mtrbasedevice/4226882-opencommissioningwindow) api and the arguments that are passed are:- 
Discriminator: 3840
duration: 30
queue: .main

bzbarsky-apple, 2024-04-30:
Well, 30 is not a valid value for duration.  That's why you get " Malformed Values in Cluster".

I wish the documentation you linked to included the docs on that function.   See https://github.com/project-chip/connectedhomeip/blob/335ae1912d9356e3710fb74ee1e7d1ddeda9515e/src/darwin/Framework/CHIP/MTRBaseDevice.h#L449-L462

"duration" is documented as being seconds.  Per spec, valid values are between 3 and 15 minutes, so 180-900 seconds.

aryanarayantiwari, 2024-05-01:
> > I was wondering if it's documented that the commission window supports duration between 3 to 15 minutes.
> 
> It's in the published Matter specification, no? Just like all sorts of other constraints on various arguments.
> 
> But yes, in terms of API documentation there are some lacks here.

Yes It's mentioned in the [Announcement Section(5.4.2.3) of Matter Specification](https://csa-iot.org/wp-content/uploads/2022/11/22-27349-001_Matter-1.0-Core-Specification.pdf) at page 239. Thanks for the help @bzbarsky-apple

closed 2024-05-01 by aryanarayantiwari

aryanarayantiwari, 2024-05-15:
Hi! @bzbarsky-apple . Could you please guide me onto how to discover what type of device I'm pairing with, meaning what kind of clusters does it support I would appreciate it if you would help.

bzbarsky-apple, 2024-05-16:
You could read ServerList from the Descriptor cluster on all the endpoints, right?