project-chip/connectedhomeip #33245: [1.3] Stress Test of Pair and Unpair on Raspi accessory Failed in some iterations

cjandhyala, 2024-04-30:
### Reproduction steps

1. Using ToT , build all-cluster-app on raspi and build the python controller. 
2. Using the stress test scripts, execute pair and unpair in a loop of iterations. Script flow is , 

- Factory reset DUT (all-cluster-app running on raspi)
- pair DUT (using on-network)
- sleep 2 secs 
- unpair DUT
- repeat the above steps in a loop.

### Bug prevalence

random

### GitHub hash of the SDK that was being used

8a4dffc5dd20d5343c6ea384390aeedb0c6b2e4c

### Platform

raspi

### Type

Core SDK Crash

### Anything else?

multiple iterations failed with following Matter stack signature. logs attached for all these fail scenarios. 

CommissionFailure(Pairing with DUT failed Reason src/app/CommandSender.cpp:328: CHIP Error 0x00000032: Timeout
CommissionFailure(Pairing with DUT failed Reason src/protocols/secure_channel/CASESession.cpp:1984: CHIP Error 0x000000C9: No shared trusted root ))
CommissionFailure(CommissionFailure(Pairing with DUT failed Reason src/app/ClusterStateCache.cpp:303: CHIP Error 0x00000010: Key not found )
CommissionFailure(Pairing with DUT failed Reason src/controller/python/ChipDeviceController-ScriptDevicePairingDelegate.cpp:89: CHIP Error 0x00000003: Incorrect state ))

[logs.zip](https://github.com/project-chip/connectedhomeip/files/15171169/logs.zip)

gokul-bhalodiya-7, 2024-05-31:
Hope this may help you https://github.com/project-chip/connectedhomeip/issues/33688#issuecomment-2143007459

cjandhyala, 2024-06-17:
closing this issue, following it up in another other issue.

closed 2024-06-17 by cjandhyala