project-chip/connectedhomeip #33287: [BUG] Node resolution attempt marked complete incorrectly.

su-shanka, 2024-05-03:
### Reproduction steps

In MinMdns Resolver (Resolver_ImplMinimalMdns.cpp), when `resolver->Take()` fails for `ResolvedNodeData`, the corresponding active resolver is marked complete and `OnOperationalNodeResolved` is called with arbitrary garbage in `ResolvedNodeData`.

### Bug prevalence

NA

### GitHub hash of the SDK that was being used

fcfc9bc

### Platform

core

bzbarsky-apple, 2024-05-03:
@andy31415

andy31415, 2024-05-03:
@su-shanka do you have a path that makes Take fail?

The bug seems valid, however I also wonder under what operations we would have that code path taken.

andy31415, 2024-05-03:
Basically what happens is that `Take` will verify that `IsActiveOperationalParse()` and return an error only in that case. The unchecked Take is on a `if IsActiveOperationalParse` branch, so Take would never fail.

I will fix the code to always look ok, however right now I am not sure there is a path to trigger the bad behavior.

PR #33296 (2024-05-03): Add an explicit skip processing on error (do not attempt to use invalid data)
This fixes a missing `continue` on error when parsing operational data.

I do not believe this error path of using invalid data can actually be reached since the only way the Take will fail is if `IsActiveOperationalBrowse` is false and this is exactly the top condition on that branch. However correctness-wise, if the Take can fail, we should propperly handle the failure.

This fixes #33287
closed 2024-05-03 by mergify[bot]