project-chip/connectedhomeip #33326: [CERT-TEST-FAILURE] Test_TC_CC_5_3 Min/Max values checked in step 3d and 3c are the same which fails transitions

Apollon77, 2024-05-07:
### Feature Area

Other

### Test Case

TC_CC_5_3

### Reproduction steps

The testcase Test_TC_CC_5_3 is testing some hiue and saturation transitions. Here in step 3c some values are checked after 10s of transitionioning. Then it waits another 5s and checks in 3d again ... but here the same values are used which indeed fails becaus ethe transition is 5s further in time.

https://github.com/project-chip/connectedhomeip/blob/master/src/app/tests/suites/certification/Test_TC_CC_5_3.yaml#L154-L170

When I use the same rate for min/max used between 3b->3c and half them I come to the followijg boundaries

```
    - label: "Step 3d: TH reads CurrentX attribute from DUT"
      command: "readAttribute"
      PICS: CC.S.F03 && CC.S.A0003 && CC.S.C08.Rsp
      attribute: "CurrentX"
      response: 
          constraints: 
              minValue: 5925
              maxValue: 10075
      
    - label: "Step 3d: TH reads CurrentY attribute from DUT"
      command: "readAttribute"
      PICS: CC.S.F03 && CC.S.A0004 && CC.S.C08.Rsp
      attribute: "CurrentY"
      response:
          constraints: 
              minValue: 10850
              maxValue: 14150
```

I came across this while implementing ColorControl for matter.js.
Here the values while checking 3b are:
* currentX: 13000
* currentY: 14000

And in 3d they are:
* currentX: 8000
* currentY: 12500

The other question would be why chip tool tests in CI did not have the same issue, unless they somehow transition differently (which would be very strange).

### Bug prevalence

always for me

### GitHub hash of the SDK that was being used

master

### Platform

core

ReneJosefsen, 2024-05-08:
Hmm, looking at the test plan it seems to indicate that the transition/step should end at 3c. 3a starts the step with a transition of 20s and both 3b and 3c waits for 10s respectively, so once 3d have waited 5s it should not have changed from 3c. With that said, there does seems to be a slight discrepancy between the ranges mentioned in some of the steps and what is used in the yaml script e.g. 3b and 3d 🤔 

But I do think that the transition should stop for 3c so while the range does not match the test plan, I do think the intention of verifying the transition has stopped is okay.

Apollon77, 2024-05-08:
Hm ... Ohh you are right ... I somehow missed the 20s and read 30s ... hhm ... ok ... I found a bug in my Step implementation. Thank you very much! Then this test is completely fine for now!

closed 2024-05-08 by Apollon77