project-chip/connectedhomeip #33373: [Test Failed] EFR32 TestExchange.cpp asserts

jmartinez-silabs, 2024-05-08:
### Test issue(s)

While Running the test driver on efr32 hardware,
the device asserts in the test `TestExchange.cpp ; CheckBasicMessageRounTrip` in the function 

```
CHIP_ERROR ExchangeContext::SendMessage(Protocols::Id protocolId, uint8_t msgType, PacketBufferHandle && msgBuf,
                                        const SendFlags & sendFlags)
```
line 
`VerifyOrReturnError(mExchangeMgr != nullptr, CHIP_ERROR_INTERNAL);`

Before the assert, logs show events failing to get posted to the chip event queue because the queue is full. 

### Platform

efr32

bzbarsky-apple, 2024-05-09:
@jmartinez-silabs Is something supposed to be pumping that queue?  I would have expected `ctx.DrainAndServiceIO();` to do that sort of thing, but maybe the cut-down event loop it's using is broken on efr32?  If so, I expect lots of other message-send tests are failing there for similar reasons?

jmartinez-silabs, 2024-05-09:
@bzbarsky-apple Trying to run any test under src/messaging/tests/BUILD.gn ran is the seemingly same issue.

We had a series of tests running fine on our hardware not long ago. 
Currently, there I am looking into  an issue where strangely no tests are run.. They are compiled and part of the binary, But nothing runs. I'm not sure what is occurring yet and therefore can't say if the event loop is generally broken on the ef32 test driver.

The only one running yesterday was actually `TestExchange`

```
python -m nl_test_runner.nl_test_runner -d /dev/cu.usbmodem0004402662211 -o out3.log

All tests completed
Total of all tests failed: 0 of 0
```

I'll try to find and fix it so all built tests are run then investigate the event loop

PR #33430 (2024-05-14): [Silabs]Fix efr32 test driver event loop, run gtests, enable more test on efr…
fixes #10447
fixes #33373

- gtest were not run by the runner. add Call `chip::test::RunAllTests();`
- event loop processing in tests changed and was not working on efr32 anymore.  Add `CHIP_SYSTEM_CONFIG_USE_OPEN_THREAD_ENDPOINT` conditions in TestInetCommonOptions.cpp and TestInetCommonPosix.cpp to enable it.
- re-enable MessagingLayerTests tests.
closed 2024-05-14 by mergify[bot]