project-chip/connectedhomeip #33413: [1.3] The default null values from the Thermostat cluster are not correctly set

kkasperczyk-no, 2024-05-13:
### Reproduction steps

It seems that the default value NULL value for the attributes from Thermostat cluster like LocalTemperature or OutdoorTemperature is not correctly set. I have this attributes filled with NULL value in the .zap file and I have not modified them in my application, but the IsNull check returns false and the value is 0.
 
Even more interesting is that after performing read operation for these attributes using CHIPTool I got:

[1715586371.626553][74241:74243] CHIP:DMG: ReportDataMessage =
[1715586371.626561][74241:74243] CHIP:DMG: {
[1715586371.626567][74241:74243] CHIP:DMG: 	AttributeReportIBs =
[1715586371.626579][74241:74243] CHIP:DMG: 	[
[1715586371.626585][74241:74243] CHIP:DMG: 		AttributeReportIB =
[1715586371.626596][74241:74243] CHIP:DMG: 		{
[1715586371.626602][74241:74243] CHIP:DMG: 			AttributeDataIB =
[1715586371.626610][74241:74243] CHIP:DMG: 			{
[1715586371.626618][74241:74243] CHIP:DMG: 				DataVersion = 0x6972abdc,
[1715586371.626625][74241:74243] CHIP:DMG: 				AttributePathIB =
[1715586371.626633][74241:74243] CHIP:DMG: 				{
[1715586371.626641][74241:74243] CHIP:DMG: 					Endpoint = 0x1,
[1715586371.626649][74241:74243] CHIP:DMG: 					Cluster = 0x201,
[1715586371.626657][74241:74243] CHIP:DMG: 					Attribute = 0x0000_0001,
[1715586371.626664][74241:74243] CHIP:DMG: 				}
[1715586371.626674][74241:74243] CHIP:DMG: 					
[1715586371.626684][74241:74243] CHIP:DMG: 				Data = -1, 
[1715586371.626690][74241:74243] CHIP:DMG: 			},
[1715586371.626705][74241:74243] CHIP:DMG: 			
[1715586371.626714][74241:74243] CHIP:DMG: 		},
[1715586371.626730][74241:74243] CHIP:DMG: 		
[1715586371.626738][74241:74243] CHIP:DMG: 	],
[1715586371.626753][74241:74243] CHIP:DMG: 	
[1715586371.626762][74241:74243] CHIP:DMG: 	SuppressResponse = true, 
[1715586371.626772][74241:74243] CHIP:DMG: 	InteractionModelRevision = 11
[1715586371.626781][74241:74243] CHIP:DMG: }
[1715586371.626848][74241:74243] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0201 Attribute 0x0000_0001 DataVersion: 1769122780
**[1715586371.626868][74241:74243] CHIP:TOO:   OutdoorTemperature: -1**

I don't know when the behavior changed, but on the 1.2.0.1 tag it was working fine, and the read value in the CHIPTool was null.

I can be worked around by setting null value to these attributes on init and then read works as expected, but this is probably not the best solution.

 

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

v1.3.0.0 tag

### Platform

core

### Platform Version(s)

1.3

### Type

Spec Compliance Issue

bzbarsky-apple, 2024-05-13:
The XML says: `type="temperature"`

The type definition says: `<type id="0xD8" description="Temperature"              name="temperature"       size="2"  analog="true"     signed="true" />`

The ZAP output for that null default is `0xFFFF`, which is NOT how one represents null for a _signed_ 2-byte type.

This needs to be fixed in https://github.com/project-chip/zap/issues

Looks like #30758 changed the type in the XML from `int16s` to `temperature`, and we have no useful test coverage.

paulr34, 2024-05-29:
@kkasperczyk-no how did you set NULL in your .zap file?

bzbarsky-apple, 2024-05-30:
@paulr34 By clicking the "NULL" button in the ZAP UI.

kkasperczyk-no, 2024-06-03:
> @paulr34 By clicking the "NULL" button in the ZAP UI.

@bzbarsky-apple actually it didn't work for me. I had to workaround it in my application by calling `app::Clusters::Thermostat::Attributes::LocalTemperature::SetNull();`

bzbarsky-apple, 2024-06-08:
@kkasperczyk-no Sure, that's what @paulr34 is trying to fix in ZAP...  He was just asking about steps to reproduce the ZAP issue.

closed 2024-06-10 by mergify[bot]