project-chip/connectedhomeip #33421: [BUG] Commissioning android tv-app fails: tv-app can't be discovered

JelenaZdr, 2024-05-13:
### Reproduction steps

The platform app is initiated on the phone, while the chip tool is launched on the computer running Ubuntu 20. However, during commissioning (command: pairing code ), the chip tool fails to discovered the platform app. This issue persists even when utilizing a third-party controller.

### Bug prevalence

Whenever i do this

### GitHub hash of the SDK that was being used

bcebbc6124620d992ca4a8f7816cafbc42043cf7

### Platform

android

### Anything else?

[bug_no_advertising_tv_app_android.txt](https://github.com/project-chip/connectedhomeip/files/15294543/bug_no_advertising_tv_app_android.txt)

xvortep, 2024-05-27:
@bzbarsky-apple @andy31415 @yunhanw-google @tennessee-google  Hi, can you please take a look at this? This is a serious blocker for us. Thank you.

sharadb-amazon, 2024-05-30:
Is operational discovery (service type: \_matter.\_tcp) failing or something else? Can you attach logs from both the tv-app as well as the chip-tool?

Also, can you use any third-party discovery tool (like "Discovery" app for iOS/MacOS or "Service Browser" to see if the above service type can be discovered using them? Or test using the dns-sd / avahi CLI commands?

JelenaZdr, 2024-05-31:
Hello, atteched are the logs from both application. 

Also, we tried using a thrid-parthy controller and had the same problem

[logChipTool.txt](https://github.com/user-attachments/files/15510989/logChipTool.txt)
[logTV.txt](https://github.com/user-attachments/files/15510990/logTV.txt)

bzbarsky-apple, 2024-06-01:
That last logTV.txt: I don't see it ever trying to advertise anything with CM=1.

Which means it does not have a commissioning window open, so of course commissioning it will not work.

xvortep, 2024-06-03:
Do you know what could be a problem here? Do we have to manually open the commissioning window, or are there some errors that prevent the app to do that automatically?
If there's a problem with DNS implementation used in this app, we could try changing it to "minimal". Do you think that's a good idea?

Thanks!

bzbarsky-apple, 2024-06-08:
> Do we have to manually open the commissioning window, or are there some errors that prevent the app to do that automatically?

It depends on the situation.  Why do you expect the commissioning window to be open in this case?

xvortep, 2024-06-10:
> > Do we have to manually open the commissioning window, or are there some errors that prevent the app to do that automatically?
> 
> It depends on the situation. Why do you expect the commissioning window to be open in this case?

Maybe I didn't understand your question, the device should be commissionable right away, right? So the commissioning window should be open on app start, like in the implementation of a linux tv-app.

JelenaZdr, 2024-07-24:
In file examples/tv-app/tv-common/include/CHIPProjectAppConfig.h,  CHIP_DEVICE_CONFIG_ENABLE_PAIRING_AUTOSTART  is set to 1. However, during commissioning (command: pairing code ), the chip tool fails to discovered the platform app. 
[logTV.txt](https://github.com/user-attachments/files/16359943/logTV.txt)
[logChipTool.txt](https://github.com/user-attachments/files/16359946/logChipTool.txt)

xvortep, 2024-07-24:
To be clear - we want to create an android tv device which can be commissioned like every other device, we don't want to use UDC. Linux app can be commissioned like that, but android app can't and we need a way to do so. 

Thank you very much