project-chip/connectedhomeip #33442: CHIP Error 0x00000032 when trying to send cluster commands

mulligan252, 2024-05-14:
### Reproduction steps

I have an ESP32-C6-DevKitM-1. I have flashed this with the Matter light example from the esp-matter SDK. I am using Ubuntu 22.04 as a Virtual machine on Windows for doing the cross compilation and running chip-tool etc. 

I then commission my device as follows : `chip-tool pairing ble-wifi 0x7283 'My Network' 123456789 20202021 3840`

This all works fine. My device connects to my local network and and I can ping its IP address etc.

I don't know, why but I am unable then to turn on or off the light, or even send any cluster command using the chip tool for example : 

`chip-tool onoff toggle 0x7283 0x01`

I get the following logs below :  

[code][1715694336.819940][4472:4472] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_tool_kvs
[1715694336.821260][4472:4472] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1715694336.821301][4472:4472] CHIP:DL: ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_kvs
[1715694336.822772][4472:4472] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1715694336.822880][4472:4472] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1715694336.822947][4472:4472] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1715694336.823209][4472:4472] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-WSN4ht)
[1715694336.823563][4472:4472] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1715694336.823929][4472:4472] CHIP:DL: NVS set: chip-counters/reboot-count = 17 (0x11)
[1715694336.824140][4472:4472] CHIP:DL: Got Ethernet interface: enp0s3
[1715694336.824322][4472:4472] CHIP:DL: Found the primary Ethernet interface:enp0s3
[1715694336.824621][4472:4472] CHIP:DL: Failed to get WiFi interface
[1715694336.824671][4472:4472] CHIP:DL: Failed to reset WiFi statistic counts
[1715694336.824715][4472:4472] CHIP:IN: UDP::Init bind&listen port=0
[1715694336.824781][4472:4472] CHIP:IN: UDP::Init bound to port=46740
[1715694336.824798][4472:4472] CHIP:IN: BLEBase::Init - setting/overriding transport
[1715694336.824813][4472:4472] CHIP:IN: TransportMgr initialized
[1715694336.824873][4472:4472] CHIP:FP: Initializing FabricTable from persistent storage
[1715694336.824918][4472:4472] CHIP:TS: Last Known Good Time: 2023-10-14T01:16:48
[1715694336.825519][4472:4472] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x35CFE666CE838CD1, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1715694336.826572][4472:4472] CHIP:ZCL: Using ZAP configuration...
[1715694336.827095][4472:4472] CHIP:CTL: System State Initialized...
[1715694336.827193][4472:4472] CHIP:CTL: Setting attestation nonce to random value
[1715694336.827270][4472:4472] CHIP:CTL: Setting CSR nonce to random value
[1715694336.827431][4472:4474] CHIP:DL: CHIP task running
[1715694336.827562][4472:4474] CHIP:DL: HandlePlatformSpecificBLEEvent 32786
[1715694336.827676][4472:4474] CHIP:CTL: Setting attestation nonce to random value
[1715694336.827716][4472:4474] CHIP:CTL: Setting CSR nonce to random value
[1715694336.827934][4472:4474] CHIP:CTL: Generating NOC
[1715694336.828143][4472:4474] CHIP:FP: Validating NOC chain
[1715694336.828544][4472:4474] CHIP:FP: NOC chain validation successful
[1715694336.828586][4472:4474] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1715694336.828602][4472:4474] CHIP:TS: Last Known Good Time: 2023-10-14T01:16:48
[1715694336.828617][4472:4474] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1715694336.828631][4472:4474] CHIP:TS: Retaining current Last Known Good Time
[1715694336.829780][4472:4474] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1715694336.835566][4472:4474] CHIP:TS: Committing Last Known Good Time to storage: 2023-10-14T01:16:48
[1715694336.845253][4472:4474] CHIP:CTL: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 35CFE666CE838CD1)
[1715694336.859605][4472:4474] CHIP:TOO: Sending command to node 0x7283
[1715694336.860118][4472:4474] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000007283]
[1715694336.860377][4472:4474] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1715694336.860615][4472:4474] CHIP:DIS: OperationalSessionSetup[1:0000000000007283]: State change 1 --> 2
[1715694337.063755][4472:4474] CHIP:DIS: Checking node lookup status after 203 ms
[1715694356.834065][4472:4472] CHIP:CTL: Shutting down the commissioner
[1715694356.834211][4472:4472] CHIP:CTL: Shutting down the controller
[1715694356.834268][4472:4472] CHIP:DIS: OperationalSessionSetup[1:0000000000007283]: Cancelling incomplete address resolution as device is being deleted.
[1715694356.834350][4472:4472] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1715694356.834384][4472:4472] CHIP:FP: Forgetting fabric 0x1
[1715694356.834424][4472:4472] CHIP:TS: Pending Last Known Good Time: 2023-10-14T01:16:48
[1715694356.834519][4472:4472] CHIP:TS: Previous Last Known Good Time: 2023-10-14T01:16:48
[1715694356.835051][4472:4472] CHIP:TS: Reverted Last Known Good Time to previous value
[1715694356.835101][4472:4472] CHIP:CTL: Shutting down the commissioner
[1715694356.835134][4472:4472] CHIP:CTL: Shutting down the controller
[1715694356.835167][4472:4472] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1715694356.835263][4472:4472] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1715694356.835397][4472:4472] CHIP:FP: Shutting down FabricTable
[1715694356.835434][4472:4472] CHIP:TS: Pending Last Known Good Time: 2023-10-14T01:16:48
[1715694356.835502][4472:4472] CHIP:TS: Previous Last Known Good Time: 2023-10-14T01:16:48
[1715694356.835605][4472:4472] CHIP:TS: Reverted Last Known Good Time to previous value
[1715694356.836108][4472:4472] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-sI6X0o)
[1715694356.836685][4472:4472] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1715694356.837441][4472:4472] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1715694356.837514][4472:4472] CHIP:DL: Inet Layer shutdown
[1715694356.837539][4472:4472] CHIP:DL: BLE shutdown
[1715694356.837595][4472:4472] CHIP:DL: System Layer shutdown
[1715694356.837755][4472:4472] CHIP:TOO: Run command failure: examples/chip-tool/commands/common/CHIPCommand.cpp:617: CHIP Error 0x00000032: Timeout[/code]

### Bug prevalence

always 

### GitHub hash of the SDK that was being used

5bb5c9e23d532cea40476fc0bd1d3008522792ba

### Platform

esp32

### Type

Platform Issue

bzbarsky-apple, 2024-05-14:
Well, chip-tool can't discover your device over DNS-SD.  Is the device advertising?  Is the chip-tool on the same wi-fi network as the device?  @mulligan252

bzbarsky-apple, 2024-05-14:
That said, if commissioning succeeded, presumably chip-tool could find the advertisement at that point....

Does it help to pass a longer `--timeout` value for the toggle attempt?

mulligan252, 2024-05-14:
> Well, chip-tool can't discover your device over DNS-SD. Is the device advertising? Is the chip-tool on the same wi-fi network as the device? @mulligan252

@bzbarsky-apple Thanks for your reply. The chip-tool is on the same wi-fi network as the device. Commissioning appears to be successful. Below are the logs when the device boots up every time. There are some wi-fi connection issues initially, but seems to resolve itself eventually. I can ping the device on the network via its IP address from any computer then. 

```
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c410,len:0xe58
load:0x4086e610,len:0x2e20
entry 0x4086c41a
I (23) boot: ESP-IDF v5.2.1 2nd stage bootloader
I (24) boot: compile time May 10 2024 11:16:46
I (24) boot: chip revision: v0.0
I (26) boot.esp32c6: SPI Speed      : 80MHz
I (31) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (64) boot:  1 nvs              WiFi data        01 02 00010000 0000c000
I (72) boot:  2 nvs_keys         NVS keys         01 04 0001c000 00001000
I (79) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (87) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (94) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (102) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (109) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (117) boot: End of partition table
I (121) esp_image: segment 0: paddr=00020020 vaddr=42140020 size=42288h (270984) map
I (186) esp_image: segment 1: paddr=000622b0 vaddr=40800000 size=0dd68h ( 56680) load
I (200) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=13e9f8h (1305080) map
I (469) esp_image: segment 3: paddr=001aea20 vaddr=4080dd68 size=0e264h ( 57956) load
I (484) esp_image: segment 4: paddr=001bcc8c vaddr=4081bfd0 size=03ec8h ( 16072) load
I (488) esp_image: segment 5: paddr=001c0b5c vaddr=50000000 size=00004h (     4) load
I (496) boot: Loaded app from partition at offset 0x20000
I (497) boot: Disabling RNG early entropy source...
I (512) cpu_start: Unicore app
W (521) clk: esp_perip_clk_init() has not been implemented yet
I (528) cpu_start: Pro cpu start user code
I (528) cpu_start: cpu freq: 160000000 Hz
I (528) cpu_start: Application information:
I (531) cpu_start: Project name:     light
I (536) cpu_start: App version:      1.0
I (540) cpu_start: Compile time:     May 10 2024 11:16:34
I (546) cpu_start: ELF file SHA256:  ba163e08d...
I (552) cpu_start: ESP-IDF:          v5.2.1
I (557) cpu_start: Min chip rev:     v0.0
I (561) cpu_start: Max chip rev:     v0.99
I (566) cpu_start: Chip rev:         v0.0
I (571) heap_init: Initializing. RAM available for dynamic allocation:
I (578) heap_init: At 40832140 len 0004A4D0 (297 KiB): RAM
I (584) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (590) heap_init: At 50000004 len 00003FE4 (15 KiB): RTCRAM
I (597) spi_flash: detected chip: generic
I (601) spi_flash: flash io: dio
W (606) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (617) sleep: Configure to isolate all GPIO pins in sleep state
I (623) sleep: Enable automatic switching of GPIO sleep configuration
I (630) coexist: coex firmware version: 77cd7f8
I (635) coexist: coexist rom version 5b8dcfa
I (640) main_task: Started on CPU0
I (640) main_task: Calling app_main()
I (660) led_indicator: LED Indicator Version: 0.9.2
I (660) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (660) led_indicator: Indicator create successfully. type:LED Strips mode, hardware_data:0x408367e0, blink_lists:custom
I (670) button: IoT Button Version: 3.2.0
I (680) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (690) app_main: Light created with endpoint_id 1
[... 205 lines cut ...]
I (6800) chip[SWU]: Starting the periodic query timer, timeout: 86400 seconds
I (6800) chip[IM]: No subscriptions to resume
I (6810) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000002A's Attribute 0x00000002 is 0 **********
I (6820) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x0000002A's Attribute 0x00000002 is 1 **********
I (6830) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000002A's Attribute 0x00000003 is 0 **********
I (6850) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x0000002A's Attribute 0x00000003 is null **********
I (7690) chip[DL]: IP_EVENT_GOT_IP6
I (7690) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fd00:0000:0000:0000:5632:04ff:fe07:177c
I (7700) ROUTE_HOOK: Hook already installed on netif, skip...
I (7700) app_main: Interface IP Address changed
I (7710) chip[DIS]: Updating services using commissioning mode 1
I (7710) chip[DIS]: CHIP minimal mDNS started advertising.
I (7730) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1 cp=0
I (7730) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: D6AE8150D313F113.
I (7750) chip[DIS]: mDNS service published: _matterc._udp
I (31580) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)
I (31580) chip[DL]: Device already advertising, stop active advertisement and restart
I (31590) NimBLE: GAP procedure initiated: stop advertising.

I (31590) NimBLE: GAP procedure initiated: advertise;
I (31600) NimBLE: disc_mode=2
I (31600) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (31610) NimBLE:

```

bzbarsky-apple, 2024-05-15:
@mulligan252 That log looks fine as far as it goes, but only shows commissionable advertising, not operational.

Might I also suggest attaching, not pasting logs, to keep the issue readable?

In any case, the chip-tool log above shows that it tries to resolve the device's IP and never manages to do that before the command timeout.  Hence my suggestion to test whether raising the timeout (ideally to > 45 seconds, which is the resolution timeout) helps anything.  Even if it's just seeing whether resolution then times out.

dhrishi, 2024-10-25:
@pavel808 Is there any updates you can provide here? Please close the issue if it works fine

closed 2024-12-18 by pavel808

pavel808, 2024-12-18:
@bzbarsky-apple Apologies for the delay and thanks for your help. I have since just started using the ESP Matter Controller integrated into my code, and there are no issues like mentioned above. So it's better to close this issue now.